vroland/epdiy #3: Terminal example 'crashes' USB Serial chip

sebastius, 2020-05-24:
Somehow the terminal example annoys the USB Serial chip so much it causes to re-enumerate on the USB bus :sunglasses: 

Tried it with the Arduino IDE serial terminal, screen and IDF monitor. All three are gloriously killed :+1: 

Could be a shitty driver on my Linux Mint, could be a bad chip. Haven't seen this behaviour before in other USB serial chips, this is my first board with the CH330.

sgeisler, 2020-05-24:
Interesting, when does that happen? I haven't actively experienced that so far, but maybe I just didn't notice? But my setup is a bit hacky too: let IDF monitor configure the serial port and then pipe the output of `script` to it to mirror a local shell for testing.

sebastius, 2020-05-24:
Oh it might be power related, it hangs my entire hub.

vroland, 2020-05-25:
Ha, that's interesting. So far I have not been able to reproduce it with neither of my machines... 
Which kernel version are you using? And it does not happen with the other examples?

sebastius, 2020-05-25:
It seems to be a problem when i'm doing both serial AND updating the eink. Maybe i ordered the wrong parts and have an overcurrent problem. Or a lack of capacitors on the main 5v and 3.3v rails. No clue yet.

It also breaks the USB connection in Windows. Its probably hardware :)

vroland, 2020-05-25:
Hm, jup, seems to be hardware. When I've got the time I can test it on a Raspberry Pi, which is not known for its particularly stable usb power, so I might be able to reproduce it there.

vroland, 2020-05-26:
Hm, I've tried it with my raspberry pi, still no unexpected behaviour :/

sebastius, 2020-05-27:
[image]

My problem is definetly in the capacitor area (this is the 3.3v rail at the CH330N chip) . Will debug further!

sebastius, 2020-05-27:
More digging: this also happens on the 5v rail. Added a 10nF cap to the CH330N, no effect.

sebastius, 2020-05-28:
Quick question: the LT1945 seems to be powered of the 3.3v rail instead of the 5v rail. Why is that?

sebastius, 2020-05-28:
I fixed it. So what i did was cut the trace supplying the P-fet with 3.3v and hooked it up to the 5v rail. Apparantly the LT1945 and subsequent stuff draw too much power and they destroy the 3.3v rail. I noticed in the PDF that an older version had it hooked up to 5v, so i took a chance.

vroland, 2020-05-28:
Thanks a lot, @sebastius! And yes, as you mentioned in https://github.com/vroland/epdiy/issues/4, I should really update some artifacts.
The LT1945 was hooked up to the 3.3v rail to be able to power the board via 3.3v only (without USB), as this was in-spec according to the data sheet.
Have you tried adding a larger cap on the 3.3V rail? I'm just curious if there is an easy way to fix that, which does not drop support for 3.3V operation... Due to the Covid-19 Situation I do not have access to an ocscilloscope, buf it looks loke it could definitely be useful to have a look at such things.

sebastius, 2020-05-28:
i've doubled the 3.3v and 5v caps (both now 2x 10uF) but to no avail. 

I recommend reverting to the 5v supply on that signal but allow people to use a range of voltages there. Anything above 3.3v and below 12v will probably work (depending on the mosfet, if the CH330n isn't present and the caps are capable of that voltage. 

I've left the pullup R1 on the 3.3v rail by the way.

vroland, 2020-05-28:
Hey, that sounds reasonable. I'm just learning by doing here myself, since my background is more in software than in hardware ;)
Btw, could you me which exact regulators you used? Just to compare the model with mine.

sebastius, 2020-05-28:
Solved with the 5v patch and is fixed in #6.

I've used the C125862, UZ1086L-33-TN3-R which has nearly the same specs but apparantly not good enough :)

closed 2020-05-28 by sebastius

vroland, 2020-05-28:
I was using the AMS1086CD-3.3, but aside from a slightly lower dropout voltage they seem pretty similar...