vroland/epdiy #4: BOM list

kennethlimcp, 2020-05-26:
Nice work @vroland ! I am trying to fabricate one board and get the dusty `ED060SC4` part working.

Are you able to generate the BOM so that it's easier to have the parts ordered? 

Or if you have a spare one, i wont mind buying it as well! My previous attempt to design a board that drinks the e-ink failed and your board will be useful to help me play around, test the e-ink displays to make sure it's working before making a revision based off your design. :)

vroland, 2020-05-26:
Hmm, yes, I should compile a definitive BOM at some point...
Until then, https://github.com/vroland/epdiy/blob/master/hardware/epaper-breakout/gerbers/bom_jlcpcb.csv is a start, but it is missing the USB connector and the ESP Module (hopefully that's it).

kennethlimcp, 2020-05-26:
Ah I missed that! Should be useful thanks!!

sebastius, 2020-05-27:
Also please update the PDFs, both are outdated :)

vroland, 2020-05-28:
As soon as I get the last parts of my new order, I'll solder a board which I have a complete and accurate BOM for which will mainly be compiled from LCSC and JLCPCB orders.

commit 9184de65 (2020-05-29): add complete BOM (fix  #4)
closed 2020-05-29 by vroland