vroland/epdiy #8: How to calibrate RV1?

sebastius, 2020-05-28:
Messed with RV1 and now i've got a completely black screen :laughing: , any hints on calibrating RV1 and also recover the eink to normal?

vroland, 2020-05-29:
Hmm, aside from the sencond opamp, I'm not sure why the opamp would blow...
But yes, I calibrated by holding a multimeter to the opamp pin ;) 
I could add a test pad for VCOM (and a calibration program which just activates the power supply).  Or do you have a better Idea?

sebastius, 2020-05-29:
The main question is: What should i calibrate it to?

vroland, 2020-05-29:
There is usually a sticker with a negative number (usually -1 to -3, e.g. -1.27) somewhere on the ribbon cable. This is the voltage you should calibrate to ;)

sebastius, 2020-05-29:
> I could add a test pad for VCOM (and a calibration program which just activates the power supply). Or do you have a better Idea?
I can get a nice calibration pad in probably (i need to re-add the mounting holes anyway), if you can make a tiny calibration program, it would be great!

sebastius, 2020-05-29:
Calibrated mine, made the calibration program myself, turned out to be easier than expected. Will write a request with code + documentation tomorrow :)

PR #11 (2020-06-02): Mountingholes, VCOM testpad and 3d models
Fixes #8 and also adds the mountingholes (3x M2) and i've taken the liberty of adding in almost all 3d models.

[image]
closed 2020-06-02 by vroland