vroland/epdiy #12: Use the terminal example as an actual terminal

sebastius, 2020-06-02:
I had great fun already using the Serial Terminal example, and with a bit of foolishness i added a second USB serial to route my PC keyboard from one terminal into the e-ink terminal session (by connecting the second USB-Serial TX line to the RX of the CH330). One problem arose when trying to update the board: the second usb-serial causes the communication to fail. 

I'd like to have a proper input method on the ESP32. Could be as simple as just using a second serial port (they can be routed to any GPIO) and then pushing them towards the regular serial output. That will also allow some control options, like the classical terminals used to have. Alternatively a PS2 port, Bluetooth host or USB host could also work. I don't think we have the GPIOs left for a full keyboard matrix (i think you'd need about 20 GPIOs to do it).

Any thoughts on what you'd like to do in terms of input devices? We could make the expansion header a bit more useful by adding in as much of the GPIOs as we can. 

(besides this hardware/software upgrade the terminal code would need a pretty decent upgrade too, to get to 80columns width, work with a cursor, stuff like VIM would be useful to have. But the first tests were really pleasing!)

Heck at some point i'd love to see the ESP32 do an SSH into my desktop :P (or maybe use a seperate ESP32 for that, so the current chip can focus on being a terminal).

vroland, 2020-06-02:
Yes, a lot of those things are planned for the future ;) First of all, to test how it looks with real terminal input, just run `script -f /dev/ttyUSB0` in a terminal emulator window (of approximately the size of the display, to set $COLUMNS and $ROWS correctly). Additionally, you'll need to open a serial monitor to empty the output buffer of the ESP.
This lets you try all the fun stuff, like:
[image]
I am also working on an AVR-based board to read keyboard and trackpad input from an old ThinkPad keyboard, which I'll write about once I've got all the parts and can assemble  it.
This microntroller can then talk with the ESP and send keyboard input via UART or SPI. 
So basically, once you connect an external microcontroller via the unused and broken out pins, you should be able to use pretty much anything ;)

Running ssh directly on the ESP is a sweet idea as well, maybe I'll come around to trying it at some point...

vroland, 2020-07-04:
If you want to use the terminal, it seems like the ED097TC2 is better suited. At least it seems drawing in white is way less consistent with the ED097OC4 than with the ED097TC2... Also, exact line skipping seems to be trickier.

sebastius, 2020-07-04:
@vroland any hints on where to get the TC2 variant? AliExpress isn't selling them.

vroland, 2020-07-05:
@sebastius Sometimes you just have to ask them ;) This seller for example (https://www.aliexpress.com/item/443254687.html?spm=a2g0s.9042311.0.0.4c004c4dcwgOeF) has them and there is a picture of the TC2 in the reviews section. However, they seem to have raised the pice since my order :O

mmib, 2020-07-06:
Any idea how to connect keyboard to the board and which libs to use?

vroland, 2020-07-06:
I thought of these mini USB host shields: http://www.shop.tkjelectronics.dk/product_info.php?products_id=45 (various derivates are available on AliExpress). There is a library with a HID keyboard example. 
A second option would be a recycled thinkpad keyboard to build a portable terminal. I'll see which parts arrive first ;)
For now, I just connect an external USB UART adapter to the broken out ESP pins and assign them to a spare UART peripheral.

vroland, 2020-07-06:
Furthermore, implementing the Kitty Graphics protocol (https://sw.kovidgoyal.net/kitty/graphics-protocol.html) could be very nice for viewing the occasional PDF document, etc... Well, at least I won't get bored ;)

markbirss, 2021-01-15:
@vroland 

> For now, I just connect an external USB UART adapter to the broken out ESP pins and assign them to a spare UART peripheral.

This part is not clear to me ?

How are you doing this ? a picture might help me

vroland, 2021-01-15:
I don't have a picture right now, but the ESP allows to set up another serial connection and use basically any pins you give it. So you just have to connect two of the pins of the pin header (e.g. gpio 13 and 14) to a USB to UART adapter. That could be one of those cheap programmers for the esp8266.

markbirss, 2021-01-15:
@vroland  

This ability is already in the terminal example on gpio13 and 14?

vroland, 2021-01-15:
The serial pins are defined in https://github.com/vroland/epdiy/blob/9f49923cdbc1ad00aff992338e07fb75a3f68201/examples/terminal/main/main.c#L23. By default, the usb serial is used, but then you won't get debug messages and the ESP writes some messages there during boot.
In a real application, you should change those definitions and comment https://github.com/vroland/epdiy/blob/9f49923cdbc1ad00aff992338e07fb75a3f68201/examples/terminal/main/main.c#L89.

markbirss, 2021-01-15:
Currently I have a Raspberry Pi 3 and the V4 board flashed with terminal example connected to USB port
On my pc I have connection via USB serial to the serial port of the Pi

I have to run
script -f /dev/ttyUSB0

then start a terminal connection and kill it to have a actual interactive terminal

sudo screen /dev/ttyUSB0 115200
sudo killall screen

[image]

vroland, 2021-01-15:
That's one way ;)
You can also use the raspberry pi serial port directly. Just enable the serial console in raspi-config and connect the serial pins to the ESP pins you chose. In my portable terminal, I use the epdiy usb port just for flashing. Power and serial communication comes from the pi through GPIOs. 
Btw. for some applications like fancy TUIs you may need flow control which is a bit tricky to enable with the pi. This article might help: https://ethertubes.com/raspberry-pi-rts-cts-flow-control/

markbirss, 2021-01-15:
@vroland 

Ok, now i understand better

I need to change the code to suit the usge of connecting to Pi Serial console via GPIO

i had it hooked up but it was not configured yet in the code

markbirss, 2021-01-16:
@vroland 

Ok, now the serial console is output onto e-Ppaer

I can attach console via SSH running "script -f /dev/ttyS0"

But how to get a USB keyboard working from the Raspberry Pi ?

or 

a i2c or ps/2 keyboard via the ESP32

Here i made a CardKB USB with
https://github.com/markbirss/CardKeyBoard_PS2

vroland, 2021-01-16:
Personally, I made a board to use thinkpad keyboards and make them output SPI to the ESP: https://git.vroland.de/vroland/epd-laptop/src/branch/master
PS/2 should also be fairly easy, as you mentioned. USB is a bit more tricky, I had the idea of using a USB host board based on the max3421. I have some lying around, but didn't test them yet.

markbirss, 2021-02-04:
Ok, so I have now both working Ok now, thank you for your help @vroland 

V4 EPDIY with a Raspberry Pi 3A local USB keyboard (120x38)  (already posted picture above)

and

LilyGo 4.7 with a Raspberry Pi Zero W local USB keyboard (95x25)

Both the Pi Zero and Pi 3A can be swapped, Pi 4 still have a issue to resolve

[image]

markbirss, 2021-02-06:
With hardware flow control enabled

[image]

astranome, 2021-04-15:
Hi Mark! I recently bought Liligo 4.7 from aliexpress. I want to use this as a terminal for my ESP 32 or Raspberry Pi. I found a great code example on the voland github (thank you, Valentin!)! But I got a very pale image. And instead of black characters, you get gray characters and small ones. I changed .regular = & Firecode, to &Firecoder_Blu, and instead of unsigned int default bg = 7; set the value to 255. This action has improved the display, but the text is still small. In your photo, the text is easy to read, but I don't. I tried replacing the font files .regular = &Firecode, with .regular = &FiraSans12, (I took it from another example). But I get an infinite reload loop after executing listened.....I ask you to share your solution . Maybe you will post your successful terminal launch on github. or send me a hint or a code. I am very grateful for your work, I believe that you are a genius! astra@ya.ru

markbirss, 2021-04-15:
@astranome 

Hi, I can certainly help with the too light print as @vroland helped me with this particular issue also

We added more calls to the command that draws to the e-paper

examples/terminal/main/st.c

Add 4 more to the existing line

Line 3118
draw_mask(area,render_mask,boolean_line_dirtyness);

You can play withe the amount off calls to make the print clearer but your e-paper will get more dirty quicker

astranome, 2021-04-15:
Thank you for taking the time to respond. How do I increase the font size? The config file.h line 21 . I change .regular = & FiraCode, to .regular = &OpenSans12, and add the line #include "opensans12. h" too. But I get an endless cycle of reboots.

vroland, 2021-04-24:
@astranome Do you get a backtrace with the reboots? What do they say? For example, you may need to increase the flash partition size to store the larger fonts and the boot message would tell you that.

astranome, 2021-04-25:
I get boot messages from the device. In contrast to the successful program boot (without infinite restarts), I get the message: 0x400857a8: spi_flash_op_block_func at C:/Users/Admin/Desktop/esp-idf-3/componen
ts/spi_flash/cache_utils.c:105 (discriminator 1) , instead of messages (in normal operation): 0x40084d00: waveform_lut at C:/Users/Admin/Desktop/esp-idf-3/epdiy/src/epd_drive
r/lut. c:241
(inlined by) calculate_lut at C:/Users/Admin/Desktop/esp-idf-3/epdiy/src/epd_dr
iver/lut. c:506
(embedded) feed_display at C:/Users/Admin/Desktop/esp-idf-3/epdiy/src/epd_dri
ver/lut.c:539.
How to increase the flash partition size to store the larger fonts?

astranome, 2021-04-26:
Infinite reboots:
I (940) esp_temperature: Characterized using eFuse Vref

I (2353) term: listening
//AND NOW BEGIN RESTART:

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7468
load:0x40078000,len:14368
load:0x40080400,len:5360
0x40080400: _init at ??:?
etc...............................................

vroland, 2021-04-29:
Hm, the crash seems to occur in the driver code, so it's not the partition size. Which chip revision are you using?
And do the other examples work fine?

entozoon, 2021-12-01:
@astranome I get precisely the same error, did you figure it out in the end?:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
malloc: Success
abort() was called at PC 0x400ee893 on core 1
```

It dies on [this line for me 😭 ](https://github.com/entozoon/psion-terminal/blob/master/src/main.cpp#L121).

Perhaps it's running out of memory? 
```
RAM:   [=         ]   9.0% (used 29600 bytes from 327680 bytes)
Flash: [========  ]  80.0% (used 1048250 bytes from 1310720 bytes)
```
Although I'm not sure what can be removed. It's an Espressif ESP32-WROVER-E with 16mb

vroland, 2021-12-01:
Hm, that could indeed be an out of memory. Maybe try to reduce the columns and lines in the config:
https://github.com/entozoon/terminal-bitch-monitor/blob/master/lib/epdiy-terminal-example/config.h#L401
and see if that helps? Also don't you get a complete backtrace? Or does that not work with platformio?

entozoon, 2021-12-01:
No joy unfortunately. I wonder which parts of the software are so large? But yeah, here's the full output:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
malloc: Success
abort() was called at PC 0x400ee893 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40089e5c:0x3ffb1dc0 0x4008a0cd:0x3ffb1de0 0x400ee893:0x3ffb1e00 0x400d8093:0x3ffb1e20 0x400d3ab5:0x3ffb1e40 0x400d3ad5:0x3ffb1e90 0x400d3e9f:0x3ffb1eb0 0x400d3fa0:0x3ffb1f20 0x400d1506:0x3ffb1f40 0x400d66c2:0x3ffb1fb0 0x4008bd21:0x3ffb1fd0
  #0  0x40089e5c:0x3ffb1dc0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008a0cd:0x3ffb1de0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x400ee893:0x3ffb1e00 in _exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:95
  #3  0x400d8093:0x3ffb1e20 in exit at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/exit.c:70
  #4  0x400d3ab5:0x3ffb1e40 in die at lib/epdiy-terminal-example/st.c:3146
  #5  0x400d3ad5:0x3ffb1e90 in xmalloc at lib/epdiy-terminal-example/st.c:3146
  #6  0x400d3e9f:0x3ffb1eb0 in tresize at lib/epdiy-terminal-example/st.c:3146
  #7  0x400d3fa0:0x3ffb1f20 in tnew at lib/epdiy-terminal-example/st.c:3146
  #8  0x400d1506:0x3ffb1f40 in setup() at src/main.cpp:121
  #9  0x400d66c2:0x3ffb1fb0 in loopTask(void*) at /Users/m/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
  #10 0x4008bd21:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

vroland, 2021-12-04:
Hm, maybe the arduino framework consumes a lot of memory? Did you try using the IDF directly? You can also configure platformio to use the esp-idf instead of the arduino framework, afaik.

scumola, 2022-01-20:
I'm also working with the Lilygo T5 4.7 display w/ the ESP32 controller.  I know that there's ssh libs for the esp32, and I'd love to get something going with the terminal emulation on the ESP32 through the EPDIY library and the ssh library to get a functional SSH client working...  I'm envious of you guys!  Such good work!  :)

scumola, 2022-02-08:
Could I do something like this:
[image]

vroland, 2022-02-09:
Yes, that should work. I actually have something similar running. Btw, for this setup you won't even need to step up the battery voltage to 5V, the Pi 0 actually runs fine with ~3.7V, as long as you don't need USB devices that require 5V.

scumola, 2022-02-09:
Wow @vroland, that's fantastic news!  I'm still waiting for my T5 4.7" display (supply chain delays and all), so I'm just dreaming here, but this is what I'd like to get working.  I want to use your library to power a normal low-power news/email/reddit caching/reading app using the ESP32, with the option to turn on the Pi Zero if I want/need to use linux or ssh somewhere or do some real work.  I plan on having both controllers (ESP32 and Pi Zero) housed in the same enclosure (I'll just 3D print myself a case if it's too bulky), but having the option to have a powered-off *real* computer (the Pi Zero) in the same case to me is a real plus.  Personally, I'd love to use micropython on the ESP32 to power the display, but your library with the fast refreshes and the terminal emulation built-in is just too sweet to turn down.  :)  I expect that you'll have a *lot* more people wanting support with this display since Aliexpress.com is really pushing these displays and touchscreens.  I got mine for $50 (with touchscreen) and I'd like to have more control over the device than the m5paper device.  Thanks for all of the hard work on this library!  I can't wait for my device to arrive so I can dive in!  :)

markbirss, 2022-02-10:
> Yes, that should work. I actually have something similar running. Btw, for this setup you won't even need to step up the battery voltage to 5V, the Pi 0 actually runs fine with ~3.7V, as long as you don't need USB devices that require 5V.

I did this with the LilyGo4.7 and had to set a delay in config.txt for the GPIO 12 (high on boot issue)

extrowerk, 2024-05-01:
> LilyGo 4.7 with a Raspberry Pi Zero W local USB keyboard (95x25)

Do I understand it correctly this Lillygo display is a one-piece solution and it only needs a firmware? Would you please explain me how to accomplish it? I'd like to have a dumb terminal, which lets me interact with my SBC purely over UART.

Thank You very much!

entozoon, 2024-05-01:
@extrowerk Correct. The Lillygo devices have a programmable ESP microcontroller which can be programmed to parse and display the serial data

extrowerk, 2024-05-01:
@entozoon : thanks for the input. LEt me clarify: I'd like to make a dump-terminal for my SBC running openBSD instead of Linux. Eg. I cannot use those displays which gets connected to 40 pin GPIO because they using linux blackmagic, which i can't reproduce on OpenBSD. Therefore I'am looking for a solution which can act as a standalone terminal, and the communication happens via pure UART. On the photo from @markbirss it seems the Lillygo display connected to the UART ports of the Rpi Zero like SBC, but I1d liek to make sure this solution would work before I give out my precious money. 
Could you please confirm, this solution would work in my use case? If yes, what UART speeds are supported and how would I connect a keyboard (maybe a USB or a PS2 one). In case of USB keyboard does the display converts the keyboard data to UART or some kind of Linux blackmagic happens here aswell?

Thank You very much for your help and support!

markbirss, 2024-05-01:
@extrowerk 

It worked at the time i posted with some modifications.

Note that you actually also have to modify your BSD OS getty output options set stty output options and run the screen utility to print the serial console to the lilygo display

extrowerk, 2024-05-01:
I plan to do a headless setup (eg. no display attached over HDMI) with the console directed to the UART pins.  I think the screen command is not necessary in this case, but correct me if I am wrong. Thanks!