vroland/epdiy #14: Status der PCB

holgerlembke, 2020-06-18:
Danke für das großartige Projekt!

Ich plane ein paar epdiy-PCBs machen zu lassen und dann rumzubasteln. 

Wie ist der Status der PCBs unter https://github.com/vroland/epdiy/tree/master/hardware/epaper-breakout/gerbers? Nicht das ich bestelle und drei Tage später eine dolle Änderung gemacht wird...

vroland, 2020-06-18:
Wahrscheinlich sind die nicht mehr ganz aktuell, also vielleicht lieber noch einmal exportieren ;)
Ansonsten enthält die Gerbers_v2_rev0.zip die von mir zuletzt erfolgreich Bestellten, allerdings ohne die ab dort eingearbeiteten Verbesserungen.

holgerlembke, 2020-06-25:
Ja, ein aktueller Export wäre eine schicke Sache.

vroland, 2020-06-25:
Was meinst du mit 09446309680041?

holgerlembke, 2020-06-25:
Gemeint habe ich ED097OC4. Bin zu blöd, die Zwischenablage zu bedienen.

Danke für das Update.

vroland, 2020-06-26:
Falls du das Display meinst: Ebay / Aliexpress. Der Konnektor sollte bei den üblichen Distributoren zu finden sein.

holgerlembke, 2020-06-26:
Danke für die Hinweise. 

Ich bin mit den Einzelteilen halbwegs durch, es hat keine große Eile...

closed 2020-06-26 by vroland