vroland/epdiy #17: Potential issue with GPIO12

sebastius, 2020-06-30:
GPIO12 (MTDI pin) on the ESP32 when pulled high on boot sets another LDO voltage: 

[image]

Which will cause a brownout on the flash-chip in the wrover. We've learned that the hard way with the Hackerhotel 2019 badge project i was part of. 

No idea if the display is capable of pulling the pin high on boot, but it might be interesting to verify. I have a few V2 boards (with my modifications bodged in)  that aren't booting right now, will look into this.

vroland, 2020-07-01:
The more you know... Thanks for sharing! So far I did not encounter the issue with any of my displays, but when I've got access to as scope I can look for anything suspicious.

vroland, 2020-07-01:
If I'm seeing it corrently, GPIO12 is not connected to any display pin. It is broken out as MISO, thus the internal pull-down should do its job...

sebastius, 2020-07-01:
I think i misread the schematic then :) So it's a non-issue, might be useful to update the readme with a caveat.

vroland, 2020-07-03:
Addressed in f4ac37aca000ada40e4d3596885bfed7beea1846.

closed 2020-07-03 by vroland