vroland/epdiy #21: Questions about the components

mcer12, 2020-07-08:
Hi @vroland  I love this and I am eager to make this thing! I plan on making it as compact as possible and also as cheap as possible, so I have some questions. Hope this is the right place to ask those... 
1) in schematic IRLML6402 but it seems large mosfet is used on the pictured board (or is that a different component?). Is that just due to newer schematic revision?
2) Is there a reason why such a large regulators are used? There seems to be quite a few alternatives in SOT-89 like:
https://lcsc.com/product-detail/New-Quadratic-Unclassified-Data_MDD-Microdiode-Electronics-78L15_C414028.html
3) Is ESP-WROVER required for its larger memory or can usual ESP-WROOM-32 be used?
4) I don't see the potentiometer in the schematic, also is it possible to provide resistance values for each display type?
5) Any particular reason why 74HC4094 is used instead of more widely available 74HC595? I see there are some differences between the two.
6) I struggle with the BOM as the components are not marked with designators according to the schematic. Would it be possible for you to update the bom with component numbers?
7) I see AMS1086CD-3.3 is used in BOM and LT1086 in the schematic, both of which have very large quiescent current. I use HT7333 in my ESP project without problems. Maybe separate regulators for the display and the ESP would make sense? It would also be possible to more easilly cut power to the display circuitry that way when the refresh is done.

Please don't take these questions as nitpicking on your awesome project, it's just me being OCD. I want to optimize everything as much as possible so it can be used in compact low power projects.

vroland, 2020-07-08:
That's good to hear ;) Not all things here where conscious decisions, sometimes I just went with what worked for me. But some things I can explain:

1. Yes, I should update that some time... :sweat_smile: 
2. The IRLML6402 is actually quite small (SOT-23) package. These big things are all LDOs.
3. Mainly because I didn't optimize that much for size ;) But another reason is head dissipation and current output (0.1A vs 1.5A): At least according to the datasheet, the display can have large current spikes when powered on. And of course, normally you will disable power as soon as the drawing is done. But if you don't, such a small regulator may not handle generated heat, which can be significant when dropping from 22V to 15V. But you're welcome to try it :)
4. The first version of the PCB actually used the WROOM-Module and the driver itself does not require the additional memory, but for most applications you'll need it. For the 9.7" display, the internal memory cannot even store a full framebuffer, much less two as used in the terminal application.
5. The potentiometer is the variable resistor feeding into the amplifier for the VCOM voltage. Unfortuately, this voltage varies a lot, even for displays of the same type and is quite important for the output quality. So I think a trimmer is really needed here.
6. I used the 4094 because it has an additional output register, which holds the old state while the internal shift register is updated. With the 595 you can only set the outputs to high-impedance, which would deactivate the switching power supply for the display every time I change a value in the register.
7. As in 3. I have concerns about output power, especially if you power the display from 3.3V. This might be alleviated in the newest revision (see #3, #6) as you can also power the display via 5V, but I don't want to be forced to actually have 5V available. But maybe there is a more efficient, package-compatible regulator? Or another way to reconcile those needs?

I hope that answers some of your questions. I'm always open for suggestions and improvements, I'm only starting to learn about electronics myself :)

mcer12, 2020-07-08:
Thanks for the answers!
3. I think power dissipation isn't an issue here since it will be only for very short bursts. The datasheet claims 2W absolute highest peak for the 9.7" display which doesn't seem too bad.
4. that's unfortunate... is it only a flash memory issue (4mb vs 16mb) so 16mb wroom will also work? or is it using the additional SPI PSRAM of the wrover module?
5. I saw an issue here that the trimmer adjusting caused something else to burn, which I would like to avoid :D Any suggestions about that?
7. It might make sense to use low quiescent current for ESP32 and another beefier one with high side switch for the display, that would be cut off when in deep sleep. For low power projects the AMS1086CD-3.3 is a no-no :)

vroland, 2020-07-08:
3. Hm, I according to their data sheet those regulators have a thermal resistance of 160°C/W (at least the negative variant). But I could order those next time I order anyway and see how they do.
4. Flash memory is not a problem, If you only want to display images stored in flash , a WROOM would do. But if you want to compose an image in memory, you need the additional PSRAM.
5. I think that was about the unused second Opamp, which introduced interference. This should be fixed by Sebastius' PR.
7. Yes, that would probably make sense, tough it would use even more components... However, for low-power operation you could just leave out the AMS1086 and USB-to-UART all together, change the jumper to power from 5V (which then could be basically any voltage the LT1945 can handle) and supply 3.3V and 5V via the exposed pins. Then, you could even use a Lithium battery which is directly connected to "5V" and only and feed the output of a low-power LDO into the 3.3V pin.

mcer12, 2020-07-08:
3) It's worth noting that 
a) the peak is just for a fraction of a second, 
b) the power is divided between all the voltage inputs of the screen, there won't be 2W flowing through a single regulator ;) 
and finally
c) The power dissipation is not the same as current draw of the display as the regulator only dissipates the voltage difference between VIN and VOUT which is ~5V and the current is very small. I will definitely give it a shot in my design :)

4) I want to use this as a dashboard that will reguralely download bitmap from a website and display it. I am already working on something but with different display, see my post here: https://www.reddit.com/r/homeassistant/comments/h9yoes/first_glimpse_of_low_power_eink_dashboard/
but the displays are expensive and small-ish. So I want to take on the challenge with this one :)

5) awesome, will check it out

6) I definitely plan to embed battery & charging circuitry as well as CH340 with autoreset as I am lazy to push the buttons :) Want to make all-in-one board since the ESP32 is already embedded, there's not much sense to use external power. 
What I would like to use this for is to stream bitmaps to it from a website.

vroland, 2020-07-08:
1. I see... Yes, it's probably fine. But If you're going to design an all-in-one solution anyway, I'm happy to just see how it goes ;) So I would appreciate if you would keep me updated. Some of those regulators seem to be available locally, so I can try them out some time.

>     2. I want to use this as a dashboard that will regularely download bitmap from a website and display it. I am already working on something but with different display, see my post here: https://www.reddit.com/r/homeassistant/comments/h9yoes/first_glimpse_of_low_power_eink_dashboard/
>        but the displays are expensive and small-ish. So I want to take on the challenge with this one :)

That looks interesting. Maybe it could even be combined with one of those chap tablet touch digitizers to create a full-blown HomeAssistant control panel? Not sure how good these things are at conserving energy though.

>     4. I definitely plan to embed battery & charging circuitry as well as CH340 with autoreset as I am lazy to push the buttons :) Want to make all-in-one board since the ESP32 is already embedded, there's not much sense to use external power.
>        What I would like to use this for is to stream bitmaps to it from a website.

If you're streaming from a website using the WROVER will be a lot easier, as you can just store the image to PSRAM and decode from there. With the WROOM you would probably need to pre-calculate the image data and stream in grayscale layer frames sequentially.

mcer12, 2020-07-08:
1. I will do my best. Since I am ordering everything right now, it will take quite a while... :)
2. There is a version of the kindle display with digitizer. But I am not in a position to understand the datasheet :) I will just slap a few buttons at the side of the display.

I guess I will have to order the wrover module then...

mcer12, 2020-07-08:
ED097OD2 is the version with digitizer. Seems it might be the same thing with the digitizer slapped on.

vroland, 2020-07-08:
Yes, there are E-Ink screen with touch, but they seem to be more expensive and do not come with a touch controller, so I think they are too much of a hassle. What I meant was basically a glass plate with capacitive touch on it, which are available for ~10$ as tablet replacement parts. 
Regarding the display: If you don't yet have one, try to find the ED097TC2, as it has the best contrast and refresh characteristics of the displays I've tested so far. In my case I had to ask a seller on AliExpress (see #12 ), but they sell them for 30-40$, while Waveshare sells them for 150 (They actually removed the model number from the datasheet of their 9.7" display, but it is still present in the file if you grep for ED097TC2 ;) ).

mcer12, 2020-07-08:
I am looking at this one, $10 cheaper :) https://www.aliexpress.com/item/32232199360.html?spm=a2g0o.detail.0.0.60124ccfUvXP18&gps-id=pcDetailCartBuyAlsoBuy&scm=1007.12908.131176.0&scm_id=1007.12908.131176.0&scm-url=1007.12908.131176.0&pvid=7fb96ea8-b8e0-4809-aa46-ff751f4d1e3e&_t=gps-id:pcDetailCartBuyAlsoBuy,scm-url:1007.12908.131176.0,pvid:7fb96ea8-b8e0-4809-aa46-ff751f4d1e3e,tpp_buckets:668%230%23131923%2358_668%23808%235965%23581_668%23888%233325%2319_668%232846%238111%23464_668%232717%237558%23119_

Is this display compatible? Seems like it should be the same.

vroland, 2020-07-08:
Yes, that one should work, it will have a lower contrast though (TC2 has 50% more than OC4, not sure about OC1). And you might have to adjust the Grayscale timings if they are different from the OC4.

mcer12, 2020-07-09:
People seem to be happy with the OC1 display, so maybe the lower contrast is due to some slight voltage / timing difference? I will try to find the datasheet for it, see if there's any difference. Thank you so much for the insights, I am barely touching the surface of the e-ink world :)

simonrus, 2020-07-09:
> I am looking at this one, $10 cheaper :) [...]

I had a problem with display from this seller. The quality was awful (it had glitches and low resolution), I opened a dispute and got 15euro refund

It is an image with artifacts
[](https://ae01.alicdn.com/kf/UTB8Ix2wPiDEXKJk43Oqq6Az3XXaY.jpg)

So, I would advice to take this one
[](https://www.aliexpress.com/item/1888281990.html?spm=a2g0s.9042311.0.0.27424c4dsUVdZI)
It is the original one

mcer12, 2020-07-09:
How's the contrast on the one from your link?

simonrus, 2020-07-09:
I would say, that contrast is the same (It is not graphite??). But characters are sharp and I was able to read books for hours on Kindle DX again. With the glitchy screen, it was a disaster

vroland, 2020-07-09:
Interesting, so there definitely seem to be differences in quality, which may also be the source of @mmib 's problems. I have one screen with a few dead (always white) pixels as well.
Regarding the contrast difference: The OC4 and the TC2 seem to be just different iterations of the technology. Their data sheets specify different contrast ratios (12 vs 16) and they have different timing requirements. The OC4 also had some temporary fading when i hit it with too many partial updates in a row, which the TC2 did not have (at least with the current timings). But that is all just my observations and I have to admit that I simply do most of my testing on the TC2, so some things might just be better optimized for those ;)

mmib, 2020-07-09:
Hi, nice discussion :) I have 2x www.aliexpress.com/item/32840668114.html and it is working now, but with some artefacts. Would like to play with ED097TC2

mcer12, 2020-07-09:
I went ahead and ordered the ED097TC2, it hurt the wallet but I want the best quality and support in case there are any hiccups since @vroland uses it :) I will close this for now and address things as separate issues as I encounter them. Thanks for the patience and insights @vroland

vroland, 2020-07-10:
Well, looks like I should be careful about linking them, as they have already raised the price for the ED097TC2 to 40$ :D On the other hand, if you buy a display anyway the few dollars extra are worth it ihmo. 

And I should probably add some notes from this thread to the Readme, so I think I'll leave it open until then as a reminder :)

mcer12, 2020-07-10:
@vroland since you've reopened the thread, I have some more questions about the schematic :) These are little more specific
1) you use 22uH in the schematic but the datasheet for the LT1945 uses 10uH in the example... any reason for that?
2) there are 39k resistors across L7915 and L7815, what is the purpose of those?
3) POS_CTRL, NEG_CTRL, SMPS_CTRL all have either pullup or pulldown attached... is this necessary? Especially the R1 pullup as the shift register will be either low or high, rendering the pullup useless. Or am I wrong here? 

I will probably work on the schematic during the weekend, will share it for further discussion :)

vroland, 2020-07-10:
Ok, here we go ;)
1. The datasheet has a section on how to calculate the inductor value which yields something around 20uH, but 22uH was the closest available one.
2. That design originates [here](http://spritesmods.com/?art=einkdisplay&page=4), afaik, which has these resistors as well. I'm not sure if they serve a purpose though... And they have oddly specific values, too :D They could be relevant for discharging the display supply current when the regulator is turned of... Someone with a scope should test wether they change anything about the timing ;)
3. Those are left over from the first version without shift register and can probably be removed.

I look forward to what you'll come up with ;)

mcer12, 2020-07-10:
2) only thing I can think of is that it forces some current to flow so that the regulator doesn't shut down... or discharging the capacitance as you say.

That article is awesome, thank you for that. The author says:
_The design has one drawback, and that's that the high-voltage generating parts are pretty inefficient: when I turn on the boost converter, the battery gets loaded with almost 100mA, even without the E-ink display connected._
The 39k resistors certainly don't help with that I guess... I will try to find a way around the 78l15 and 79l15, ideally with some switching supplies.

mcer12, 2020-07-10:
But it might just be the author not being used to ESP high current consumption sinc I assume he uses the battery for everything.

mcer12, 2020-07-11:
@vroland I did some checking with the datasheet of the display. What I found out is this:
- 15V negative rail can draw up to ~70mA
- 15V positive rail can draw up to ~60mA
- -20V and 22V draw ~2.5mA max, so very little
- gate rails should tolerate 21V and -21V respectively

https://www.mouser.com/datasheet/2/149/FAN5331-92198.pdf
This regulator should be able to generate both negative and positive voltages as shown further down at the datasheet with a +-1V swing on the negative rail. But it's limited in terms of output current to 10mA on negative rail so it's only usable on the gate rails. 
Is it possible for you to check how sensitive the gate voltages are? Let's say from +-18V to +-22V, if it has any glitches? If it's ok it should be OK to use this regulator for the +-20V rails and LT1945 for +-15V. This setup should be much more effective. Otherwise I would use two LT1945s to convert all the high voltages from battery.

vroland, 2020-07-11:
As far as I know, these high voltages are the ones needed to drive TFT matrix. The matrix transistors can have large variations from manufacturing and thus high voltages are used to ensure all transistors can be driven. You can probably use (much) lower voltages than specified, but the lower you go the more pixels will not respond adequately any more. For instance the ED097TC2 is specified for +27V, -20V but my models work just fine with +22V,-20V. 
With +-21V + swings you should probably be fine for *most* displays, but you may not be for all of them. So being able to just swap a resistor in case a display doesn't work is comforting for me ;)

mcer12, 2020-07-11:
I wasn't able to find datasheet for ED097TC2, only the other mentioned models, can you provide a link?

vroland, 2020-07-11:
Maybe those rails can be supplied by separate converters?

I use the waveshare datasheet: https://www.welectron.com/mediafiles/datasheets/waveshare/Waveshare-9.7inch-e-Paper-HAT-Datasheet.PDF (If you look at the PDF source, you can see that the actual title of the document is "Microsoft Word - ED097TC2 Ver 1.0-DCC.docx", although they have carefully removed the part number from the visible parts ;) )

mcer12, 2020-07-11:
Btw I tested the negative voltage circuit on MT3608 that I own which is pretty much the same thing as FAN5331 but cheap and widely available. Some tests I did with voltage set to 20V:

- No load: both +20V and -20V are stable at 20V
- Under load: positive is stable, negative goes down with additional load on negative rail. Load on positive rail has pretty much no impact on negative rail. When applying load on negative rail, I got these results.
2mA => -19.5V, 6mA => -18V, 10mA => -17V which sounds good to me since 2mA is absulte maximum in the datasheet.

mcer12, 2020-07-11:
_Maybe those rails can be supplied by separate converters?_
I have difficulty finding inverting converter that's not insanely expensive

mcer12, 2020-07-12:
Schmematic is done though I will have to thoroughly check it. I've added a second LT1945 just in case my "hacky" solution causes issues. When making the schematic I found a dead flag EP_RL which doesn't seem to be connected anywhere... can you confirm it's just supposted to be connected to 3.3V? It's pin 11 on the connector.

**Here it is:**
https://imgur.com/a/3uvm93z

holgerlembke, 2020-07-12:
Might I add a wish: break out the unused ESP32-WROVER-pins to a 2.54 header. 

I want to add some i2c- and serial devices.

mcer12, 2020-07-12:
@holgerlembke there are only 3 GPIOs left (you can use these for soft i2c but I am going to use them for buttons in my project). This may be solved by additional shift register, maybe not, my knowledge is limited here :) @vroland could have more insight with this.

holgerlembke, 2020-07-12:
3 would be ok for i2c (50% safety margin :-). Add a MCP 23017 (or so) gives 16 io-ports, that should be enough for 64 buttons. Fortunately my CO2 sensor already has been i2c-o-fied.

[image]

vroland, 2020-07-12:
@mcer12 RL and UD are specified as "scan direction" in the datasheet. However, for my displays, they either do nothing or keep the display from working when I apply other levels. Maybe they work for some 6" displays, so I did not remove them, yet.
Judging by the schematic, that prototype board will not be simple and small ;) But I look forward to seeing the results.

@holgerlembke The great thing about the ESP32 is that you can route pretty much any peripheral to any pin using the GPIO matrix ;) So with 3 pins, you could do UART, I2S or even SPI. In the worst case, just add a small Atmega and send events to the ESP.

mcer12, 2020-07-12:
@vroland so you have the RL connected to 3.3V? Just making sure. Regarding the board, it should be fairly small, but it will hide under the screen anyway and there's plenty of space :) The buttons will be on a separate pcb.

vroland, 2020-07-12:
Yes, I do. Not sure if this is needed though. 

What exactly are you building, btw?

mcer12, 2020-07-12:
@vroland I will just share **[the reddit post again, see my explanation under the post](https://www.reddit.com/r/homeassistant/comments/h9yoes/first_glimpse_of_low_power_eink_dashboard/)**, I want to do the same thing as I did on the waveshare 7.5" display (which is way easier to work with btw :D ) but larger and hopefuly cheaper. I am waiting for the final pcb now for the waveshare display, I will share photos once it's assembled :)

vroland, 2020-07-13:
Oh, I forgot, sorry. But will be permanently connected to the network and wait for updates or will it sleep / wake up based on a timer?

mcer12, 2020-07-13:
It will wake up every few minutes ;)

mmib, 2020-07-28:
WOW, I noticed you added support for ED133UT2. Where to get it and what is the price (if it is not top secret :) ?

vroland, 2020-07-28:
See https://hackaday.io/project/168193-epdiy-976-e-paper-controller/log/180930-bigger-is-better ;)
It was sold as PENG133D for ~200$. However, it seems to take the controller to its limits, since for full contrast, the refresh is slower than I would expect. So I suspect it draws too high peak currents to deliver through the +-15V regulators. But I will have to confirm with an oscilloscope. 
However, If you do not depend on fast refreshes, it seems to be fine ;)

mcer12, 2020-08-10:
@vroland One by one the components are arriving finally, most notably the gorgeous :) I think replacing the super inefficient 15V regulators with second buck converter will help drop the consumption drastically and allow for these large displays.

holgerlembke, 2020-08-10:
Same here. Out of free time, still missing some parts... Got two ED097TC, building some mounting/holder for development. 

Meanwhile spend some time toying with

[image]

that shall run on li-io battery. currently testing runtime. no idea were it will end.

vroland, 2020-08-11:
@mcer12 I'm curious to see your results ;) I already have a pcb with a bodged MCP170033 which already helps a lot with standby current. So for infrequently updating displays, this is already a big improvement.

@holgerlembke Did you do any measurements of power consumptions with this?

holgerlembke, 2020-08-11:
I don't have equipment to measure power consumption at that scale. I use a "known" li-io cell and will wait until it stops updating (the lower number is a quick hack total runtime display). After recharging the cell I might get a ballpark consumption. I hope.

It is a "TTGO T5 V 2,3 ESP-32 e-ink paper" and as far as I see, the designer didn't had power saving in mind. It had a blue LED in the power supply path... I removed it. But what else might be bad? I didn't research it.

I toyed around with GxEPD and build a partial update with single-digit-update on demand. I did not manage to understand how to get a update region from font sizes, text boxes, cursor positions etc. I had to add some offsets, I didn't find a reason, why... There seems to be no docu, not in .h-file, nowhere...  I might have missed something.

This esp32 RTC has a runtime error of about 4%. So no accurate time keeping possible. I added a DS3231. 32 khz crystals could have been added (GPIO 32/33 is free) but I'm negative biased about their quality (ds1307 are notoriously bad...). It currently does a weekly Wifi-rtc update.

Wakeup from Deep Sleep is done via RTC IO, this consumes more power than RTC CNT, but I have no idea about its dimension in real life.

I will update about the end result....

holgerlembke, 2020-09-11:
662:15:20.

So almost 28 days.

vroland, 2020-09-13:
Well, depending on how often you update that is not bad, but not that good either...

mcer12, 2020-09-14:
@vroland I have another one for you :) Is SMPS_CTRL useful at all? From what I understand it can be ommited completely as the regulators have their own switches (NEG_CTRL, POS_CTRL). LT1945EMS draws <1uA in shutdown according to schematic :) Or did you have some other reson for it?

vroland, 2020-09-14:
Hm, good point. At least right now, I don't remember any reason :D

vroland, 2020-09-29:
@mcer12 Any experience with the FAN5331 to share, yet? ;)

mcer12, 2020-09-29:
@vroland I ditched it (MT3608) in the initial version of the pcb, simply using two LT1945EMS. Even though it should work, it can possibly cause issues and there's enough that can go wrong already. The pcbs should arrive this week so I will report back how it goes.
Regarding the MT3608/SX1308/FAN5331, given everything goes right, I will have to test first how the different gate voltages affect panel performance first. The negative voltage from MT3608 can swing quite a bit +-2V depending on load. Also some panels can be pickier  than others about the gate voltage so it might cause problems on some and not on others. So I became quite sceptical about the idea of using it.

mcer12, 2020-09-29:
@vroland Here is a model of the pcb, the coils are wrong values, gonna be 10uH :) I also already found out I put the LED on the wrong side of the board :D I already have some issues and ideas for the new pcb but let's see if this works first! Hopefully you're able to access the image.
https://imgur.com/a/4v5BrYt

vroland, 2020-09-29:
@mcer12 Ah, that where my concerns as well. And as long as we have access to cheap lt1945's they are fine. 
And it looks like the voltages are adjustable on your PCB with these trimers? This looks quite nice, I'm curious to the the schematic ;)

mcer12, 2020-09-29:
@vroland Schematic here:
https://imgur.com/a/1qGRXaA
take it with a grain of salt, I expect to be changing things as I test the pcb.

vroland, 2020-10-01:
Thanks for the schematic :) When I've got time I'll make another revision with two lt1945 that should hopefully be better at driving the large 13.3" display and consume less power.

mcer12, 2020-10-04:
@vroland The board works! Only the display seems somewhat oversaturated but I might've selected a wrong display in configuration :) Only minor changes will be made to the schematic (different resistors for lt1945 voltage setting). This is awesome! Some sizing changes will be made to the pcb itself as I didn't measure it quite right.

vroland, 2020-10-05:
That's good to hear ;)

mcer12, 2020-10-08:
Btw I've ordered some ED060XC3, it's super cheap and I hope I will make it work. There's no datasheet for it but I found one for ED060XC5 and it seems like a same thing, different flex cable. So hopefully the pinout is the same as well, will report back once I give it a try.

vroland, 2020-10-08:
Interesting... There even seems to be a more expensive version with touch and backlight, so once you have the cheap one working that might be worth a try ;) It's just funny to me how they have subtly different connector types / pinouts for every display, although the interface is always the same :D

mcer12, 2020-10-08:
I would guess it's because they make the displays custom according to customer's (read Amazon) specs. They send them data where they need the connector and what shape to best suit their design. I would guess the number of display types will be very close to the number of big e-reader manufacturers models :)

vroland, 2020-10-08:
I guess that would be the reason. At least most of them are "normal" FPC connectors...

[11 more replies]

closed 2020-10-13 by vroland