vroland/epdiy #22: Arduino IDE examples

mcer12, 2020-07-10:
Would it be possible to add an example for arduino ide? That would make development for this much more accessible for many people, me included :)

vroland, 2020-07-10:
Hm, until now I've just used Arduino as an ESP-IDF component (like the weather example), but yes, the Arduino IDE is more familiar to most people. However, I'd have to think about a good way to do configuration (like the Menuconfig option for display type) in Arduino. Additionally, the IDE uses a pre-compiled ESP-IDF, which has the PSRAM frequency fixed to 40MHz instead of 80MHz, which will make some demos inexplicably slower for these users. 
And perhaps the biggest blocker for me: The IDE is a piece of crap software with no proper build system which is unusable with a wayland window manager because it can't even be resized and menus are showing up randomly all over the place while crashing every time you even look at it in a way it doesn't like :D

So, I won't promise to get to this quickly, but I'll keep it in mind and read up on it :) You may already have some success with just copying the component sources and some `#defines`, if you can get it to compile the code should work pretty much as-is ;)

mcer12, 2020-07-11:
No worries, I will be waiting a month for the components anyway :)

holgerlembke, 2020-07-12:
It shouldn't be to difficult to build your own ESP-IDF.

I have this in my purpur files folder:
```
How to create your own Arduino-ESP32-lib:

Install debian from Microsoft App Store

!!Remember your debian password!!
!!Change shell colors: white background, black text!!
!!Do a supply run to your local deli, it takes a hell of a time!!

sudo apt update           
sudo apt upgrade

follow https://github.com/espressif/esp32-arduino-lib-builder

sudo apt-get install git wget curl libssl-dev libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache
sudo pip install --upgrade pip
git clone https://github.com/espressif/esp32-arduino-lib-builder
cd esp32-arduino-lib-builder
./build.sh

then fix compile errors: https://github.com/espressif/esp32-arduino-lib-builder/issues/12

again:
./build.sh

edit sdkconfig for changes.

again:
./build.sh

out-folder contains sdk.
```

mcer12, 2020-09-02:
@holgerlembke I mean sure I could probably do that but most people just use arduino IDE for the simplicity. Much more people will be able to use this if it's compatible with arduino IDE. I will definitely try to get it up and running there :)

mcer12, 2020-10-04:
@vroland Let me revisit this as I finally have the board working and ready to do software side of things. I understand you don't use arduino ide but for me personally, I have difficulty learning esp-idf and migrating to it for this single project with all the libraries etc I use... could you point me where to start if I want to port this to arduino IDE to make it into an arduino library?

vroland, 2020-10-05:
@mcer12 You can actually use arduino libraries with the ESP-IDF, if you use Arduino as an IDF component. 
But it should also be possible to do the other way around. In fact, the Arduino core for the ESP32 is just a pre-compiled ESP-IDF and you should still have access to the IDF-funktionality directly.  The primary reason not doing this was performance, since this pre-compiled version has a sub-optimal configuration, which makes the PSRAM run at half speed only. But i see that this is not important for every application.
So it should only be a matter of choosing the correct project structure to accomodate both, so that could be the point to start. If you just copy and paste the files from the driver component into your arduino project and adjust defines / include paths, it *should* more or less work. The challenge is to make it work cleanly.

markbirss, 2020-12-18:
@vroland 

not sure if you were aware your code was adapted for the 4.7 inch TTGO LilyGo E-Paper for Arduino

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47

vroland, 2020-12-21:
Thanks for pointing that out here, and yes, I am aware. Unfortunately this fork only supports their board and does not seem to track upstream changes. So it would be nicer to make epdiy arduino compatible, but the main problem I see is the configurability.
If anyone knows how we could set options like display dimensions and board revision at compile time with the arduino IDE, that would be nice.

mcer12, 2020-12-23:
You can use #define to switch between the displays. I would assume it works just the same as menuconfig

vroland, 2020-12-24:
But can I set #define s compiler-wide? Otherwise, the driver would be a different compilation unit than the main sketch, so I can't set defines in the skets and have them included in the driver...

holgerlembke, 2020-12-24:
#defines exist globally after the ~~compiler~~ preprocessor saw them the first time.

vroland, 2020-12-24:
Ok, I don't know about the arduino pipeline, but in most build systems (as in the IDF), the driver would first be built as a static library at which point the preprocessor has never seen the "main.cpp" file / sketch. Then the sketch is compiled and liked with the driver static library. But the arduino IDE might do that differently, I'd have to try that.

holgerlembke, 2020-12-24:
As far as I see in the end it depends how your "build" process works. Do you rely on the "#include" way so the precompiler follows the dependencies? Or do you have some sort of make script that compiles the fragments alone.

Arduino goes the #include way, I think.

vroland, 2020-12-25:
Ah, ok, I didn't know that. I guess trying it out is the easiest way to find out what works.

holgerlembke, 2020-12-25:
Basic process:
* copy together the sketch tabs from left to right (really: alphabetically, main sketch always first)
* add #include <Arduino.h> at line 1
* create some function forwards (it is an imperfect process due to imperfections in detecting what a function is...) and place them before the void setup(); (this will fail if setup is called anything else than exactly this.)
* copy the "core" folder of the microcontroller
* copy (only) the libraries found while scanning the glued sketch

vroland, 2020-12-26:
Good to know. I hope I can try that out until next year. If someone else wants to do a proof of concept, feel free ;)

vroland, 2021-01-06:
Looks like libraries are still compiled as separate compilation units (running arduino-cli in verbose mode will show that), so just `#define`-ing the config won't work.
Thus, I experimented with adding custom board definitions, with *some* success. Adding a custom boards.txt works well and the boards show up in the menu. I can even add a submenu for selecting the display type. Unfortunately, the esp32 arduino core is currently not working very well with derived platforms and assumes wrong paths (see https://github.com/espressif/arduino-esp32/issues/4691). So until that gets fixed, it looks like I need to duplicate their platform.txt and fix the paths in my board definition, which might be a bit of work to keep up to date. However, this file only seems to change on releases, so it should be managable...

mcer12, 2021-01-06:
@vroland take a look here, might be a good approach.
https://github.com/ZinggJM/GxEPD/blob/master/examples/GxEPD_Example/GxEPD_Example.ino

mcer12, 2021-01-06:
But defines should work, other libraries use it like that as well so I don't think you're correct :) 
Something like 
#define CONFIG_EPD_DISPLAY_TYPE_ED097TC2
#include "ed097oc4.h"
#include "config_reg_v2.h"
At this point I think it would make sense to rename some of the library files, especially ed097oc4.h as it is quite misleading

vroland, 2021-01-06:
Hm, that approach looks like a lot of duplicated code for me... And thus quite inconvenient to maintain. They don't define some macro which is then used in the library, but basically include separate libraries based on the display, which then use some common library. But if you look at the code, most of it is implemented in the display-specific libraries (~900 lines of code) and only very little (~90 lines of code) in the common part. 
Copying the platform.txt and running `sed -i "s/{runtime.platform.path}..."` after a esp32 core release seems much easier, plus selecting the board type and display type from the menu seems more beginner-friendly to me. I mean, if you're fiddling with pre-processor macros and you need to understand the order of inclusion in some build tool to understand why your code isn't compiling, you might just as well use the IDF ;)

vroland, 2021-01-06:
But yes, renaming some things while I'm at it would make sense.

stig3, 2021-01-18:
Here is the EPDIY Library using Arduino IDE, just add it to your arduino libaray and restart your arduino IDE. you can use epdiy on Arduino.
[EPDIY_LIB_FOR_ARDUINO_IDE.zip](https://github.com/vroland/epdiy/files/5827308/EPDIY_LIB_FOR_ARDUINO_IDE.zip)

vroland, 2021-01-18:
Thanks, that is probably a point to start. I am currently looking into custom board definitions to make the display type user-selectable without changing the library. And I'll try to make the IDF component to be build-able as the library directly, that way the two won't go out of sync.

vroland, 2021-01-18:
This is now work-in-progress on the `arduino` branch.  With this, you should be able to copy the boards.local.txt to the esp32 core directory (alongside its boards.txt) and just clone the epdiy repo into your Arduino library folder. You should then be able to select a board and display type in the menu.
I'm still working on making the examples compatible and documenting the process, but if someone wants to try and give feedback, go ahead :)

haldi4803, 2021-05-23:
the  "epd_highlevel.h" is not yet ported to arduino IDE do i see that correctly?

Because i saw this example in the master Branch https://github.com/vroland/epdiy/blob/master/examples/lilygo-t5-47-epd-platformio/src/main.cpp and tought that might help my issue with deep sleep.
I've modified the original lilygo file with deepsleep but whenever i quit deepsleep the display just becomes greyed out and doesn't respond anymore (on my T5 47 board)

vroland, 2021-05-25:
Glad you got it working :) I have to admit that the arduino support is still a bit unpolished, but at least it's there. Unfortunately, making it more ergonimic takes time I don't have atm.

haldi4803, 2021-05-25:
on Another note... the Library from lilygo has the epd_poweroff_all(); which also turns off the blue LED.
Found it here: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/dc05af2a794185abf3540b6027b6ab2a13ec63c6/src/ed097oc4.c#L147
I can't find this in your repo anywhere. is that some special addition from them?

martinberlin, 2021-05-25:
If I see this right this is specific of the Lilygo display. I also have this display, but on most things I do send the whole ESP32 to sleep, and that turns also the blue Led for me (I though also about desoldering this one ;)

vroland, 2021-05-29:
Yes, that is specific to them. If needed, we could add that functionality as part of epd_poweroff or the like, but only for the lilygo board type.

martinberlin, 2022-05-14:
As far as I know support for Arduino-esp32 is in place. So we are safe to close this one.
In case we need more examples please make a new issue.

closed 2022-05-14 by martinberlin