vroland/epdiy #23: Question about Parts List / BOM

SimonSelg, 2020-08-27:
Hi @vroland,

thanks for the cool project! Looks very nice :) I would like to build a board too! 

I was wondering which parts list to use:
- [hardware/BOM.csv](../tree/master/hardware/BOM.csv) or
- [hardware/epaper-breakout/partlist.ods](../tree/master/hardware/epaper-breakout/partlist.ods)

I see that in the newest version of the PCB a jumper was added, I guess that one could be added into the parts list too. Or did I get this wrong?

Or should I refer to the schematics and get the required parts from there? 

Thanks a lot!

vroland, 2020-08-29:
Hi,

they should refer to the same parts, but the CSV does not contain specific part numbers. But it won't hurt to double check :)
The jumper is a solder jumper, which you have to solder / scratch, so there is no part for that ;)

Regards, 
Valentin

SimonSelg, 2020-08-29:
Hi, thanks for quick reply. That answers the question, I'll start compiling a  list of parts I need now.

Have a great weekend!

closed 2020-08-29 by SimonSelg