vroland/epdiy #24: SMPS Power Chip Decision Question and Idea

aldwinhermanudin, 2020-09-01:
Hi @vroland! 

First of all, I'm really excited for this project and really one of the thing I've been searching for. This is awesome!
I'm currently, building the newest design based on the current gerbers file on `hardware/epaper-breakout/gerbers` and is also trying to understand the schematics and comparing it to other available e-paper display driver like the [Inkplate6](https://github.com/e-radionicacom/Inkplate-6-hardware/blob/master/Schematics%2C%20Gerber%2C%20BOM/Inkplate6%20Schematics.pdf) and [Waveshare's IT8951](https://www.waveshare.com/w/upload/f/f6/E-Paper-IT8951-Driver-HAT-Schematic.pdf). In the design I saw you use LT1945, L7915 and L7815  to supply the necessary power to the e-paper display and I believe you also mentioned that the design is taken from https://hackaday.io/project/21607-paperback-a-desktop-epaper-monitor.

My question are, 
1) What's the reason behind choosing those instead of using TPS65186, which is designed specifically for e-paper?
2) Is there a plan to use TPS65186? Looking at the [Inkplate6](https://github.com/e-radionicacom/Inkplate-6-hardware/blob/master/Schematics%2C%20Gerber%2C%20BOM/Inkplate6%20Schematics.pdf) is seems pretty similar.

Thanks again!

vroland, 2020-09-01:
Hi,

Yes, the power supply circuitry is based on paperback. With the TPS65186 specifically, it seems to me that it is harder to source and more expensive. But yes, it may be more efficient. 
However, while there are no immediate plans to use the TPS65186, @mcer12 is currently working on another design based on a second, cheaper SMPS (see #21 ). So that may already improve efficiency a lot ;) 
On the other hand, if you need high integration and efficiency, you could probably adapt the design relatively easily ;)

Hope that answers your questions somewhat.

Regards,
Valentin

mcer12, 2020-09-01:
@aldwinhermanudin @vroland Since you guys tagged me here, Ill chime in :)

TPS65186 package-wise is a pain in the ass and also it's not particularly cheap. You can very easily replace the linear regulators with second LT1945. But I will be trying an even cheaper solution for the +-20V rail in my design with MT3608 or SX1308 using external charge pump for the negative voltage. This solution is good for low-ish currents so probably not usable for +-15V rail. I tested this at home and it seems to work. Will have to properly test everything with the display but am still waiting for the LT1945.

aldwinhermanudin, 2020-09-02:
@vroland got it! yes, that definitely answer my question. I'm new to electronics, and basically wanted to understand the design decision of the components. Thanks for the explanation!

@mcer12 yeah VQFN is definitely will be a pain to solder haha (it will be relatively fun tho :D). Nice, MT3608 sounds interesting, been using that for basic usage on my other project. Looking forward on how that goes!

Regarding the LT1945 cost, where did you guys got the chip? I got it from [Mouser](https://www.mouser.com/ProductDetail/Analog-Devices/LT1945EMSPBF?qs=ytflclh7QUVNZJUIwyL8mg%3D%3D) for $5.89 which is more expensive than [TPS65186](https://www.mouser.com/ProductDetail/Texas-Instruments/TPS65186RGZR?qs=KC2rBGieF4rGy3NAz1yOuA%3D%3D) for $4.73.

Am I getting the correct LT1945 chip?

Thanks!

mcer12, 2020-09-02:
@aldwinhermanudin I have the LT1945 ordered from Ali. 8USD for 5 chips I think it was.

mcer12, 2020-09-02:
You might be able to pull it off with two MT3608s sourcing +-20V and +-15V respectively. That's the super economic option as these chips cost pretty much nothing. But +-15V rail is the main power rail and negative voltage using external charge pump is really only good for low-ish currents and drops down with higher current. While the positive output will be rock solid +15V, you could end up with -12V at peaks which is not great. You can probably get around that with a capacitor but the voltage can still swing a bit.

vroland, 2020-09-02:
> @aldwinhermanudin I have the LT1945 ordered from Ali. 8USD for 5 chips I think it was.

Some sellers are even cheaper there and so far I did not encounter problems with them. Plus, you'll probably need to order the displays there anyway ;)

mcer12, 2020-09-02:
@vroland yeah it was 10pcs for $7 not 5pcs :D Just double checked.

aldwinhermanudin, 2020-09-03:
@mcer12 @vroland oh man, can't believe I didn' think of buying the chip from Aliexpress :( there goes my 25 bucks haha.

closed 2020-09-03 by aldwinhermanudin

SimonSelg, 2020-09-03:
@aldwinhermanudin was about to make the same mistake. Thanks for reminding me guys that the chip also is being solved on aliexpress.