vroland/epdiy #27: Kindle touch(d01200) screen SPI interface

ashbuilds, 2020-09-13:
Hi :) 
Sorry, This is not an issue but just a newbie question.

Really happy to see this repo exist, Thank you for your work. I have an old kindle touch that uses an E-Ink screen `ED060SCG(LF)T1`. I wanted to use this as my DIY project and make use out of it, I'm new to this field so my searches are limited and I'm unable to find a way to connect that screen connector to my Arduino uno. It would be great if you can give me the advice to find a proper SPI interface or display driver for this model. Thank you so much. 

[image]

vroland, 2020-09-13:
Could you post a picture of the actual connector?
With these displays, it is unfortunately not as easy as just plugging it in some adapter. This is because they need some special voltages and have some timing requirements that might are not possible to achieve with an Arduino Uno.
But this is basically why this project exists ;) So if the connector for this display works with this board, you can write firmware for the ESP32 that either drives the display directly or talks to an Arduino over SPI.

ashbuilds, 2020-09-14:
Hi :) 
Sounds great. Below are the photos of the connector let me know if not it's clear I'll try to take another. Thanks and have a nice day. 
[image]
[image]
[image]

vroland, 2020-09-14:
Hmm, I have not yet seen a connector like this on any displays I've tried... So if you could get your hands on a data sheet to find out the pinout and connector type, you could modify the PCB to use this connector :/

ashbuilds, 2020-09-14:
Okay great. I'll do some research and get back to you. Thanks :)

ashbuilds, 2020-09-14:
Hi @vroland :)
After some intense research, I found the connector type and its pinout. Its 34 pin connector and below is the screenshot from the datasheet: 
[image]

Link to datasheet: https://wenku.baidu.com/view/793fba7919e8b8f67d1cb94e.html

Let me know if it's possible to modify the PCB for this connector. Thank you for your help :)

vroland, 2020-09-14:
Hey, nicely done! :) A quick search for the axt434124 looks like it's fairly easy to come by and the pin functions are pretty much the same as for the ED097OC4/TC2. So you should be able to create /find a KiCad part for the connector, wire the pins accordingly and you should be good to go.  Once you have the PCB you might have to experiment a bit with the grayscale timing. 
So from what I've seen so far, i am fairly optimistic that this should work :)

mcer12, 2020-09-16:
@ashbuilds @vroland I am more curious about the touch stuff :)

vroland, 2020-09-16:
@mcer12 From what I've seen so far, the touch screen is usually broken out via another connector. Some seem to already have a controller integrated, some have the raw sense lines. So @ashbuilds, is there some other connector as well?

ashbuilds, 2020-09-17:
@mcer12 @vroland Hi :)
In other screens, I saw online they usually do have another connector for touch. I didn't find time to crack open my full kindle but it seems like this screen `ED060SCG` only got one connector as it shows in replacement displays online: http://got.by/21kynj but soon I will do the full teardown of my kindle and keep you updated. 

[image]

mcer12, 2020-09-17:
@ashbuilds  If that's the case then the pinout is incomplete. You will need to give us full datasheet of the display ;)

mcer12, 2020-10-05:
@ashbuilds this should be the same panel, same connector, just differently shaped cable. 
https://datasheetspdf.com/pdf/1098542/PrimeView/ED060SCE/1

vroland, 2021-02-28:
The connector is now implemented in V5 and the ED060SCG works.

closed 2021-02-28 by vroland