vroland/epdiy #28: Demo crash

mcer12, 2020-10-04:
So I managed to compile the demo file but it crashes. This is the output in command prompt... any ideas?

W (993) main: Hello World!

Heap summary for capabilities 0x00000800:
  At 0x3ffae6e0 len 6432 free 0 allocated 6312 min_free 0
    largest_free_block 0 alloc_blocks 22 free_blocks 0 total_blocks 22
  At 0x3ffb2a20 len 185824 free 170644 allocated 15036 min_free 169740
    largest_free_block 169740 alloc_blocks 26 free_blocks 2 total_blocks 28
  At 0x3ffe0440 len 15072 free 15036 allocated 0 min_free 15036
    largest_free_block 15036 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 113804
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x4008c9dc len 79396 free 79360 allocated 0 min_free 79360
    largest_free_block 79360 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 378844 allocated 21348 min_free 377940 largest_free_block 169740
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4194264 allocated 0 min_free 4194264
    largest_free_block 4194264 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 4194264 allocated 0 min_free 4194264 largest_free_block 4194264
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d8ac5  PS      : 0x00060f33  A0      : 0x800d53a0  A1      : 0x3ffb8ea0
0x400d8ac5: rmt_set_tx_intr_en at C:/Users/LGT/Desktop/esp-idf/components/driver/rmt.c:378 (discriminator 2)

A2      : 0x00000003  A3      : 0x00000008  A4      : 0x3ffb13f4  A5      : 0x00000000
A6      : 0x3ffb13f4  A7      : 0xdfffffff  A8      : 0x00000000  A9      : 0x00000004
A10     : 0x00000002  A11     : 0x00060f20  A12     : 0x3ffb654c  A13     : 0x00000001
A14     : 0x000000fe  A15     : 0xffffffbf  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 50fa5ed7bb33b1ef

Backtrace: 0x400d8ac2:0x3ffb8ea0 0x400d539d:0x3ffb8ed0 0x400d4662:0x3ffb8ef0 0x400d4540:0x3ffb8f40 0x400d424b:0x3ffb8f70
0x400d8ac2: rmt_ll_enable_tx_end_interrupt at C:/Users/LGT/Desktop/esp-idf/components/soc/esp32/include/hal/rmt_ll.h:183 (discriminator 2)
 (inlined by) rmt_set_tx_intr_en at C:/Users/LGT/Desktop/esp-idf/components/driver/rmt.c:378 (discriminator 2)

0x400d539d: rmt_pulse_init at C:/xampp/htdocs/epdiy/components/epd_driver/rmt_pulse.c:42

0x400d4662: epd_base_init at C:/xampp/htdocs/epdiy/components/epd_driver/ed097oc4.c:102

0x400d4540: epd_init at C:/xampp/htdocs/epdiy/components/epd_driver/epd_driver.c:865

0x400d424b: epd_task at c:\xampp\htdocs\epdiy\examples\demo\build/../main/main.c:133

Rebooting...

mcer12, 2020-10-04:
All other sketches crash as well :(

vroland, 2020-10-04:
That's with your custom board or with the original one? And which version of the ESP-IDF are you using?
Since this error seems to come from the RMT driver could be either the driver itself or wrong configuration...

mcer12, 2020-10-04:
@vroland My own, doesn't seem to be HW related as blink sketch works. I use the latest ESP-IDF for windows though it comes with cmake 3.13, so I had to change cmake_minimum_required to 3.13 in the CMakeLists.txt. Could this be it? Or what configuration could be wrong?

mcer12, 2020-10-04:
The ESP-IDF is 4.1

vroland, 2020-10-04:
Not sure about the build process on windows, but so far I was on IDF version 4.0 and cmake 3.18.3. Could you try again with an updated cmake and ESP-IDF 4.0?

mcer12, 2020-10-04:
I am just in the process of installing ESP-IDF 4.2 from release branch, seems to have cmake 3.16. I installed cmake 3.18 separately but ESP-IDF refuses it as unsupported and uses its own instead...

mcer12, 2020-10-04:
It works! had to fix some wrong dependencies, probably because windows installer doesn't expect you to install different branch :) Will report back in case of more issues, thanks!

vroland, 2020-10-04:
Ok, I'll make a note to check compatibility with 4.1/4.2 :) 
I'm closing this for now, feel free to re-open if you find more issues ;)

closed 2020-10-04 by vroland