vroland/epdiy #29: Check Compatibility with ESP-IDF 4.1/4.2

vroland, 2020-10-04:
see #28 .

aldwinhermanudin, 2020-10-06:
I've checked on this. It doesn't work for me when using ESP-IDF 4.1

vroland, 2020-10-06:
@aldwinhermanudin Ok, thanks. Then I'll have to look into it at some point.

mcer12, 2020-10-06:
@aldwinhermanudin  For me 4.1 (currently default on windows) didn't work, 4.2 branch works.

iromeo, 2020-11-17:
With ESP-IDF v4.2 + master branch (or revision_3 branch) works ok. With  v4.1 also doesn't work for me, get kernel panic errors on display init.

vroland, 2020-11-18:
There seems to be a driver bug (or change) with the RMT peripheral driver in 4.1. But since it works with 4.2+, a note in the readme might do?

2dom, 2020-12-06:
Add 

 ` rmt_driver_install(row_rmt_config.channel, 0,0);
`
before

`  rmt_set_tx_intr_en(row_rmt_config.channel, true);`

rmt_pulse.c

vroland, 2020-12-07:
Ok, I'll try that, thanks :)

vroland, 2020-12-08:
Unfortunately, when using the RMT driver I can't use interrupts in the way I need to because it installs its own ISR. Since this is fixed in IDF 4.2 which is officially released now, I just added a compile-time error for IDF 4.1.

closed 2020-12-08 by vroland