vroland/epdiy #30: Ghosting issue

mcer12, 2020-10-05:
When I run demo for like 10 minutes and then I run dragon, the text (inverted) stays slightly visible on dragons wing. It needs tens of refreshes before the burned text disappears, so at this point it's not useful for static content like a dashboard. I tried to play with the refresh cycle but to no avail. I think what's missing here is to add an inverted image to the refresh cycle as can be seen on the waveshare displays.

Also, sorry to keep spamming issues!

vroland, 2020-10-06:
Yes, displaying an inverted image can definitely help, if feasible with the application. I did not implement this as part of the driver, because i do not want to require saving and updating a complete framebuffer in every application, even for partial updates. That would be something for a higher-level interface.
But if your application simply displays an image and then clears it without any partial changes, you can simply draw the inverted image yourself ;) You should be able to use the epd_draw_image() function with the appropriate DrawMode.

mcer12, 2020-10-06:
Yea, that's an option as well. In my other project I stream the image directly from server to the display but here since I would have to invert it, I have to buffer it first. I see the images are supposed to be packed 2 pixels per array item, how should I go about this to generate correct image array from a simple bitmap? Or does this simply mean the display has half the resolution than it should have?

vroland, 2020-10-06:
You will have to buffer anyway in order to render it. But with the DrawMode WHITE_ON_WHITE (if i remember corrently) it should already draw the inverse, so you don't have to write another routine for it.
For converting the image, you could take inspiration from this python script: https://github.com/vroland/epdiy/blob/master/scripts/imgconvert.py or the MPD example https://github.com/vroland/epdiy/blob/d9466542310177123d363097e34be751c95e1b69/examples/mpd_status/main/mpd_image.c#L189. 
This packing simply means that in order to save space, each byte contains the 4-bit grayscale data for two pixels, one in the lower and one in the upper nibble. I guess I should elaborate on this in the documentation...

mcer12, 2020-10-06:
Awesome, thank you!

mcer12, 2020-10-06:
It is slower but seems to work reliably for images. No more ghosting. :) If I remove clear cycles completely and use just the inverted image instead, it takes around 1s. Kindle works the same way, it doesn't go through full black / white cycle, it shows inverted old screen -> inverted new screen -> new screen. I think this should be handled on the driver level on any full refresh as it's not easy to do on drawed content and it's very likely not going through inverted image damages the display in the long term.

vroland, 2020-10-08:
Hm, 1s for an image draw seems to be quite slow. Maybe have a look at the config if PSRAM speed and clock speed are set to max ;) But it could also be the wifi driver, since it usually occupies one core. 
I think the best approach would be to add an Arduino-like abstraction layer (maybe a C++ class), that hides most of the details. So if you need the flexibility you can still use the lower-level driver.

mcer12, 2020-10-08:
Oh I am talking about clear cycle, not arduino ide :) What I meant is that the inverted image should be always there to prevent permanent damage to the display, so it should be in the driver itself. ESP-IDF or arduino, doesn't matter. It's slower but it protects the display, that 1s is actually still very good.

vroland, 2020-10-08:
Yes, I know that ;) I was talking about the style of API. Currently, the C-API requires some knowledge of E-Paper displays in order to use it effectively, because I need the flexibility for my projects. But for many users, a simpler API for straightforward use cases on top of the current one would suffice. And this API could be Arduino-like in style, as this is what many hobbyists are familiar with. 
Porting it to the actual Arduino framework would be another step.

closed 2021-09-15 by mcer12