vroland/epdiy #31: Resources and Suggestions

vroland, 2020-10-13:
Feel free to post useful resources and suggestions here. Older comments of this nature can be found in #21 .

mcer12, 2020-10-14:
Found this digitizer, it's not quite the same size as 6" display but might be interesting.
https://www.aliexpress.com/item/32691474609.html

mcer12, 2020-10-14:
Another thing I found is that ED060XD4 (ED060XC3) should be using Cypress TMA445 touch IC (if you order version with touch).
http://www.cypress.com/files/cytma445-truetouch-multi-touch-all-points-touchscreen-controller-datasheetpdf
And linux driver, might come handy:
http://read.pudn.com/downloads648/sourcecode/embedded/2633827/kernel/drivers/input/touchscreen/cyttsp4_device_access.c__.htm

vroland, 2020-10-14:
For touch screen overlays there are bigger ones available as well: https://www.aliexpress.com/item/32892604142.html?spm=a2g0s.9042311.0.0.27424c4dw73ARr These have a Goodix controller chip and afaik there is already an Arduino driver out there.

mcer12, 2020-10-22:
Adding external PSRAM to WROOM module
https://www.reddit.com/r/esp32/comments/j7kxx2/psram_on_wroom/

vroland, 2020-10-23:
Right, I guess since we're already forego specialised E-Ink power supply ICs, why not de-integrate the rest as well ;)

mcer12, 2020-11-02:
There is a version of ED097TC2 with touchscreen and backlight, so just throwing it out there.
https://www.aliexpress.com/item/1005001268418194.html

vroland, 2020-11-04:
> There is a version of ED097TC2 with touchscreen and backlight, so just throwing it out there.
> https://www.aliexpress.com/item/1005001268418194.html

That's a nice one. Btw, do you know any tricks for searching in AliExpress? When i search for ED097TC2 directly nothing turns up, but sometimes it is listed in the suggestions.

mcer12, 2020-11-04:
@vroland I don't... maybe it's your region. Does it let you order the one I linked?

vroland, 2020-11-12:
Yes, I could order it, but I cannot find it via search :D

mcer12, 2020-11-18:
ED0970D2 seems to be the same as ED097TC2/OC4 but also have digitizer. In this case it looks like the digitizer is analog. No backlight.

mcer12, 2020-12-01:
Received new pcbs and should also receive the ED060XC3 finally today, it was on the way for what, over two months? lol! So I will do some testing. I dont have very high expectations since I worked without datasheet but lets see :)

mcer12, 2020-12-01:
Just managed to make it work! Both ED060XC3 and ED060XD4 work out of the box using ED097TC2 definition in menuconfig. ED060XC3 looks just as good as ED097TC2, same contrast, awesome. ED060XD4 has very bad contrast but that can be due to VCOM setting or slightly different timing. Neither of the displays came with VCOM specified so I used 1.62 from my ED097TC2 :)

mcer12, 2020-12-01:
https://imgur.com/a/CPPmDYQ

vroland, 2020-12-02:
Nice work! Just wondering: These displays have a 34-pin connector, right? Would you mind documenting the pinout, so I can make an adapter board from the 33pin connector?
And the VCOM value might be right, when I search for the XD4 I get this site: https://utradioguide.com/products/ed060xd4-c200-touch-screen-display-lcd-fur-pocketbook-626-plus-pocketbook-626-plus-pocketbook-626-p-nicht-zu-verwenden-um-pocketbook-626-2/ . One of the pictures shows a label with a -2.85V VCOM ;)

mcer12, 2020-12-02:
Yes, the connetor and pinout is the same as XC5, just with pin numbers reversed (!!!). I was searching for XC5 images to see what the usual vcom is and its usually between 2.5-2.9 which works great for my XD4. The contrast is now on par with XC3 and even looks little more crisp.

mcer12, 2020-12-02:
This assentially should add support for:
ED060XC3
ED060XD4
ED060XC5
ED060KD1
ED060KC1
ED060XD6
ED060XH2
ED060XC9

I was also doing some digging on ED060SCF but that one is most probably same connector but unknown pinout (from visual inspection of the flex cable, the traces don't match).

vroland, 2020-12-02:
Pins reversed? Wow :D The ED060XH7 looks promising as well, with a specified contrast of 16:1 (vs. 12:1) and touch + front light for ~50$.

> ED060XC3
> ED060XD4
> ED060XC5
> ED060KD1
> ED060KC1
> ED060XD6
> ED060XH2
> ED060XC9
Modulo pin permutation :D

vroland, 2020-12-02:
Have you had any luck getting touch to work, yet?

mcer12, 2020-12-02:
I am a cheap guy so I only have the cheap displays without touch :D

mcer12, 2020-12-02:
Jokes aside, I was curious about it but I barely know linux and porting touch drivers from it to the ESP is something I don't have time for nor do I expect to succeed :D

vroland, 2020-12-02:
Ah, ok. Fair enough ;) Currently I do not really have a project for a 6" panel, so this will have to wait. 
Usually, these are just I2C-connected and you have to read some registers to get events, but writing a nice driver for the ESP is of course a bit more time-consuming.

mcer12, 2020-12-06:
Turns out ED060SCF also works flawlessly with ED060XC5 pinout. I didn't think so at first because the traces looked pretty different but following key traces to the chip, turns out it's actually pin-compatible. That means:
**ED060SCF
ED060SCN
ED060SCP**
should all be compatible using my connector pinout :) 
This covers all kindle paperwhite generations and most kindle 4/5/7s. 

Adding ED060SC7 connector would add support for:
ED060SC7
ED060SCG
ED060SCE
ED060SCM
ED060SCT
This would make the board pretty much compatible with all widely available book reader displays

vroland, 2020-12-06:
So only the ED060XC3 has its pins reversed? But that would make it way easier to drive recycled screens :)
I guess it would make sense then to remove the 39-pin connector from the bottom and provide an adapter board instead, leaving the board with a 33pin and 34 pin connector? Or add it to the top at the cost of a bigger form factor...

mcer12, 2020-12-06:
Sorry I guess I said it unclearly, ED060XC3 has only pin numbers switched on the flex cable but otherwise it's exactly 1:1 like XC5. Another confusing thing is that XC3 (and the same displays) have the flex cable glued to the display, XC5 does not so it may look different when it's not. These images should be self-explanatory :)
XC3:
https://ae01.alicdn.com/kf/HTB1OaJKSFXXXXbTaXXXq6xXFXXX7.jpg?size=87752&height=1000&width=1000&hash=9c1cb17af28fbb6006537e52c23133a7
XC5:
https://rounded.com/images/detailed/123/sony-reader-prs-t3-e-ink-display-ed060xc5-lf_image-1.png

vroland, 2020-12-06:
Ah, ok, interesting. So there are two types of 34pin connector pinouts, the XC5-like (http://www.universaldisplay.asia/wp-content/uploads/2012/10/ED060XC5.pdf) and SC7-like (https://www.waveshare.net/w/upload/7/7c/6inch-e-paper-datasheet.pdf, http://www.universaldisplay.asia/wp-content/uploads/2012/10/ED060SC7-2.0.pdf). Or am I missing one?

mcer12, 2020-12-06:
Yes, it should be just these two. Only annoing thing si that the SC7 connector is not available on lcsc and costs like $2 a piece on mouser :/

vroland, 2020-12-06:
Don't they both have 34 pins with a 0.5mm pitch? This one https://lcsc.com/product-detail/FFC-FPC-Connectors_XKB-Connectivity-X05A20H34G_C528053.html is priced reasonably and looks like it will survive connecting / disconnecting a few times.

mcer12, 2020-12-06:
No, these have the 90degree connector
https://ae01.alicdn.com/kf/HTB1NUpKp.R1BeNjy0Fmq6z0wVXam/6-inch-ED060SC7-LF-C1-E-ink-LCD-For-AMAZON-KINDLE-3-D00901-k3-ebook-reader.jpg

vroland, 2020-12-06:
Oh, those weird mezzanine connectors :/ The AXT434124 seems to be available on aliexpress though. It's still not the most convenient source, but at least cheaper...

vroland, 2020-12-06:
With some luck, something like this https://lcsc.com/product-detail/Mezzanine-Connectors-Board-to-Board_HRS-Hirose-DF40C-40DS-0-4V-51_C424644.html would work. I'll have to compare datasheets to be sure though.

mcer12, 2020-12-06:
probably won't fit because of larger inner circle

vroland, 2020-12-06:
Jep, they only sell even pin counts, like 30,40,50, but not 34 :/

2dom, 2020-12-06:
I just got the board to work. I am reather happy with it - thanks for the great work. One Question though - the contrast and black levels of the display are not very good. Is this limited by the hardware or can this also be influenced / improved by driving patterns etc?

vroland, 2020-12-07:
Did you calibrate the VCOM voltage to your display's value?
Other than that, especially for older displays, the contrast may not be great. Most older ones have their Contrast ratio specified as 10:1 or 12:1, the newer ones (ED097TC2, etc) as 16:1. Since they're reflective this is not really comparable with LCDs, but afaik as long as your VCOM is correct you can't really do much more.
There seems to be some sort of "overdrive" effect if you push e.g. the white pixels very hard, but this returns to the "normal" white levels pretty quickly, even with disconnected power. So this is probably limited by particle diffusion and overdriving the display in this way will probably shorten its lifespan.

2dom, 2020-12-07:
Yeah - I tweaked VCOM for best results. I have the ED097OC4 - is this considered an "old" display?

mcer12, 2020-12-07:
@2dom yes, did you try to switch to ED097OC4 in menuconfig? 
https://epdiy.readthedocs.io/en/latest/getting_started.html#selecting-a-display-type

2dom, 2020-12-07:
Yes .. also the _OLD variety

2dom, 2020-12-07:
Just bought an ED060XC3 (saw your conversion from other issue)... still not sure how to wire this one up though...

mcer12, 2020-12-07:
You will have to either design your own board or wait for the new revision which will include different connecto for this display.

2dom, 2020-12-08:
I think I will wait for now :) A few cool additions to the board would be:

- LiPo connector & charger
- Accelerometer for detecting orientation
- free up more gpios for other stuff

Just a thought :)

vroland, 2020-12-08:
@2dom Yes, the OC4 is one of the displays with 12:1 contrast, so when you look for pictures of the Kindle DXG that should more or less match that.

> * LiPo connector & charger
> * Accelerometer for detecting orientation
> * free up more gpios for other stuff

Yes, more GPIOs would definitely be nice, I'll see what i can do without compromising performance. 
I'm not sure how much of a benefit a lipo charger and accelerometer would be, since you can get modules for that for ~1$ and it would make the board larger and more complex if you don't need it. Also, I'm not sure I have the expertise to design lipo charging circuitry correctly...

2dom, 2020-12-08:
Another finding - running those displays with some gamma corrections really made a difference for me (currently using 1.3). This pushes the lows a little bit and brings out the details

vroland, 2020-12-08:
That is mostly for images, right? But yes, I should add that to the documentation.

vroland, 2020-12-08:
Another (more or less) crazy idea: We could use some SRAM as lookup tables that are programmed for every frame, eliminating the need to do that on the CPU side. This would effectively allow to stream the raw pixel data to the SRAM address bus which outputs to the display bus. 
This way, we could either save four pins (as we only need a 4-bit wide I2S) or get really fancy:
Some of the displays @mcer12 mentioned seem to have flash chips with their waveform data on them. With the waveforms we could then generate a 256 byte look up table for each frames which handles each (old color, new color) combination. With an 8bit wide I2S, we could output 4 bits of the old image and 4 bits of the new image as lookup table address to the SRAM.
That would allow for any grayscale-to-grayscale transitions (provided we can use the vendor waveforms), at pretty good speeds. Or improve refresh speeds with the curent approach. Only synchronizing the horizontal clock signal and data correctly would require some thought.
Hope that sounds somewhat reasonable :D

2dom, 2020-12-08:
Sounds like a solid plan - I like it 😃. Do you have a data sheet for such displays ?

2dom, 2020-12-08:
I'm getting there with the image side of things ... Looking good. The key is dithering - maybe we can push this to the DMA part as well :)

2dom, 2020-12-08:
[image]

2dom, 2020-12-08:
[image]

2dom, 2020-12-08:
Has less contrast in real life though

mcer12, 2020-12-08:
@2dom I am developing my own boards compatible with epdiy with battery charging, 4 buttons and 28uA deep sleep current. One for ED060XC3 and one for the ED097OC4/TC2. I shared an early version (ED060XC3 attached) here few days ago: https://www.reddit.com/r/esp32/comments/k4uxvh/6_kindle_paperwhite_display_with_esp32/
I intend for it to be fully compatible with V5 epdiy pcb so we first need to establish final pin scheme for the V5 before I can finalize the pcb.

2dom, 2020-12-08:
Hey @mcer12 - this looks great. Exactly what I was looking for. :) 

Will you be sharing the gerbers when you are done too?

mcer12, 2020-12-08:
Sure, wouldn't mention it otherwise ;)

2dom, 2020-12-08:
Great! - I was just going to make one or two boards as a Christmas present (epaper photo frame) but this is way too much fun to stop here!

mcer12, 2020-12-08:
@2dom Especially once you realize these 6" displays are same price as 1.54" waveshare displays. I am very tempted to buy bunch of them :D

vroland, 2020-12-09:
@2dom I'm actually up to something similar :D And the dithering is a good idea. I guess you did the dithering on a desktop computer and not on the ESP32? For my picture frame version, I want to read JPEGs off a sd card requiring as little pre-processing as possible.

vroland, 2020-12-09:
Regarding the flash chip: The SC7 (http://www.universaldisplay.asia/wp-content/uploads/2012/10/ED060SC7-2.0.pdf) has it described in its data sheet.

mcer12, 2020-12-09:
@vroland the 6" displays with 34pin connector all have the chip on the flex cable. Some have winbond branded chip, some have other (can't remember the brand now) but the chips look the same so I would assume they behave the same.

mcer12, 2020-12-09:
@vroland you can do the dithering simply with by adding a parameter in your python script when compressing the image. ;)

vroland, 2020-12-09:
Yes, but I for the picture frame I don't want people to have to run the script ;)

2dom, 2020-12-09:
I implemented it on the esp

[113 more replies]