vroland/epdiy #32: Bom quantities for v3

kubasienki, 2020-10-28:
Hi,
I'm trying to build a v3  pcb. In bom, however, there are _cosmic_ quantities entered for the parts. 
For example: 
`General Purpose Amplifiers General Purpose 2 3V ~ 32V, ±1.5V ~ 16V 700kHz 0.3 V/us SOIC-8_150mil RoHS` has quantity of 20. 

As far as I noticed that bom is not for one board and more like for 20? Is that correct? 

And also is bom created for v3 board or just v2 board?

vroland, 2020-10-28:
First things first: v3 is still completely untested, that is why it is on a separate branch. I am currently testing it out however, so I would recommend to wait a few days so I can verify it is working ;)

In the current master branch, there are currently two BOMs: hardware/BOM.csv is a BOM added by a pull requests, where the quantities are just the quantities ordered. In hardware/epaper-breakout/epaper-breakout.csv the parts are listed with a list of their designators, so you should be able to get an exact quantity from there.
Sorry, for this mess, I hope to clean that up when v3 is done. :/

kubasienki, 2020-10-28:
Hi,
I'm aware it's untested yet, it was clearly stated on the main readme. I've already bought 5 boards and screen few weeks ago and they are currently in delivery. That's why I've started looking for the parts. I was happy to test v3 but wanted not to contact you too soon.

No problem about the mess. It's easy to figure out.

closed 2020-10-28 by kubasienki

vroland, 2020-10-28:
Ah, ok. So you where using the current master, not https://github.com/vroland/epdiy/tree/revision_3?
The  current master should work, I think someone already made it :) So if you need any further help, let me know ;)

kubasienki, 2020-10-28:
Yes, so the mess also includes a readme :D It's ok then.

vroland, 2020-10-28:
I guess I should rather call the new branch v4 then ;)