vroland/epdiy #33: Update BOM

hanneslehmann, 2020-11-02:
Hi,

I was trying to upload the gerber files and the BOM to a manufacturer, as I don't have the skills to produce a PCB and to solder. Unfortunately on some parts the manufacturer was not able to determine the LCSC parts. Could you pls help me out, as I don't know which alternative parts can fit/have same pinouts? Thanks a lot!
[image]

vroland, 2020-11-02:
Unfortunately, those parts do not seem to be available for the JLCPCB assembly service :/ The LT1945 and ESP32-Wrover are cheaply available from AliExpress, the others should be available from LCSC.
However, if you don't want so solder yourself you could try to either use a different assembly service (there are some other chinese ones) or find someone to solder it for you. Hackspaces are most probably closed atm, but finding someone else to split the costs might be worth it anyway ;)

hanneslehmann, 2020-11-03:
Ok, thanks for the quick reply. Any ideas - or did you already order the full service (I guess you soldered yourself) with some provider which you know works? I pinged a lot and waiting for reviews...
Thanks for your effort!!

vroland, 2020-11-03:
So far, I just ordered the boards with the parts they had and hand-soldered the rest... Most of them work quite well if you have enough flux and solder wick. But on the other hand, probably not the most beginner-friendly project ;)

hanneslehmann, 2020-11-04:
Indeed - I have already issues with just normal soldering, I guess it would be big trouble with SMT :)

hanneslehmann, 2020-11-04:
Sorry - can you help me in finding the right Part number for "LT1945 Dual Micropower DC/DC Converter with Positive and Negative Outputs, MSOP-10"? Thanks again!

vroland, 2020-11-04:
Just search for "LT1945EMS". LCSC does not currently have them in stock however (and they would be quite expensive, too), so I got mine from AliExpress... But other distributors have them, too.

hanneslehmann, 2020-11-04:
Ok, thanks!

closed 2020-11-04 by hanneslehmann