vroland/epdiy #34: Draw Image API Question

aldwinhermanudin, 2020-11-11:
Hi!

I was tinkering and working on the epdiy board to display images to the e-paper display.
I know that in the API you can use **epd_draw_grayscale_image()** to display image with the assumption that the previous display state is all white(cleared).

Is there a way or an API that I can use, so that I can update the display from an image buffer with the assumption that the previous display is non-white? For example, the current display is showing an image from an image buffer and I wanted to update the display with a new image buffer without clearing the display.

Thanks!

vroland, 2020-11-12:
Unfortunately, in order to provide such an API in general, I'd need the proprietary vendor waveforms.  But for some scenarios there are workarounds: For fonts, you can just calculate a mask of pixels which need to be darkened / lightened and thus have no visible refresh. This is implemented in the terminal example.
I'm not sure how well this would work for images though.

aldwinhermanudin, 2020-12-08:
Sorry for the delayed response and thanks for the reply!
Got it, that's very unfortunate. I'll take a look on the terminal implementation and will see how it works for images.

Will close the issue for now, thanks!

closed 2020-12-08 by aldwinhermanudin