vroland/epdiy #35: Beef up power supply

2dom, 2020-11-14:
Not an issue but a question ... is there a reason why you beefed up the power supply / regulators in the later version of the PCB ?. Cannot believe that the display needs that much power ?

vroland, 2020-11-15:
You mean the double-LT1945 in the latest revision? For most displays, it's about power consumption. Boosting up to 22V, then regulating down wastes energy in battery-powered devices, since the 15V rails actually draw the most current.
Additionally, the displays draw a lot of current during refreshes, at least for a short time. With the 9.7" I haven't noticed any problems with the previous design, but with a 13.3" display I tried the display did not react as fast as it should. With the new revision, this problem went away.

vroland, 2020-11-15:
The reasoning is also that the LT1945 is quite cheap when you source it from AliExpress, but if you don't need the lower sleep current and only use up to 9.7" displays, you will be fine with the previous revision.

2dom, 2020-11-15:
Thanks for the explanation...makes a lot of sense.

closed 2020-11-16 by vroland