vroland/epdiy #36: Differences between epaper-breakout.pdf and bom.csv

kubasienki, 2020-11-16:
Hi, 
bom.csv lists designators which are non existent on the board and also is not coherent with epaper-breakout.pdf.

Here are differences that I've found:
designator | BOM | PDF
------------|-------|------
C6 | 4.7uF | 0.33uF
R1| not listed|10kOhm
C7|0.1uF|0.33uF
C10|4.7uF|1uF
C11|100pF|1uF
C19, C20|1uF|not listed
L3, L4| 22uH| not listed
R8, R9|10kOhm|39kOhm
R19|10kOhm|not listed
R14|390kOhm|400kOhm
R16|300k| not listed
R17|27k| not listed
R18|110kOhm|not listed
U3|LT1945|L7915
U4|not listed| L7815

Which of these sources should be treated as the ground truth?

kubasienki, 2020-11-16:
I believe the bom is the correct one. Also I have ordered a boards from previous gerbers (23/10/2020) which were located here: hardware/epaper-breakout/gerbers . Since then they were upgraded. I'm looking at historical commit (from 1/9/2020) - where can I find the good part list with the designators?

kubasienki, 2020-11-16:
I've used sch from the project. Now the issue is just about false bom.

vroland, 2020-11-16:
Hi,
Sorry, the PDF seems to be really old... I assume the version you used was 1497a12b01c8fc17ebd56f5eb268dcaee88a8aa8?
Which BOM file did you look at? For this point in time, the matching BOM should be https://github.com/vroland/epdiy/blob/1497a12b01c8fc17ebd56f5eb268dcaee88a8aa8/hardware/epaper-breakout/epaper-breakout.csv.

kubasienki, 2020-11-17:
I've finally created my own bom from the kicad file :) 
But it seems like the one you provided looks the same.

Maybe a good idea would be to have revisions separated in folders? I mean v2, v2 new, v3?

closed 2020-11-17 by kubasienki