vroland/epdiy #38: Framerate

amadeok, 2020-11-18:
Hello, 
This looks really good. How many fps can it pull ? Could it be suitable for office work like programming?

vroland, 2020-11-18:
Well, that depends. I have implemented a terminal application with an ED097TC2, which is the fastest of the currently supported displays. You can see a demo video here: https://cdn.hackaday.io/files/1681937195969312/terminal_demo.mp4
I'd say it is usable, but with these partial updates you get some artefacts, so you will have to do a full clear once in a while, which takes about ~1s. 
Also, the screen might be a bit small for regular office work, but I've build a portable terminal with such a display and that works fine for writing things or light programming.

amadeok, 2020-11-18:
The speed in that video is actually quite impressive. I've had  a Dasung e-ink monitor once and it is i think almost the same speed. I'd really like to make a driver myself but i would like to put the board together myself. If i get the list of components in `epaper-breakout-top-pos.csv` , would i be able to put everything together with breadboards?

vroland, 2020-11-18:
The Dasung will probably have better quality though and in contrast to this terminal output pictures as well, just so you know ;)
The list of components is in `BOM.csv`, but you won't have much luck with breadboards, many components are only available in SMD packages and some control signals are fairly high frequency.

amadeok, 2020-11-19:
I see. Say  i wanted to display a text  browser webpage or a GUI app like VS code, would i be able to do it at the same speed of the terminal example?

vroland, 2020-11-19:
GUI applications probably won't really work that well, especially if you want accurate "colors".
For that, I would recommend devices with a dedicated controller like a hacked e-Reader or the Waveshare controllers (and buy the display from aliexpress, since theirs seem way overpriced).
If you want to build this anyway, I had the most luck with assembly services in china. For boards, there is Aisler for example, but they don't offer assembly :/

amadeok, 2020-11-20:
So Aisler is chinese ? Do you know any good chinese assembler? thanks

vroland, 2020-11-20:
Aisler is german, but they don't offer assembly. I used JLCPCB for some boards. However, they only have a subset of the parts, so you would have to solder some by hand.

amadeok, 2020-11-20:
> Aisler is german, but they don't offer assembly. I used JLCPCB for some boards. However, they only have a subset of the parts, so you would have to solder some by hand.

HI,
Yes JLCPCB looks good even if it has some missing parts.  I'm  new to pcb prototypes, when doing the order there are some things to choose, could you tell me if this configuration is ok?:
```
Gerber file - gerbers_v4_Y1
Build Time: 2-3 days
PCB Qty:5
Layers:2
Dimension:100mm*31mm
Different Design:1
Delivery Fomat:Single PCB
PCB Thickness:1.6
Impedance:no
PCB Color:Green
Surface Finish:HASL(with lead)
Copper Weight:1 oz
Gold Fingers:No
Weight:50g
Flying Probe Test:Fully Test
Castellated Holes:no
Remove Order Number:No
4-Wire Kelvin Test:No
Paper between PCBs:No
Confirm Production file:No
```
As far as the LT1945 is concerned, there is a listing in aliexpress for ` LT1945EMS MSOP10 LT1945E MSOP-10 LT1945IMS LT1945 1945 LTTS `, that's the one right?
Thanks. it's ok if the colors are not good, as i only really need this for programming and reading. If i need to display some images i can use my Pixel Qi display or a regular LCD.

vroland, 2020-11-20:
Those settings look good, I'd maybe prefer a lead-free finish. The files for SMT assembly should be in the gerbers/ directory.
In principle, any WROVER-Module will work, but the WROVER-I does not have an integrated antenna. So I use a WROVER-B.
The LT1945 description seems correct.

Another word of warning though: Software-wise you will be largely on your own, since the terminal example is purely text-based. So for GUI-Programs, you will not only have to implement the rendering, but also the interfacing with whatever outputs the image. I don't want to discourage you, just know that this might be a lot of work ;)

amadeok, 2020-11-24:
Hello again, 
I've been taking a look at the source code especially the files inside the epd-driver folder. 
As i mentioned id like to make a sort of pc monitor with your board, and i wanted to ask a couple questions if you don't mind, Petteri who made also a controller board for the ED060SC4 explained how to drive the display on his website http://essentialscrap.com/eink/ , your board works in the same way right? You have the Gate driver and Source driver pins all directly connected to the GPIOs of the esp32 or you using some chip in between them? If i adapt his code by changing the pins to the corresponding ones of your board as well as the timings, could i potentially use his code?
thanks

vroland, 2020-11-25:
Yes, the board works basically in the same way and the wiring seems to be similar. With my board, the data pins are directly connected as well, but some control pins and power configuration pins are moved to a shift register. 
Additionally, my code makes use of the ESPs I2S peripheral which allows way faster data transmission and a higher refresh rate. Also, the display driver calculation is parallelized. So except for if you *really* want to use a WROOM-Module instead of a WROVER, I don't see an advantage with not using this driver. 
For a simple version, you'd probably just transfer a framebuffer via SPI and then use the driver API to display it, so you don't have to touch the internals. But in principle, you can use either.

amadeok, 2020-11-25:
> Yes, the board works basically in the same way and the wiring seems to be similar. With my board, the data pins are directly connected as well, but some control pins and power configuration pins are [...]
Hi thanks for reply.
Yes i'll have to look into it very closely. I was planning to use TCP to transfer the frames as i've been told is the fastest data transfer method the esp32 has?. So far i was able to put together a script that captures 1200x800 portions of the screen and turns them into 1bpp images  in about 20ms (on a i5 3570k), if anyone is  interested i've committed it here:  https://github.com/amadeok/1bpp-screen-capture-for-E-ink
I was about to order the esp32-wrover-b modules and i see there are 3 types: ESP32-WROVER-B, ESP32-WROVER-B(8MB) and ESP32-WROVER-B(16MB), which should i pick?
In BOM.csv, the row number 14 doesn't have a LCSC part number, how can i choose it?
Same for row 30 "TestPoint_Probe test point (alternative probe-style design)" and row 11 "MountingHole Mounting Hole without connection" how do i choose those?
thanks

vroland, 2020-11-25:
The current kicad schematic does, other than that there should be no schematic in the current master...
WiFi is probably one of the faster interfaces of the ESP. However, I would assume that Ethernet, QSPI or parallel i2s are even faster. But I guess before you hit that limit, you'll have to have everything else working pretty smoothly ;)
Regarding the module: I think they have different PSRAM sizes. But I would assume that with your project you shouldn't runt into problems even with the 4MB version, since a 1bpp frame buffer will only be 1/8MB in size anyway (Or you could use a 2bpp representation and off-load some calculations the driver would do on the board to the host).
Test points and mounting holes don't need parts, they are just - well holes. However, you'll need to order the LT1945 externally, e.g. from aliexpress. Maybe read through some other issues, some contain useful information.

amadeok, 2020-11-29:
> The current kicad schematic does, other than that there should be no schematic in the current master...
> WiFi is probably one of the faster interfaces of the ESP. However, I would assume that Ethernet, QSPI or parallel i2s are even faster. But I guess before you hit that limit, you'll have to have everything else working pretty smoothly ;)
> Regarding the module: I think they have different PSRAM sizes. But I would assume that with your project you shouldn't runt into problems even with the 4MB version, since a 1bpp frame buffer will only be 1/8MB in size anyway (Or you could use a 2bpp representation and off-load some calculations the driver would do on the board to the host).
> Test points and mounting holes don't need parts, they are just - well holes. However, you'll need to order the LT1945 externally, e.g. from aliexpress. Maybe read through some other issues, some contain useful information I did not yet aggregate into documentation.

Yes sorry about dummy questions. Is there the possibility to attach a JTAG debugger to the board? I'm seeing that IO15 is used for the LE connection to the display.

vroland, 2020-11-30:
So far, I used the serial output to debug, but I see that JTAG would be a better approach ;) But using IO15, while it should not damage anything, will probably leave you unable to output to the screen, if the JTAG interface is using it. But in many situations you might not need actual screen output? 
Alternatively, you could try if you can change the JTAG pins or bodge the board to use a different pin for LE, freeing up IO15.

martinberlin, 2022-05-14:
There is no more feedback about this since about a year an a half so I guess we can safely close it.

closed 2022-05-14 by martinberlin