vroland/epdiy #40: README for terminal?

BackSlasher, 2020-11-27:
I just got a device to run the demo.
However, I can't get the terminal to run. I assumed that it's the same method of building and running, and then writing stuff to `/dev/ttyUSB0` (e.g. using `script -f /dev/ttyUSB0`), but no joy, reading from `UART_NUM_1` returns with 0.
Should I have built something differently?

I'm using the rev2/3 config

Thanks!

vroland, 2020-11-27:
Hm, that sounds like it should work. Is /dev/ttyUSB0 set up correctly? (115200 baud, no flow control)
You could also try to use different pins and hook up an external usb to serial device, so you still have serial output.

BackSlasher, 2020-11-27:
@vroland I'll try. Could you dump your tty setup if it's no bother? I'll try copying it :D

vroland, 2020-11-27:
I just use some serial monitor (e.g. minicom, idf monitor) to set it up ;) Maybe one thing you could try: uncomment the //esp_log_set_vprintf(log_to_uart); line.

BackSlasher, 2020-11-27:
If I uncomment the line it actually prevents anything from being printed to idf monitor.
Without it, I can still debug-print stuff and catch them on `idf monitor`, and see that `uart_read_bytes` always returns 0.
I see that idf monitor seems to set the TTY correctly (at least for baud)
```
$ stty -F /dev/ttyUSB0
speed 115200 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```
Anything obvious I'm missing here?

vroland, 2020-11-27:
Reading through the code I found that the file in master actually uses pins GPIO_NUM_15, GPIO_NUM_14, which are broken out pins and not the RX / TX pins. Can you check which ones you are using?

BackSlasher, 2020-11-27:
I want to say "uhhhh the USB ones" but I know it's a bad answer :D
How easy is it to use the RX/TX ones, that I assume are already connected to the USB?
I'm not sure which layout the board I have is based on, would be happy to give you any number printed on the PCB if it helps

vroland, 2020-11-27:
Try GPIO_NUM_1 and GPIO_NUM_3 (don't know which one is which, sorry :D), these are the pins labeled TX/ RX on the module and those are the ones the ESP is flashed with. When using those, you will also have to uncomment the line described earlier, because the default serial output will be overridden.

BackSlasher, 2020-11-27:
```
#define ECHO_TEST_TXD  (GPIO_NUM_1)
#define ECHO_TEST_RXD  (GPIO_NUM_3)
#define ECHO_TEST_RTS  (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS  (UART_PIN_NO_CHANGE)
```
I have a suspicion :D 
Will try!

BackSlasher, 2020-11-27:
```
-  ESP_ERROR_CHECK(uart_set_pin(UART_NUM_1, GPIO_NUM_15, GPIO_NUM_14, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
+  ESP_ERROR_CHECK(uart_set_pin(UART_NUM_1, ECHO_TEST_TXD, ECHO_TEST_RXD, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
```
Solves it. Thanks for the help! :D

vroland, 2020-11-27:
Yeah, I should change that :D

BackSlasher, 2020-11-27:
let me PR this plz

BackSlasher, 2020-11-27:
(if you don't mind)

vroland, 2020-11-27:
Sure, I'd appreciate it :)

closed 2020-11-28 by vroland