vroland/epdiy #43: Grainy image

mcer12, 2020-12-03:
While testing the new displays I found out that image on ED060XD4 looks grainy. Have you encountered such behavior on your displays, what do you think might cause this? This only happends on ED060XD4, the XC6 is perfect.

https://www.dropbox.com/s/kylm27qvu3jc300/Photo%2002.12.2020%2010%2030%2003.jpg?dl=0

vroland, 2020-12-04:
I saw something similar with another display of mine, but not to the same degree... Maybe a display quality issue? Or some more susceptible to noise? I don't know.

mcer12, 2020-12-04:
That's what I thought. It's not that bad at all, I was just curious. I thought maybe it needs a little higher voltage like TC2 but no dice. Thanks anyway :) I wouldn't be surprised if these cheap displays were actually discarted garbage from factory :D

closed 2020-12-04 by mcer12