vroland/epdiy #44: Help us improve our products

lewisxhe, 2020-12-04:
Hi , @vroland 

We are making a 4.7-inch ink screen board. It just happens that the driver you write is suitable for our screen. I have changed it to the Arduino method. It is suitable for novices who are not familiar with IDF. Link:

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47

We want you to help us improve this product, do you have any intentions? If you like, please send your address to this mailbox jiangcuifeng@lilygo.cc, thank you

vroland, 2020-12-04:
Hi, that's interesting to see ;) If you really want to make it more beginner-friendly though, maybe wrapper class with a more Arduino-Like API would be appropriate? Furthermore, If you have an official source for your displays, you might be able to obtain better colour accuracy by using the original waveforms instead of my reverse-engineered timings.

> We want you to help us improve this product, do you have any intentions? 

I guess that depends on what kind of improvements you have in mind and how, but we can get in touch via e-mail.

Also, please note that the driver and examples (at least the ones written by me) are licensed under the GPLv3 (https://github.com/vroland/epdiy/blob/master/components/epd_driver/LICENSE), which requires that derivatives must be licensed under the same terms.

lewisxhe, 2020-12-04:
Hi @vroland 

We will contact you via email! And send you the manual!

closed 2020-12-04 by lewisxhe