vroland/epdiy #45: Copyright infringement  'Weather' example

G6EJD, 2020-12-04:
You have blatantly copied my G6EJD weather station software  without recognising me as the author as required by the GitHub licence and my licence, see here:

**This software, the ideas and concepts is Copyright (c) David Bird 2018. All rights to this software are reserved.**
  Any redistribution or reproduction of any part or all of the contents in any form is prohibited other than the following:
  1. You may print or download to a local hard disk extracts for your personal and non-commercial use only.
  2. **You may copy the content to individual third parties for their personal use, but only if you acknowledge the author David Bird as the source of the material.**
  3. You may not, except with my express written permission, distribute or commercially exploit the content.
  4. You may not transmit it or store it in any other website or other form of electronic retrieval system for commercial purposes.
  **The above copyright ('as annotated') notice and this permission notice shall be included in all copies or substantial portions of the Software and where the
  software use is visible to an end-user.**

I have today issues a copyright infringement take-down notice on LilyGo who included your code and now require you to do the same, or I shall raise a GitHub Take-Down notice within 5-days if you have not made the necessary changes.

vroland, 2020-12-04:
Hello, thank you for notifying me. This example was contributed as part of a pull request and I did not check the source of the port thoroughly enough. 
Since this project is non-commercial and clearly references your original project in https://github.com/vroland/epdiy/tree/master/examples/weather, would you allow hosting this port, as long as your orignal license is reproduced and you are clearly mentioned as the original author in its readme?
If not, I will take it down ASAP.

mcer12, 2020-12-04:
@G6EJD This seems like a misunderstanding ;)

G6EJD, 2020-12-04:
Yes please, that's all I ask for. Thank you.

G6EJD, 2020-12-04:
@mcer12 Would you do the same please.
Thank you both for your understanding.

vroland, 2020-12-04:
> Yes please, that's all I ask for. Thank you.

So, take it down or add the license?

G6EJD, 2020-12-04:
Add the licence details please. Or you decide either way.

vroland, 2020-12-04:
Could you check wether the changes made in 86dc98a are sufficient or if I am missing something?

G6EJD, 2020-12-04:
That’s ok, thank you.

closed 2020-12-04 by G6EJD