vroland/epdiy #46: Difference between V2 and V4

mcer12, 2020-12-04:
I can't find much information on this. Just some pin differnences and one additional pin GPIO15 is now used. Can you elaborate on that please? :)

vroland, 2020-12-05:
The main difference is using 2 LT1945 to power the +-15V and +-22V rails. As a result, some pins on the config shift register where changed and the latch enable pin was moved to a GPIO, as it needs to be toggled for every line. By using a GPIO directly instead of the shift register, the refresh is ~60ms faster ;)

mcer12, 2020-12-05:
I managed to import the schematic in easyeda. I see you split power CTRL for each of the high votlage rails but that isn't required (according to the datasheet) and  eats up 2 additional ESP32 pins. +20V and +15V are supposed to be started simultaneously as well as -20V and -15V. I use it like that in my design and it works without issue. Or did you have some other reason for that?

mcer12, 2020-12-05:
Now I see that EP_UD pin is no longer used in V4. So 2 pins could be moved from ESP32 back to the shift register, one pin of the ESP32 freed :)

mcer12, 2020-12-05:
here's my current schematic version, the 3.3V high side switch needs to be tested for current leakage but should be working fine and since it uses SMPS_CTRL pin it saves another pin that would be otherwise needed. I think it would be great if we made something like "V5 (low power)" board that would take advantages from both (the latch pin from V4 and high side switch from my board and saving pins) and also you can copy my pinout for new 6" displays :)
https://www.dropbox.com/s/00wqpnsl07a27ph/Schematic_9.7inch%20e-ink%20controller_2020-12-05_17-17-09.pdf?dl=0

vroland, 2020-12-06:
Jup, that makes sense. I'll open a "V5" project to track this :)

mcer12, 2020-12-06:
I tested the high side switch with and without the additional n-channel mosfet for logic inverting. These are the results:
Using your approach (SMPS_CTRL + single P-channel mosfet) draws additional ~30uA, so the shift registers HIGH state is little leaky. To archieve 28uA deep sleep current, additional n-channel fet (or npn transistor) needs to be added to flip the logic back to active HIGH. So that SMPS_CTRL basically becomes "power_enable" instead of "power_disable" in the code. See my updated schematic:
https://www.dropbox.com/s/8k0oe8dnklqosmj/Schematic_9.7inch%20e-ink%20controller_2020-12-06_20-17-59.pdf?dl=0

Regarding the V5 project, is it possible for you to grant me access to the project task board? Maybe I could help out.

mcer12, 2020-12-10:
From what I've gathered, no remaining pins can be moved back to the shift register because they are all timing sensitive... so I don't see any reason not to use current V4 scheme (even though the CTRL-15V and CTRL+15V can be ommited as I pointed out, but there would just be two unused pins). So only difference between V4 and V5 would be inverting SMPS_CTRL logic :) Or do you have any other plans @vroland ? I ask because I would like us to establish final pin scheme so I can finish my board and have it compatible with the V5 :)

vroland, 2020-12-11:
I'm currently thinking about wether we can double some of the I2S data pins to use as shift register input pins and only have a dedicated STROBE pin, so it only changes state when we want it to. That would save two additional pins. 
Do you know I the display could take damage if a voltage is applied to the data lines with no supply voltage present? Usually these should just be FET gates, but I'm not sure if it is guaranteed to be safe...

mcer12, 2020-12-11:
The display does suck power from other pins (not sure if this includes the data pins) when 3.3V is cut. Didn't damage mine but it's certainly nothing healthy for the display... Since the shift register inputs are well... inputs :) There should be no leakage but that would require some testing. Any tiny leakage will bump the deep sleep current significantly.

vroland, 2020-12-12:
Since the data and clock inputs of the shift register are not inverted, tying them down to 0V in idle should prevent the display from drawing current though?

mcer12, 2020-12-13:
Yes unless the leakage comes from them. But that would be easily measurable
when esp gpios are pulled low so it should be ok.

vroland, 2020-12-24:
> Using your approach (SMPS_CTRL + single P-channel mosfet) draws additional ~30uA, so the shift registers HIGH state is little leaky. To archieve 28uA deep sleep current, additional n-channel fet (or npn transistor) needs to be added to flip the logic back to active HIGH

Just using N-Channel mosfets for switcher and display power would do the job as well, right?

mcer12, 2020-12-24:
You need high side switch so you need p channel. For logic inverting n channel or pano transistor ;)

vroland, 2020-12-25:
But the inverting can simply be done in software, to make it high-active ;) At least with the Mosfet I used previously, I noticed that when powering with 5V it cannot properly close the channel, since even 3.3V-5V = -1.7V where enough to keep the channel open. With an n-channel mosfet, switching booster power would work independent of the booster input voltage.

mcer12, 2020-12-25:
Yes, you can just invert the logic in the code but as I mentioned, it draws more power when the logic is active low (as it is now). At least in my case. Thats the only reason for the logic inverting. 
I don't understand the last part, can you clarify?

vroland, 2020-12-26:
Sorry, I was briefly confused about how N-channel mosfets work in that last part :D
The problem I wanted to describe is that if I'm powering the board from 5V, the p-channel mosfet keeps a low resistance, regardless of the SMPS_CTRL signal, because 3.3V was low enough to open the gate. With a Lipo battery, that problem went away, because the logic-high of 3.3V was high enough.
So that's one more reason for an inverting transistor.

vroland, 2020-12-26:
[image]
@mcer12 Does that look right?

mcer12, 2020-12-26:
The r20 can be omitted, leave only the 3.3V pull-up it will pull up both of the fets. I would use n-channel mosfet instead of Mmbt3904. AO3400 is very cheap and you can use it in other projects since it is logic level and can switch significant current.

vroland, 2020-12-27:
Ok, I just realized with the above setup I'd have a constant current flow from 5V to 3.3V. So I need to separate those:
[image]

Regarding using an n-channel FET: I'm using the mmbt3904 for an auto-reset circuit anyway, so using the AO3400 would be one more part on the BOM...

mcer12, 2020-12-27:
There’s no reason to separate them it will work just fine as I said before
;)

vroland, 2020-12-27:
Yes, but wouldn't I have a constant current flow of (5V-3.3V)/10k from the 5V rail to the 3.3V rail through the pullup resistor?

vroland, 2020-12-27:
And unrelated: Do think it would be useful to be able to switch the BORDER voltage to turn it white or black? If we would add that, I'd probably need an N-Channel mosfet anyway.

mcer12, 2020-12-27:
That’s why you should omit the 5V pull-up ;) the p channel mosfets will be
pulled to 3.3V, no way for 5V to get there

vroland, 2020-12-27:
Ah, sorry, I misread omitting the resistor as connecting it directly to 5V ;)
That makes more sense, but only pulling it up to 5V may still turn Q2, as its gate-source voltage would be -1.7V. But leaving out the 3.3V pull-up and only leaving the 5V pull up should work, right?

mcer12, 2020-12-28:
You're correct, I totally missed that :( Using previous schematic and leaving only the 5V pullup should work, Vgs max is +-12V so it should be just fine. Sorry for the misunderstanding...

mcer12, 2020-12-28:
Regarding the border, how would you use it? I can not imagine any use case other than having it white. The actuall border of the display is white anyway so it would only be distracting imo :)

vroland, 2020-12-29:
No worries :)
Regarding the border: I will probably leave it tied to ground, since the voltage ranges seem to differ for some displays (this might be a mistake in the data sheet, but better be safe).

vroland, 2020-12-29:
I also noticed I have some left over board space, so I might be able to add lipo charging after all...

markbirss, 2021-01-11:
I am waiting for my finished V4 board to see if it works with a 9.7 inch display, will report back once tested

[image]

[image]

markbirss, 2021-01-15:
My V4 PCB arrived today and is working ! (ED097TC2) 

Flashed the demo with your instructions provided 

[image]

[image]

I adjusted the VCOM to -1.78 for my display
[image]

vroland, 2021-01-15:
good to hear :)

markbirss, 2021-01-15:
@vroland 

thank you for your and others contributions, looking forward tio V5

vroland, 2021-02-28:
further discussion has moved to matrix / slack, so this issue is obsolete.

closed 2021-02-28 by vroland