vroland/epdiy #47: Low power optimizations

mcer12, 2020-12-05:
I am working on low power optimizations for my board and I was able to go just below 30uA in deep sleep. Probably more can be done but I'm quite happy with this result.

Changes required:
1. High side switch for EPD 3.3V power rail, should be possible to use SMPS_CTRL pin for it
2. Software optimizations before going to sleep - setting all shift register pins LOW (SMPS_CTRL HIGH as it's inverted) (or drive OE pin low, requires additional ESP32 pin), i2s deinit (issuing PR now)
3. run epd_deinit(); right before going to sleep (added with PR request).
4. (outside epd driver) use rtc_gpio_isolate on GPIO12 and GPIO0 before going to sleep

vroland, 2020-12-05:
That sounds nice! So the 30uA is with the display attached? So isolating that seems to make a big difference.

mcer12, 2020-12-05:
Yes, display attached :)

vroland, 2020-12-06:
I'll add this to the V5 project for tracking :)

vroland, 2021-02-28:
Implemented in V5.

closed 2021-02-28 by vroland