vroland/epdiy #49: I want to make my own font file, but  Error encountered

lixy123, 2020-12-26:
I want to make my own font file, but even with your example practice, the font Successfully generated，but can't run in esp32. I'll revise it according to the guide you gave "fontconvert.py" The exercise generated a custom font file,  The firmware generated by compilation will report an error when running to the reference font statement ：
   , guru mediation error: core 1 panic'ed (storepromoted). Exception was unhandled. 
But the firasans.h in the example runs normally, 
can you give us some suggestions?
   Here are my practice steps:
  1. modify fontconvert.py here:
intervals= [
    # arrows
    (0x2190, 0x21FF),
    # math
    (0x2200, 0x22FF),
    # symbols
    (0x2300, 0x23FF),    
]
  2. run command in Raspberry where can run python:
python3 fontconvert.py msyh 10 FiraCode-Regular.ttf Symbola.ttf  >hz500.h  

    FiraCode-Regular.ttf Symbola.ttf   the same 

[hz500.zip](https://github.com/vroland/epdiy/files/5743199/hz500.zip)

as your example.

  3. esp32  run error：
  
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c46c  PS      : 0x00060130  A0      : 0x800d0e74  A1      : 0x3ffb1f70  
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x0003f480  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00003f48  A8      : 0x00000000  A9      : 0x3ffb1f40  
A10     : 0x4008ed10  A11     : 0x000001e0  A12     : 0x00000000  A13     : 0x3ffd1efc  
A14     : 0x00060120  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00003f47  

Backtrace: 0x4000c46c:0x3ffb1f70 0x400d0e71:0x3ffb1f80 0x400d4617:0x3ffb1fb0 0x40088d2d:0x3ffb1fd0

vroland, 2020-12-26:
It's hard to make anything of the backtrace with only the addresses, but I would guess that the problem occurs because the ASCII characters are missing.
E.g. if a glyph is not found, "?" is used by default, which is not present in that font. Could you try again with the ASCII range included?  (32, 126)

lixy123, 2020-12-27:
oh, you are right! I success, thanks you!

vroland, 2020-12-27:
Hm, that's strange. I'll mark it as a bug and have a look in the next days. 
But the non-alphanumeric characters that are present in the font work?

lixy123, 2020-12-28:
> Hm, that's strange. I'll mark it as a bug and have a look in the next days.
> But the non-alphanumeric characters that are present in the font work?

There are 3500 Chinese character libraries, but because not all of them are Chinese characters, it is inevitable to encounter a library outside the library  see:
[hz3500_16.zip](https://github.com/vroland/epdiy/files/5745813/hz3500_16.zip)

vroland, 2021-01-01:
Could you give me an example string? Right now I have some trouble reproducing this.

lixy123, 2021-01-04:
> Could you give me an example string? Right now I have some trouble reproducing this.

for example
  writeln((GFXfont *)&msyh36, (char *)now_string.c_str(), &cursor_x, &cursor_y, NULL);

1.when now_string be equal to 'abc', which 'abc' exist in the font library, No mistakes
2.when now_string be equal to '嗯',  which '嗯' no exist in the font library, run error.
3.when now_string be equal to 'abc嗯',  which 'abc' exist in the font library and  '嗯' no exist in the font library, No mistakes

vroland, 2021-01-04:
Thanks, looks like the code still wanted to allocate a buffer with negative height. This should be fixed now.

closed 2021-01-04 by vroland

lixy123, 2021-01-05:
> Thanks, looks like the code still wanted to allocate a buffer with negative height. This should be fixed now.

by the way,  function writeln  string wrap is not supported now, whether there are plans to support string wrap  in the future, or provides the function of auxiliary judgment

vroland, 2021-01-05:
You mean breaking the string into multiple lines based on some line width?
I'm not sure if the display library is the right place for that, since the rules for breaking strings into lines vary from language to language and can be quite complex.

lixy123, 2021-01-06:
> You mean breaking the string into multiple lines based on some line width?
> I'm not sure if the display library is the right place for that, since the rules for breaking strings into lines vary from language to language and can be quite complex.

That's what I asked, You can do the simplest case first, for example, like office word software, when a line of text cannot be displayed in the line, you can wrap it normally

commit 8b42085c (2021-01-04): font: fix crash in write_mode with only non-present characters (fix #49)
commit fef8982c (2021-01-05): font: fix crash in write_mode with only non-present characters (fix #49)