vroland/epdiy #51: How to use fontconvert.py

LilyGO, 2021-01-05:
[image]
I observed the code and found that three parameters are required.
They are name, size, and fontstack.

But I cannot execute it correctly. Can you give some pointers?

vroland, 2021-01-05:
Looks like you are missing the full path to the TTF. For a reference on how to use the script, you can also look at the Makefile in the MPD example: https://github.com/vroland/epdiy/blob/master/examples/mpd_status/Makefile

LilyGO, 2021-01-05:
Thanks for the reply.
TTF is in the **current** path. So I didn't add the absolute path.
I read the Makefile, and the content is to convert **3 fonts** at once. 
Do you have an easier way to operate Powershell or **Bash shell**?
E.g: python3 fontconvert.py [Parameter 1] [Parameter 2] [Parameter 3]

vroland, 2021-01-05:
When running fontconvert.py with `-h`, it prints the following:
`fontconvert.py [-h] [--compress] name size fontstack [fontstack ...]`

Which means the first parameter is a name, then a size. Then, multiple fontstack arguments can follow. You can specify multiple fonts to use others as a fallback font. In the MPD example, some symbols are not present in Fira Sans. When that is the case, the script tries to find it in Symbola (sencond specified font). If it is not there, it looks in Fira Code. If it is not present in any, the script will fail. So only the first font will be "converted", but the others are used as fallback fonts. Does that answer your question?
But you are definitely right in that this should further be explained in the documentation.

lixy123, 2021-01-06:
> Thanks for the reply.
> TTF is in the **current** path. So I didn't add the absolute path.
> I read the Makefile, and the content is to convert **3 fonts** at once.
> Do you have an easier way to operate Powershell or **Bash shell**?
> E.g: python3 fontconvert.py [Parameter 1] [Parameter 2] [Parameter 3]

I have been following the ink screen development library of epdiy. You can ask me how to select a batch of Chinese characters and make the source code file for the development of the library. According to the steps described in it, You will be able to make the right content for certain languages of the font file you need
step is here:
https://github.com/lixy123/LilyGo-EPD47-ShowNowDay/tree/main/%E6%B1%89%E5%AD%97%E5%BA%93%E5%88%B6%E4%BD%9C

G6EJD, 2021-01-29:
How do we create extended fonts to include special characters like °

I see you have them included in your examples, but I cannot replicate that for a new version say a new point-size?

I'm able to create the fonts using your fontconvert.py without issues, just the missing/extended characters.

vroland, 2021-01-29:
@G6EJD You need to add the required code point intervals in the script over here: https://github.com/vroland/epdiy/blob/322bcea64f909f1cd7a7df514cc89ba505ca96e1/scripts/fontconvert.py#L26
Sorry, looks like I forgot to mention that in the documentation.

martinberlin, 2022-05-14:
@iromeo what is the status of this?
Can you make a pull request or should we close it?
It has been quite a long time since the last feedback.

G6EJD, 2022-05-14:
It works as is

martinberlin, 2022-05-14:
Ok so unless I'm missing something we can safely close this. Documentation is linked from [the Readme](https://epdiy.readthedocs.io/en/latest/filegen.html)

Please reopen if there still things to do in this point.

closed 2022-05-14 by martinberlin