vroland/epdiy #52: Noisy / grainy image with ED097TC2

2dom, 2021-01-06:
I just received my ED097TC2 panel to replace the broken ED097OC4. While it appears to be of better contrast it is also a lot more "grainy" (see right side of attached picture). Any suggestions what I can tweak to get rid of this?

Fun fact - if I place my hand on the display during update (placed it o the left side) the grain appears to be much reduced. 

[image]

vroland, 2021-01-06:
You're right, when comparing the displays directly the TC2 seems noisier (though maybe not as much?) for me as well. I don't seem to have your magical hands though :/ Can you localize where your hand has an effect? I don't have an oscilloscope myself, but if I know what to look for I may be able to borrow one.
Is it the same if you first darken the display and then draw the image in white (The grayscale test example does both)?

2dom, 2021-01-06:
Will try white on black. I presume that the hand has some electrostatic effect. Maybe changes the off-set or something. I noticed this before - on another display I had put sum electrical tape on the back and it was clearly visible in the image.

vroland, 2021-01-06:
Hm, in my e-paper laptop I have the whole controller board taped to the back of the display with electrical tape but it's not visible in any way. Maybe because it is shielded? 
Which revision of the board are you using, btw?

mcer12, 2021-01-06:
@2dom just to be sure, did you adjust the voltages? According to datasheet it requires +27V (instead of usual +22V). And did you adjust VCOM according to the sticker?

2dom, 2021-01-06:
I did adapt vcom but not to 27V ... Will have to give this a go ☺️

2dom, 2021-01-06:
Thanks for the reminder - did not actually read the datasheet

2dom, 2021-01-06:
@vroland Checking the Revision I have to postpone until I take the picture frame apart next time

2dom, 2021-01-11:
@mcer12 ... Can you send me the datasheet please?

mcer12, 2021-01-12:
@2dom https://www.welectron.com/mediafiles/datasheets/waveshare/Waveshare-9.7inch-e-Paper-HAT-Datasheet.PDF

2dom, 2021-01-13:
Great - thanks!

vroland, 2021-01-14:
@2dom Did you already check if the voltage has any effect?

2dom, 2021-01-19:
No difference unfortunately

mcer12, 2021-01-19:
@2dom what version of the board are you using?

2dom, 2021-01-20:
Version 2

mcer12, 2021-01-20:
@2dom that board uses linear regulators for +-15V rails as far as I know. And I also think the booster is driven from 3.3V which si pretty bad so maybe it's having trouble driving the large display. The V4 uses powerful booster for +-15V. @vroland can probably elaborate on that.

vroland, 2021-01-21:
Could be a reason, But I noticed one of my TC2s being grainy with the newer versions as well. 
@2dom Does it make a difference if you make the display black first and then draw in white vs directly drawing in black on a white display?

2dom, 2021-01-21:
I tried both...pretty much the same effect

vroland, 2021-01-23:
Hm, not sure what else to try. Maybe in the long therm we could reverse engineer the original waveform files. Unfortunately, I can't really find any information about the file format.

mcer12, 2021-10-06:
@2dom @vroland I bought a second ED097TC2 and the new one is grainy, the old one isn't. So it seems like it's display-speific or some displays are more susceptible than others. I will investigate a bit further on this but I can't think of anything particular that could cause this on one display and not the other. Maybe it's just cheap low quality panels that were stolen from factory garbage, who knows.