vroland/epdiy #53: Compatible displays list

mcer12, 2021-01-06:
I made this a while ago and kinda forgot about it :) Please take a look, maybe it will be useful.
https://www.dropbox.com/s/8g3380bcyct6b7w/displays.xlsx?dl=0

vroland, 2021-01-06:
Neat :) I think this should be converted into a documentation page on occasion.

2dom, 2021-01-09:
Maybe augment with column for driving voltages? Could be helpful to some 2doms

markbirss, 2021-01-11:
since support was added for the LILYGO T5-4.7 inch epaper board the list can include it?

https://github.com/vroland/epdiy/commit/8c6545245c4296890f1bbfdb8e63cfe0e040339f

mcer12, 2021-01-12:
@markbirss the list is specifically for the displays. Maybe separate list for compatible boards would make sense but I think this is up to the boards author to link to this library.

mcer12, 2021-01-12:
@2dom From the datasheets I've seen (and it was quite a few) the only exception is the ED097TC2, other displays all run at +22V, -20V and +-15V. The TC2 also seems to run just fine (at least for me) at +22V ;)

mcer12, 2021-01-12:
@vroland how did you make the dable of display in readme.md? Did you use some tool to convert the table from excel github format? I would like to update the display list in readme but I am not willing to do it by hand :D

markbirss, 2021-01-13:
> @vroland how did you make the dable of display in readme.md? Did you use some tool to convert the table from excel github format? I would like to update the display list in readme but I am not willing to do it by hand :D

Have a look at 
https://docs.github.com/en/free-pro-team@latest/github/writing-on-github/organizing-information-with-tables

markbirss, 2021-01-13:
> @markbirss the list is specifically for the displays. Maybe separate list for compatible boards would make sense but I think this is up to the boards author to link to this library. If you happen to know the serial numer of that 4.7" display, let me know.

Ok, why not just put it in the "compatible since pcb version" column

with the M5Paper it is the ED047TC1 e-paper pictured below

Model Number：EPD_ED047TC1 | 540*960@4.7" | Grayscale : 16 Levels | Display area : 58.32*103.68mm | Display Driver : IT8951

[image]

mcer12, 2021-01-13:
Sure, makes sense

mcer12, 2021-01-13:
@markbirss the M5Paper looks much better than the Lilygo board, very nice thing. Definitely would love to get my hands on the schematic :)

markbirss, 2021-01-13:
@mcer12 

I hope to get some soon.

They are driving the display with the IT8951E 

https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/schematic/Core/m5paper/M5_PAPER_SCH.pdf

mcer12, 2021-01-13:
Yea I just found it as well :) Very nice. They are using quite expensive parts, neither IT8951E or TPS65185 are particularly cheap... do you know what it draws in sleep?

markbirss, 2021-01-13:
@mcer12 

Im not sure, i see some discussion here

https://community.m5stack.com/topic/2581/m5paper-epd-power-consumption/9

mcer12, 2021-01-13:
Seems pretty bad, I was suspicious why they don't advertise that :D Kinda defeats the purpose.

vroland, 2021-01-13:
> @vroland how did you make the dable of display in readme.md? Did you use some tool to convert the table from excel github format? I would like to update the display list in readme but I am not willing to do it by hand :D

I'd export it as csv and then pipe it through `sed`, that should be enough. I just have to get around to writing some documentation :)

vroland, 2021-01-13:
> @vroland definitely take a look at it, they managed to implement the touch screen which I imagine will be the same as with the larger displays.

From what I've seen in the pictures, there are two kinds of displays with tough: Some have a touch controller which usually break out an I2C bus. This is what the m5stack seems to use as well and many ebook readers two. These should be relatively straight forward and pretty much the same as those cheap tablet touch screens. 
The other kind exposes the capacitive sensing lines directly with no controller. I wouldn't bother with them.
I'm occasionally looking for cheap broken kindles with a touch display to try them, but on aliexpress displays with touch are a lot more expensive than without...

mcer12, 2021-01-13:
@vroland What about ED060XH3?
https://www.aliexpress.com/item/32980103286.html?spm=a2g0o.productlist.0.0.48d266f6LOw8jS&algo_pvid=1a602e4d-553b-4188-8ec0-46beb7126fab&algo_expid=1a602e4d-553b-4188-8ec0-46beb7126fab-0&btsid=2100bb4c16105470952453145e4ed0&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
$30 and I like that it already has the black frame attached,

mcer12, 2021-01-13:
Also found this one:
ED060SD1
https://www.aliexpress.com/item/32968701660.html?spm=a2g0o.productlist.0.0.365f4c5aldUqWN&algo_pvid=e3305cd7-b209-4244-bccf-4a62984326d0&algo_expid=e3305cd7-b209-4244-bccf-4a62984326d0-0&btsid=2100bb4916105473759052231e0fd4&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

martinberlin, 2021-01-13:
@vroland awesome work this component. Big fan of IDF so with this I will learn how to drive paralell displays
Please add support for this new 4.7 epaper from Lilygo ed097oc4:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47

Resolution: 
EPD_WIDTH 960
EPD_HEIGHT 540

I've one in my work so I can help testing. I built one [epaper component for ESP-IDF](https://github.com/martinberlin/cale-idf/) but is so far to drive only SPI einks.

vroland, 2021-01-13:
@martinberlin I've added support for the LilyGo board in 8c6545245c4296890f1bbfdb8e63cfe0e040339f, so feel free to test it out :)

vroland, 2021-01-13:
> @vroland What about ED060XH3? (will update supported displays list as I just found it now)
> https://www.aliexpress.com/item/32980103286.html?spm=a2g0o.productlist.0.0.48d266f6LOw8jS&algo_pvid=1a602e4d-553b-4188-8ec0-46beb7126fab&algo_expid=1a602e4d-553b-4188-8ec0-46beb7126fab-0&btsid=2100bb4c16105470952453145e4ed0&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
> $30 and I like that it already has the black frame attached,

That one looks interesting, 30$ for touch and backlight is a good price... Did you find a data sheet anywhere? Looks like it uses the XC5 connector, but I'm only guessing from images.

mcer12, 2021-01-13:
I'm quite certain it will be the same thing. Since XC3 works (which has different looking connector) this will work as well ;)

martinberlin, 2021-01-14:
> @martinberlin I've added support for the LilyGo board in [8c65452](https://github.com/vroland/epdiy/commit/8c6545245c4296890f1bbfdb8e63cfe0e040339f), so feel free to test it out :)

Awesome. Tried two demos and worked out of the box:
demo
grayscale_test -> Worked great but leaves a ghost of last print (see [video](https://twitter.com/martinfasani/status/1349674742169690112))

I really like it. I would like to research how it works and based on your work, but with my C++ approach, to add this epaper also as the first model in my component Cale-idf. No idea if I have the knowledge to do it, but will give it a try. And of course I will use your library for other projects. 
What I will really want to try in this epaper is the touch screen. It has all prepared for touch, including a demo code from LILYGO, with the detail that the hardware is not there (At least I don't see a touch overlay)

mcer12, 2021-01-14:
@martinberlin I already reported this. By repeatedly displaying the same image, it will get burned in and takes several refreshes to get rid of the ghost. What you can do to mitigate this is to add another frame with inverted image before every change.
Instead of image1 => clear => image2
should be inverted image 1 => image 1 => clear) => inverted image 2 => image 2
You can see what kindle does when it refreshges. I think this should in some way be implemented by the library and we discussed it... but I don't remember what the conclusion was @vroland

vroland, 2021-01-14:
@mcer12 @martinberlin Currently you can draw the inverse image by setting the draw_mode parameter of the image drawing functions: https://epdiy.readthedocs.io/en/latest/api.html#c.DrawMode
My idea was to provide a high-level API that automatically does this before every draw, maybe as part of adding Arduino IDE-compatibility. However, for some applications I want detailed control, this is why the current API does not do this automatically.
But it could be highlighted in the documentation as a best practice.

martinberlin, 2021-01-16:
Quick question @vroland 
is there any chance that someday this nice PCB you designed will be available to buy in Tindie or to buy one already made from you directly?

My father has a broken Epaper and would like to try it out. We do electronic projects together so for sure he will find his way around

vroland, 2021-01-16:
@martinberlin Commercially selling hardware in Germany is unfortunately quite expensive, so unless I find a good way to do that not from me. I might getting rid of leftover prototypes on eBay though ;)

stig3, 2021-01-18:
I have tested ED043WC3,ED047TC2,ED038TH1  and it also works fine. And only ED047TC2 have to change the resolution to 960x540, the other 2 don't need to change anything.

mcer12, 2021-01-18:
@stig3 where did you source these displays?

stig3, 2021-01-19:
@mcer12 i got few from Eink agent.

vroland, 2021-01-21:
@stig3 Do you know if there is a source for a 13.3" display for a reasonable price?

stig3, 2021-01-22:
@vroland sorry, the largest epd i've seen is 9.7".

martinberlin, 2021-01-27:
>I might getting rid of leftover prototypes on eBay though ;)

That sounds awesome. My father has some dead Kindles and wanted to try it out. So if you ever sell them just send me an email to martin @ luckycloud.de and I will buy it from you directly. 
I added the TTGO T5S to my epaper component using your C code as a base. Documented this here:
https://github.com/martinberlin/cale-idf/wiki/Parallel-epapers-driven-by-Epdiy-with-I2S-Data-bus

Still testing but so far works great, this is how the C++ class looks like:
https://github.com/martinberlin/cale-idf/tree/refactor/oop/components/CalEPD/models/parallel

That way the users of my component can still use this parallel display same way as the others. I had only one little conflict, since Adafruit GFX that I use for the fonts / Graphics, uses the same structs as your class: GFXFont & GFXGlyph so I had to make a copy and rename them to GFXFontDiy & GFXGlyphDiy but that is not optimal. 
I'm really happy I found this component. And I'm also following your project in Hackaday. It's really very interesting the way you sent I2S like this to the display.
I tried to use this with an [Inkplate 6](https://github.com/e-radionicacom/Inkplate-Arduino-library) since I like this library more than the original one but I could not make it because they put a I2S bus expander in the middle for the CFG lines. This is what the library owner told me

>EPDiy uses I2S Parallel interface to drive epaper, while we are using classic bit-banging method. In order to get fastest refresh rate from our code, we have to optimize it a little bit. That means that we have to set data and clock it in as less instructions as possible. In order to do that, in one line of code, we clear data lines and clock line and in the other we set data and pull clock high. Because you can't set data and clock it at the same time, we have to slow clock down on hardware level and for that we used capacitor from GPIO0 (Epaper CL) to GND (capacitor C44 of 100pF). If you are using EPDiy, that can cause problems. In that case, you should remove it. But, if you ever wanted to use Inkplate library again, you have to put it back.

But I'm really eager to try it out and make EPDiy work in the Inkplates too (I need to check first this hardware thing with the 100pF capacitor though)

vroland, 2021-01-28:
Hey @martinberlin, 
Nice work! The C++ wrapper looks quite nice as well. I was thinking about going a similar route of wrapping the component for Arduino IDE support.  Since you seem to have some experience with embedded C++, maybe we could chat a bit about that?

> Adafruit GFX that I use for the fonts / Graphics, uses the same structs as your class: GFXFont & GFXGlyph so I had to make a copy and rename them to GFXFontDiy & GFXGlyphDiy but that is not optimal.

Yes, I'm not happy with this as well. Originally, I planned to be compatible to the GFX fonts, but with large displays and the resources the esp32 offers I really wanted proper support for more unicode glyphs, anti-aliasing and compression for large fonts. So the implementations diverged. 
There are some other things that need fixing: e.g. the DrawMode enum is misleading, there is no proper error propagation (errors like unknown glyphs are just ignored) and the code structure could be improved. Additionally, I have a working prototype to allow using the vendor waveform files, which will be slower but offer more accurate colours. I want the option to use them seamlessly if you have them. 
Since these are a lot of breaking changes anyway it would be the perfect opportunity to get rid possible future conflicts like this or think about a high-level C++ wrapper.

> But I'm really eager to try it out and make EPDiy work in the Inkplates too (I need to check first this hardware thing with the 100pF capacitor though)

Interesting approach they took there :D It would be interesting to play around with them, but I don't think I want to spend that money with plenty of working boards lying around.

martinberlin, 2021-01-29:
Thanks for your comments. I must say started like I year ago with C++ and to make embedded Firmware so still a long way to get the object oriented part right.
Cale-idf is my first project where I tried to use OOP in a complex way (Inheritance, Injection, and all the other goodies)
But it's still a long way to be an ideal implementation. Of course I will be very happy to help and learn more, next step that I want to take is to add the [Touch component](https://github.com/martinberlin/FT6X36-IDF) for this T5S from Lilygo.
I ported an Arduino-esp32 project to ESP-IDF there and it's working nicely, at least to detect Tap events.

martinberlin, 2021-03-22:
Yesterday we bring back a very old ebook to life thanks to EPDiy. Just wanted to thank for the support! 
Still need to calibrate colors and copy here the model details. Here a small preview:
https://twitter.com/martinfasani/status/1373937158709448705
Info about this epaper: 
LB060S01-RD02  used in menuconfig-> Component config -> E-Paper driver this settings
**Display type:** ED097OC4  and **Board:** epdiy v5

vroland, 2021-03-24:
Feel free to add it to the library once you have it calibrated :)

markbirss, 2021-03-24:
@vroland @martinberlin 

It turns out that the kindle has the waveforms for the panel it used stored on the kindle

Mine for example Kindle 4
`[root@kindle scripts]# cd /var/local/eink/panel/
[root@kindle panel]# ls
06_05_009a_3c_152131_05_58_00000db2_85_07.wbf  panel_id
06_05_009a_3c_152131_05_58_00000db2_85_07.wrf  version
[root@kindle panel]# pwd
/var/local/eink/panel
`

`[root@kindle panel]# more /opt/eink/scripts/pvi_barcode_table.txt
E0R     M01     60      # ED060SC4
E1M     M03     60      # ED060SC4H1
E1X     M04     60      # ED060SC4H2
E1U     M06     60      # ED060SC5
E23     M0B     60      # ED060SC5H1
E22     M0C     60      # ED060SC5H2
E2N     M12     60      # ED060SC7
E32     M13     60      # ED060SC7H1
E31     M14     60      # ED060SC7H2
E2R     M23     60      # ED060SC5C1
E30     M24     60      # ED060SC7C1
E43     M3D     60      # ED060SC7T1
L01     MA1     60      # LB060S03-RD01
L02     MA2     60      # LB060S03-RD02
E40     M33     60      # ED060SCF
E48     M34     60      # ED060SCFH1
E49     M35     60      # ED060SCFH2
E4A     M36     60      # ED060SCFC1
E4B     M37     60      # ED060SCFT1
E45     M38     60      # ED060SCG
E4E     M39     60      # ED060SCGH1
E4F     M3A     60      # ED060SCGH2
E4C     M3B     60      # ED060SCGC1
E4D     M3C     60      # ED060SCGT1

E0D     M01     97      # ED097OC1
E0X     M01     97      # ED097OC1
E2C     M0D     97      # ED097OC1H1
E1Y     M05     97      # ED097OC1H2
E29     M0E     97      # EF097OC3
E2B     M0F     97      # EF097OC3H1
E2A     M10     97      # EF097OC3H2
E2S     M15     97      # EF097OC4
E33     M16     97      # EF097OC4H1
E34     M17     97      # EF097OC4H2
E35     M20     97      # ED097OC4
E36     M21     97      # ED097OC4H1
E37     M22     97      # ED097OC4H2`

vroland, 2021-03-24:
That sounds great :) Are waveforms for all the displays on there or just for your specific one?

markbirss, 2021-03-25:
On mine just for my one display, but it seemed from the forum post I left
that at some stage the Kindle DX had other files also?  I can't confirm

cxw0102, 2021-06-04:
I have tested ED052TC2 ,same same as ED047TC2 .

vroland, 2021-06-05:
Good to know :) Do you want to submit a pull request or should I just add it to the table?

cxw0102, 2021-06-06:
You can add it to the table,use ED047TC1 ,don't need to change anything .
I have another display-ED058TC7,it need 50pin connector.but no datasheet :(

mcer12, 2021-06-06:
@cxw0102 find similar one, there are usually alternatives. Try ED058...

vroland, 2021-06-07:
@cxw0102 The ED052TC2 is added :)

martinberlin, 2024-05-06:
Converting this to Discussions since I would like to keep Issues for firmware updates and resolving programming issues. I really like the idea to make a nice repository with details and datasheets of every supported displays.
I know many are not supposed to be distributed but is no problem I will put it under my GitHub and gave the internet police 👮