vroland/epdiy #55: Inductors

amadeok, 2021-01-14:
Hello,
I got the board, i'm trying to solder it. could you tell me if the inductors have polarity and so how to solder them? thanks

vroland, 2021-01-14:
Hi, inductors don't have polarity, so soldering them should be fairly straightforward. Use a large enough iron top to conduct enough heat. For the FPC connectors, using copious amounts of flux helps.

amadeok, 2021-01-14:
Hi thanks for reply. I was able to flash the demo but then the screen turned like this(and hasn't changed again after that):
[image]
is this the correct way to connect it?
[image]
Before i tried connecting it the other way is it possible i damaged it by doing that?

vroland, 2021-01-14:
Make sure to select the correct board revision in `idf.py menuconfig`, getting that wrong sometimes causes such artifacts.
However, connecting it the wrong way can indeed case damage, since in the worst case your connecting +-22V to a 3.3V logic line...

amadeok, 2021-01-14:
Yes i forgot to change the revision and the screen type. but the worst is i connected the connector the other way around. After correcting the board revision and the screen type it worked but had some vertical bars but now is ok, but there seems to be a problem with the greyscale Would you mind checking it this video? 
https://www.youtube.com/watch?v=0xATHM6l_qo&feature=youtu.be

vroland, 2021-01-14:
Which display are you using and do you have the correct display selected in `idf menuconfig`?

amadeok, 2021-01-14:
In the aliexpress insertion is says "ED097TC2 (LF) " and in the menuconfig i put : ED097TC2

vroland, 2021-01-14:
Did you calibrate the VCOM voltage?

amadeok, 2021-01-15:
Yes, -1.33. Perhaps i did damage it? right now i'm using esp-idf version v4.3, the tested one is 4.0 right, later i'll try it?

vroland, 2021-01-15:
Hm, damage in such a subtle way seems unlikely, but it may be possible. I'd recommend to play around with the VCOM and display settings and see if something looks right.
I tested IDF versions 4.0 and 4.2. Not sure why 4.3 would be so different, but it could be worth a try.

amadeok, 2021-01-15:
Hi, 
Yes i'm in a cold room, the sensor of the demo example says "current temperature: 18.700001". do you know what maximum values i can try for the vcom?

vroland, 2021-01-15:
Ok, 18.7°C is not that low... But just for fun, you could also try if a room temperature of ~24°C makes a difference.
The VCOM value range is -0.3 to -4V, according to the data sheet.

amadeok, 2021-03-07:
Hey i got 3 more ED097TC2  yesterday, i tested them and they all look the same as the video. Strange thing is that i also soldered five v4 boards so far, and also with all of them it's the same. This would seem to imply that there is a problem with my esp-idf setup, or maybe my room? it's not so cold right now but there's really a lot of humidity.

vroland, 2021-03-08:
I'll try with my v4 boards and see if something broke in the driver.

vroland, 2021-03-08:
Could you try this firmware image: 
[fw.zip](https://github.com/vroland/epdiy/files/6101008/fw.zip)

amadeok, 2021-03-23:
It seems that .bin file does no longer work because of an update, can you send another?

amadeok, 2021-03-23:
Hey i tested the latest version of the epdiy's demo example, the one with the waveform and it looks similar to the old one:
https://youtu.be/2fZNyx38vl8
I noted that the sensor temperature on the terminal says 21 degrees celsius, but my house's heating system temperature sensor and my multimeter's thermal sensor say it's 16-17. i'm not sure if this is somehow relevant.

vroland, 2021-03-24:
Yes, the ESP heats the board relatively quickly, so the temerature reading will be a few degress to high unfortunately. However, the epdiy waveforms are not yet teperature dependent and calibrated for ~25°C. I don't know how much different the display behaves with 10°C difference, but it could be significant. If you want, you could try to re-calibrate it for your temperature.
For that, you have to edit the scripts/gen_epdiy_waveforms.py with the timings for your display and then run `make clean` and `make` in the repository root. Then test it with the grayscale example. 
I would also be happy to add the calibration for this temperature when it's working :)
Also: which board revision are you using?

amadeok, 2021-03-24:
Hi,
What i have to chage are these? 
`FRAME_TIMES_WHITE_TO_GL16 = {
    "ED097OC4": [3.0, 3.0, 2.0, 2.0, 3.0, 3.0, 3.0, 4.0, 4.0, 5.0, 5.0, 5.0, 10.0, 20.0, 30.0],
    "ED097OC1": [3.0, 3.0, 2.0, 2.0, 3.0, 3.0, 3.0, 4.0, 4.0, 5.0, 5.0, 5.0, 10.0, 20.0, 30.0],
    "ED060SC4": [3.0, 3.0, 2.0, 2.0, 3.0, 3.0, 3.0, 4.0, 4.0, 5.0, 5.0, 5.0, 10.0, 20.0, 30.0],
    "ED047TC1": [3.0, 3.0, 2.0, 2.0, 3.0, 3.0, 3.0, 4.0, 4.0, 5.0, 5.0, 5.0, 10.0, 20.0, 30.0],
    "ED097TC2": [1.5, .8, .8, .8, .8, .8, 1.0, 1.0, 1.0, 1.0, 2.0, 2.0, 5.0, 10.0, 20.0],
    "ED133UT2": [6.0, 3.0, 2.0, 2.0, 2.0, 2.0, 2.0, 3.0, 3.0, 3.0, 5.0, 5.0, 10.0, 20.0, 30.0],
}

FRAME_TIMES_BLACK_TO_GL16 = {

    "ED097OC4": [1.0, 1.0, .8, .8, .8, .8, .8, 1.0, 1.0, 1.5, 1.5, 2.0, 2.0, 10.0, 30.0],
    "ED097OC1": [1.0, 1.0, .8, .8, .8, .8, .8, 1.0, 1.0, 1.5, 1.5, 2.0, 2.0, 10.0, 30.0],
    "ED060SC4": [1.0, 1.0, .8, .8, .8, .8, .8, 1.0, 1.0, 1.5, 1.5, 2.0, 2.0, 10.0, 30.0],
    "ED047TC1": [1.0, 1.0, .8, .8, .8, .8, .8, 1.0, 1.0, 1.5, 1.5, 2.0, 2.0, 10.0, 30.0],
    "ED097TC2": [1.5, .8, .4, .3, .3, .3, .3, .6, .6, .6, .6, 1.5, 2.0, 5.0, 15.0],
    "ED133UT2": [2.0, 1.5, 1.5, 1.5, 1.5, 1.5, 1.5, 1.5, 1.5, 2.0, 3.0, 3.0, 5.0, 10.0, 20.0]
}

FRAME_TIMES_DU = {
    "ED097OC4": [100, 100, 100, 100],
    "ED097OC1": [100, 100, 100, 100],
    "ED060SC4": [100, 100, 100, 100],
    "ED047TC1": [100, 100, 100, 100, 100],
    "ED097TC2": [100, 100, 100, 100, 100],
    "ED133UT2": [100, 100, 100, 100]
}
`
I have to change the values of the display i'm using until it looks good?
I'm using the v4 revision

vroland, 2021-03-24:
Yes, that's right. In `FRAME_TIMES_WHITE_TO_GL16`, the larger the time values the darker in white to black transitions,
for `FRAME_TIMES_BLACK_TO_GL16` larger values mean lighter in black-to-white transitions.
The FRAME_TIMES_DU are only from going from completely black to completely white, you can just copy the other displays there.
The, you also need to add your display to the makefile over here: https://github.com/vroland/epdiy/blob/240f1f792a99eade18411e1a1bb04970e8c8842b/Makefile#L3

amadeok, 2021-03-24:
The only display i'm using is the ED097TC2, do I still have to add it that list?

vroland, 2021-03-25:
Ah, right. Since you are just modifying the existing one you don't have to add it.

mcer12, 2022-04-22:
Stale, closing.

closed 2022-04-22 by mcer12