vroland/epdiy #58: some suggestions

lixy123, 2021-01-22:
1. Because there is no corresponding API, it is difficult to calculate the word wrap. Can you provide the ability of automatic line wrap?
2. Can you provide the API function of drawjpg() or drawbmp() ?

This is another API about ink screen project, With the ability to read and display JPG, BMP, PNG,
for reference
https://github.com/m5stack/M5EPD/blob/main/src/M5EPD_Canvas.cpp

aceisace, 2021-01-23:
I agree. This project has some great potential. It would be awesome to add native support to display a BMP file e.g. by downloading it as a BMP file or simply downloading it's buffer.

vroland, 2021-01-23:
Those would be indeed nice to have. I'm planning on implementing a Arduino-Style layer on top of the C API that implements more convenience functions. This is where those would fit in. 
I can't give a timeline for that though. In the meantime, if you want an example for jpeg decoding you can have a look at the mpd example.

lixy123, 2021-01-25:
Thanks for the sample code of parsing and displaying jpg image. I'll take time to try it. However, it should be better if this code is in the API library.  I don't know if there is an API for screen coordinate rotation.

vroland, 2021-01-25:
Yes, that would make sense. I'll mark it as a possible enhancement.

martinberlin, 2021-03-25:
Hi Valentin,
Small suggestion (Since I don't think makes sense to open an Issue for this)
**demo Example** 
https://github.com/vroland/epdiy/blob/master/examples/demo/main/main.c#L23

Does not compile for EPD_WIDTH <1000 since it does not find the file: 
#include "firasans_12pt.h"

The correct file name is firasans_12.h

DavidM42, 2021-04-20:
I think those are great suggestions too :+1:  .
One more nice thing would be to add a return boolean if the wrapped text fits onto the area/display. This could be coupled with some kind of pagination feature, so any amount of long text can be displayed via pages

martinberlin, 2022-06-21:
Closing here since we already have another thread on suggestions #31
Please add any further suggestions there.

closed 2022-06-21 by martinberlin