vroland/epdiy #59: Terminal: Text Color fades

markbirss, 2021-01-25:
After some idle time using the Terminal example the text fades away

[image]

what could be causing this ?

vroland, 2021-01-25:
Maybe the common voltage is not adjusted for your display? And is the display always powered or only when writing? How long does it take until it fades?

markbirss, 2021-01-25:
@vroland 
I have adjusted VCOM for my display

I wil try check how long it actually takes before fades

Running a 

`watch uptime`

the text color does not fade

vroland, 2021-01-25:
Actually I noticed that I had something with my terminal as well: After a while (1-2h) of use, it suddenly started to have very low contrast. Just from one key press to the next. Just resetting the board fixed it, so i didn't bother to really look into it.
Is it the same for you, or does it fade gradually?
Maybe be some kind of memory corruption or overflow in the terminal code?

markbirss, 2021-01-25:
yes same for me just suddenly after a couple minutes

if i run top at this stage, top looks ok but when i quit returnong to the
prompt all text entered is with very low contrast still

yes pressing reset restores contrast

so i appear o have same issue

vroland, 2021-01-26:
Ok, good to know. Is there any output on the debug serial?

markbirss, 2021-01-26:
@vroland 

will have a look and report back

markbirss, 2021-01-26:
@vroland 

when screen contract very low ,I do not see any message

`--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7416
load:0x40078000,len:14368
ho 0 tail 12 room 4
load:0x40080400,len:5360
0x40080400: _init at ??:?

entry 0x40080710
I (30) boot: ESP-IDF v4.2 2nd stage bootloader
I (30) boot: compile time 17:43:50
I (30) boot: chip revision: 1
I (33) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32: SPI Speed      : 80MHz
I (43) boot.esp32: SPI Mode       : QIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (76) boot:  1 phy_init         RF data          01 01 00017000 00001000
I (83) boot:  2 factory          factory app      00 00 00020000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0xafd00 (720128) map
I (342) esp_image: segment 1: paddr=0x000cfd28 vaddr=0x3ffb0000 size=0x002f0 (   752) load
I (343) esp_image: segment 2: paddr=0x000d0020 vaddr=0x400d0020 size=0x1ca30 (117296) map
0x400d0020: _stext at ??:?

I (388) esp_image: segment 3: paddr=0x000eca58 vaddr=0x3ffb02f0 size=0x02c48 ( 11336) load
I (392) esp_image: segment 4: paddr=0x000ef6a8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/user/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (395) esp_image: segment 5: paddr=0x000efab4 vaddr=0x40080404 size=0x0f56c ( 62828) load
I (437) boot: Loaded app from partition at offset 0x20000
I (437) boot: Disabling RNG early entropy source...
I (437) psram: This chip is ESP32-D0WD
I (442) spiram: Found 64MBit SPI RAM device
I (447) spiram: SPI RAM mode: flash 80m sram 80m
I (452) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (459) cpu_start: Pro cpu up.
I (463) cpu_start: Application information:
I (468) cpu_start: Project name:     terminal
I (473) cpu_start: App version:      hw5.0-rc1-dirty
I (478) cpu_start: Compile time:     Jan 26 2021 17:43:46
I (484) cpu_start: ELF file SHA256:  004f7f658ba4e55a...
I (490) cpu_start: ESP-IDF:          v4.2
I (495) cpu_start: Starting app cpu, entry point is 0x40081af4
0x40081af4: call_start_cpu1 at /home/user/esp/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (977) spiram: SPI SRAM memory test OK
I (978) heap_init: Initializing. RAM available for dynamic allocation:
I (978) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (984) heap_init: At 3FFB7500 len 00028B00 (162 KiB): DRAM
I (990) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (997) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1003) heap_init: At 4008F970 len 00010690 (65 KiB): IRAM
I (1009) cpu_start: Pro cpu start user code
I (1014) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1034) spi_flash: detected chip: generic
I (1035) spi_flash: flash io: qio
I (1035) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1049) esp_temperature: Characterized using eFuse Vref

I (2553) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (2553) term: listening

I (7098) term: clear mode: 2
I (7099) term: epd clear.
I (14162) term: clear mode: 2
I (14163) term: epd clear.
I (17498) term: clear mode: 2
I (17499) term: epd clear.
I (25938) term: clear mode: 2
I (25939) term: epd clear.
I (495793) term: clear mode: 2
I (495794) term: epd clear.
I (515422) term: epd clear.
I (516413) term: clear mode: 2
I (518978) term: epd clear.
I (525547) term: epd clear.
I (526537) term: clear mode: 2
I (1624646) term: epd clear.
I (4924057) term: clear mode: 2
I (4924058) term: epd clear.
`

vroland, 2021-01-27:
Hm, looks like we need to dig a bit deeper here.

vroland, 2021-01-31:
Fixed in #63.

closed 2021-01-31 by vroland