vroland/epdiy #60: Unable to flash the board

amadeok, 2021-01-26:
Hello,
I'm having issues flashing projects to the board. Sometimes i can, then after a while i can no longer flash, i get error: "A fatal error occurred: Timed out waiting for packet content"
Would you happen to know any solution? I was thinking about replacing the HL-340 
thanks

vroland, 2021-01-26:
Which board revision are you using? Maybe try a different cable or reduce upload speed?

amadeok, 2021-01-26:
> Which board revision are you using? Maybe try a different cable or reduce upload speed?

Hi thanks for reply. I have revision 4, i did try another cable and a lower upload speed. I was thinking about soldering some wires to the controller board and using a standard esp32 development board( such as : https://images-na.ssl-images-amazon.com/images/I/61wqT2HjXxL._AC_SL1000_.jpg) to power the controller board and control the signals, could this work?

vroland, 2021-01-26:
You could just solder to the RX / TX lines to a normal usb to UART converter, like a esp8266 programmer.
Bodging such a complete module seems more difficult.

vroland, 2021-02-28:
@amadeok Do you still have problems here or can this issue be closed?

amadeok, 2021-02-28:
Hi, yes you can close it, it seems that that board got damage somehow

closed 2021-02-28 by vroland