vroland/epdiy #65: Discontinued part

bartlammers, 2021-01-31:
In the BOM, the 200k potentiometer is discontinued. I can't find a replacement on LCSC either I'm afraid. Ordered them off AliExpress now and Western vendors have Bourns 3x3 pots so it's not impossible to get, but do you know of a replacement part on LCSC perhaps for other future builders?

It's the EVM3ESX50B25 btw

vroland, 2021-02-02:
Ok, good to know. I'll look around for suitable replacements some time.

mcer12, 2021-02-06:
You can grab that 200k trimmer anywhere else, it's a generic part. Aliexpress is an obvious choice, or any other source.

bartlammers, 2021-02-06:
I know, was mainly mentioning it for the bom

markbirss, 2021-02-06:
@bartlammers 

If I try a quick quote for PCB assembly for the just your recently updated V5 board incl issue - Adjust solderpaste after JLCPCB feedback

A total of 6 parts are not selected ?

What is the accepted process to fix this should one want to correct this ?

RV1 | 200k Potentiometer | EVM3ESX50B25 | No part selectedSearch | 1
-- | -- | -- | -- | --
U6 | X05A20H34G | X05A20H34G | No part selectedSearch | 1
U10 | ED060SC7 | AXT334124 | No part selectedSearch | 1
U2,U3 | LT1945 Dual Micropower DC/DC Converter with Positive and Negative Outputs, MSOP-10 | MSOP-10_3x3mm_P0.5mm | No part selectedSearch | 2
J2 | C10418 USB Micro Type B connector | USB_Micro-B_A01SB141B1-067 | No part selectedSearch | 1
J1 | XF2M-3315-1A | CONN_ED097OC4 | No part selectedSearch | 1

mcer12, 2021-02-06:
Well the accepted process would be to get these parts elsewhere (and solder them yourself) ;) LT1945 and compatible connectors are available on aliexpress. USB is also on ali but there are plenty compatible connectors on lcsc to choose from. Some stuff just isn't available on lcsc so you can't have it assembled from jlcpcb.

markbirss, 2021-02-06:
Ah Ok, i got some quotes here in South Africa, the one company does not assemble pcb and the other quoted on assembly but not the parts. 

Already too expensive here

mcer12, 2021-02-06:
@markbirss in that case I would suggest to push the boundries and do the assembly ourself!

ACE1046, 2021-02-13:
U6 (X05A20H34G) is available on lcsc, C528053.
J1 (XF2M-3315-1A) - C231420
J2 C10418 - available.

closed 2022-05-14 by martinberlin