vroland/epdiy #68: Memory footprint of epd_init()

lyusupov, 2021-02-28:
The issue was found while using LilyGO's port of the **epdiy** library for Arduino/ESP32:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47
However, is seems to be applicable for the 'master' library as well.

Ok. Lest add few debug print messages near epd_init() function, as follows:

```
Serial.print(F("before epd_init: Free heap size: ")); Serial.println(ESP.getFreeHeap());

    epd_init();

Serial.print(F("after epd_init: Free heap size: ")); Serial.println(ESP.getFreeHeap());
```

This is a console output for the test above running on WROVER-E with 8 MB of PSRAM:

```
before epd_init: Free heap size: 194912
after epd_init: Free heap size: 97856
```

So, the **epd_init()** takes about 97 Kbytes of built-in ESP32 static RAM. 

Is it good or bad? The answer depends on whether or not we gonna use other functions of the ESP32 SoC...
The space remaining is Ok to bring up Wi-Fi stack or Bluetooth, but not both. :-(
  
The **epd_init()** function code contains the following:
```
conversion_lut = (uint8_t *)heap_caps_malloc(1 << 16, MALLOC_CAP_8BIT);
```

When we alter the line onto this (valid for built together with Arduino/ESP32) :

```
    conversion_lut = (uint8_t *)heap_caps_malloc(1 << 16,
                          psramFound() ? MALLOC_CAP_SPIRAM : MALLOC_CAP_8BIT);
```

&nbsp;- it's getting much better now:

```
before epd_init: Free heap size: 194912
after epd_init: Free heap size: 163408
```

My sketch works fine to me when connversion LUT table is located in PSRAM. However, we yet need to:
- find an equivalent of psramFound() in ESP-IDF, and
- make sure that there is no significant performance reduction.

vroland, 2021-02-28:
Hi,
yes, 97k may be a bit high for some applications... 
Moving the lookup table to PSRAM however will incur a large performance hit, because it is larger than the PSRAM cache and interferes with reading the framebuffer from PSRAM.
But I could implement the option to reduce the size of the LUT: 64k allows to look up 4 pixels with one load, which is also a full cycle of parallel data to the display. But I could split that up in two loads + shift + or, which will not be as fast but reduce the LUT size to 256 bytes.
I'll have a look if the changes needed are small enough to implement it in #61.

ashald, 2022-02-14:
Beating a somewhat dead horse here - I'm trying to minimize energy consumption of a run cycle in between deep sleep, and I found that enabling PSRAM adds ~800ms to the boot time, which is about 1/3 of the entire runtime.

If I only need black & white (I'm experimenting with Lilygo FWIW), and `MODE_GC16` (which gives me the crisp image I'm looking for), can I reduce the size of the framebuffer from `EPD_WIDTH / 2 * EPD_HEIGHT` somehow, so that I could fit it into the main memory?

Thanks!

vroland, 2022-02-14:
You can, using the low-level api: https://github.com/vroland/epdiy/blob/master/src/epd_driver/include/epd_driver.h#L136 as a starting point. This packs 8 pixels in one bye. I don't have an example for this mode though, and haven't used it in a long time...

martinberlin, 2022-02-14:
@ashald if you check the codebase a bit more you will realize that epd_hl_update_area uses MODE_PACKING_2PPB as default (exactly coherent with the slow mode you are using that is MODE_GC16). In order to use that 8PPB mode you will have to make your own function if I see that correctly and call epd_draw_base using that mode (There is no example so far I think).
Anyways will all due respect this question is a bit off topic for this issue considering that is about the initial memory footprint on the initialization of the library and not about the drawing modes of EPDiy.

ashald, 2022-02-14:
@martinberlin thanks! Would appreciate if you could suggest a better place to ask such questions.

One thing that I'm confused about is that it seems that the base drawing function for a single pixel is explicitly using 2-pixels-per-byte approach https://github.com/vroland/epdiy/blob/f3feb5a93c2770260d46671a05e8341f3e6d06ae/src/epd_driver/epd_driver.c#L87, which in my understanding means it cannot work with any other packing mode (as is), and therefore would require a buffer of the same size.

martinberlin, 2022-05-14:
Closing here, since there is no more feedback since mid-February, and does not seem a bug that can be solved with EPDiy.

closed 2022-05-14 by martinberlin