vroland/epdiy #69: Wrong voltage

amadeok, 2021-03-06:
Hey, 
I was soldering a v4 board today i finished it then when i powered it on i saw that there was around 5v  where there should have been 3.3v, and also it seemed that ground and 3.3v were shorted. After some testing i think the mistake was that i soldered the D4 diode the wrong way (it's the one that is direclty connected to +15v on one side and to  one of the inductors on the other)
I corrected that and now 3.3v and ground are no longer shorted but still i'm getting around 4-5v where there should be 3.3v. THe usb to uart transceiver is not recognized by the pc and i cannot flash the esp32 with an external usb to uart board. should i just discard the board?
thanks

vroland, 2021-03-07:
I'ts possible that you fried the converter already, but the voltages should not be wrong once the mistake was corrected. Maybe check that all the other diodes and ICs are correctly oriented? That way, you'll at least have a correct one as a template.

amadeok, 2021-03-07:
Hi, ok thanks. is it ok if i power a board without the transceiver, the esp32 module and the U8 shift register soldered?

vroland, 2021-03-07:
It shouldn't hurt it if that's what you mean.

closed 2022-04-22 by mcer12