vroland/epdiy #70: ED060KG1(LF) Support

adyanth, 2021-03-08:
This is the display for Kobo Glo HD [here](https://e-ink-reader.ru/screen_info_en.php?dev_id=135). I see the ED060KD1 and ED060KC1 in the README which match the resolution and pin count and should work in V5 of the board. Any reason ED060KG1(LF) is not supported or is it just left out of the list?

vroland, 2021-03-09:
As far as I can tell it should be compatible. I can't confirm it first hand, but since the pinout is equivalent to the XC5 it should be drivable. You may have to add a configuration option to set the correct resolution and calibrate the grayscale timings.

mcer12, 2021-03-16:
Seems similar enough, 99% it's compatible. Once you test it, let us know so it can be added to supported displays @adyanth

adyanth, 2021-03-16:
Thanks! Will try to build and test. Although getting the PCB printed near me is a task and a half right now.

mcer12, 2021-03-16:
@adyanth use jlcpcb assembly, it's cheap and quite fast.

adyanth, 2021-03-16:
That's the 1st thing I looked at, but with minimum 5 pieces and all of them soldered with components gets expensive with 5 ESP32s.
I did not find an option of only getting one of them built.
And I don't have the equipment to use the solder balls to do it myself :(

mcer12, 2021-03-16:
assembling 5 boards will be probably still cheaper than doing one board localy... sadly. You can save quite a bit by skipping ESP and LT1945 and buy those separately from aliexpress and solder yourself.

markbirss, 2021-03-18:
@adyanth @vroland 

To get my ED060KC1 (1072 x 1448) to work I had to modify 
`../../src/epd_driver/include/epd_internals.h `

with a EPD_WIDTH value of 1442

then select running 
`idf.py menuconfig`

`(Top) → Component config → E-Paper Driver
                                   Espressif IoT Development Framework Configuration
    Display Type (ED097TC2)  --->
    Board / Board Revision (epdiy v5)  --->`

`  GNU nano 5.4                          ../../src/epd_driver/include/epd_internals.h                                    
/**
 * @file epd_internals.h
 * @brief Internal definitions and auxiliary data types.
 *
 * Unless you want to extend the library itself (Which you are very welcome to do),
 * you will most likely not need to know about this file.
 */

#ifndef EPD_INTERNALS_H
#define EPD_INTERNALS_H

#include <stdbool.h>
#include <stdint.h>

/// minimal draw time in ms for a frame layer,
/// which will allow all particles to set properly.
#ifndef MINIMUM_FRAME_TIME
#define MINIMUM_FRAME_TIME 12
#endif

/// Frame draw time for monochrome mode in 1/10 us.
#define MONOCHROME_FRAME_TIME 100

#if defined(CONFIG_EPD_DISPLAY_TYPE_ED097OC4) ||                               \
    defined(CONFIG_EPD_DISPLAY_TYPE_ED097TC2) ||                               \
    defined(CONFIG_EPD_DISPLAY_TYPE_ED097OC4_LQ)
/// Width of the display area in pixels.
#define EPD_WIDTH 1442
/// Height of the display area in pixels.
#define EPD_HEIGHT 1072
`

[image]

vroland, 2021-03-24:
I'll mark it as a bug and look into it some time :)

schroedi83, 2022-02-20:
I would appreciate if someone could help, same issue here...

[image]
[image]

mcer12, 2022-02-20:
@schroedi83 not the same issue ;) That panel has a resolution of 1440x1072, I don't think that resolution is supported.
You can edit the resolution in src\epd_driver\include\epd_internals.h
Not sure if there's an easier solution to this at this point.

schroedi83, 2022-02-20:
@mcer12 Thank you. I guess my second photo was misleading. I used your
inkster platformio example and edited the resolution in
src\epd_driver\include\epd_internals.h as @adyanth did and reproduced the
error. When using 1442 the display is updated but there is a small black
bar (those 6 missing lines I guess). Using full 1448 the refresh didn't
work anymore.

[image: ED060KG1.jpg]
[image: ED060KG1 1448.jpg]

mcer12, 2022-02-21:
@schroedi83 Sorry you're correct, there's some weird shenanigans happening. Try to set width to 1600, it should work as a workaround.

mcer12, 2022-02-21:
@schroedi83 Looks like the problem is that the width is not divisible by 16 (1442 works but that's more by accident :D), so you can use the closest higher number which is:
#define EPD_WIDTH 1456
@vroland maybe worth looking at.

closed 2022-04-22 by mcer12