vroland/epdiy #71: Adding sofware sofware rotation

martinberlin, 2021-03-25:
I started a [EPDiy Fork](https://github.com/martinberlin/epdiy-rotation) with the mission to add sofware rotation to the Epaper.
I want to leave this ticket open to document the challenges I go into but also to ping @vroland and other developers that have more experience than me with this library in case I get stuck. 

### So far I have this 3 points open:

1. Add software rotation in **epd_draw_pixel** ( epd_driver.c ) This I already did in Lilygo repository and works well. Then automatically all geometric functions that draw using epd_draw_pixel will be already rotated
2. Add rotation to **fonts**. According to Valentin this should be easier in this new version of the component
3. Add rotation to **image** functions like **epd_copy_to_framebuffer**. This one looks the most complicated, since the whole "buffer" should be rotated a part of the initial x & y coordinated.

About 3 I'm not sure if it's the best to do it in epd_copy_to_framebuffer or on a helper function that will rotate the pixels before sending it to the mentioned function.

### Possible rotation values

0 Default landscape (As is now)
1 90° clockwise Portrait
2 180° inverted landscape
3 270° inverted Portrait

martinberlin, 2021-03-25:
Update: Working on point 2
[image]
If anyone can take a look and point me what I'm doing wrong I will appreciate it. It renders the font but there is a vertical empty line so I'm obviously missing something. Note that the original function prints directly on the buffer and with my update I'm printing it via the epd_draw_pixel function.

vroland, 2021-03-26:
> QUESTION: warning: unsigned conversion from 'int' to 'unsigned char:4' changes value from '120' to '8' [-Woverflow]
> font_props.fg_color = 120;

The color here is a 0-15 value (0x0-0xF). That's why you get the overflow error. But 120 overflows to 8, which is coincidentally very similar to the scale of gray you expect ;)

vroland, 2021-03-26:
And a simple version of image rotation could be to implement a get_pixel function which basically inverterts the draw_pixel function, and then draw_pixel it to the new buffer. This won't be the fastest solution of course, but I think it's the simplest one.

martinberlin, 2021-03-26:
>get_pixel function which basically inverts the draw_pixel function, and then draw_pixel it to the new buffer

Good idea!
**One question:** How much free PSRAM is left when you use a 11 inches epaper?
Because I was thiking the best will be to download the whole image or even make a SPIFFS example, where you read the file and keep it on the PSRAM. But I don't know if it will be enough memory for that, I guess yes since the Radionica examples (Inkplate) did that. But I tested and it was not as stable as mines. 
I do not keep the BMP in memory, I just read it and "stream it" to the drawPixel function. But for JPEG or PNG I guess you need the whole image available since they are formats that need compression.

vroland, 2021-03-26:
That depends on the resolution. Which resolution does the 11" display have?

And regarding the images: At least with JPEG, you can decode it in a streaming fashion. Don't know about PNG though.

martinberlin, 2021-03-27:
I meant something like  13.3" | 1600 x 1200  but yes I think should work since Inkplate 10 is rendering images with the standard PSRAM too. 

Thanks for the feedback. Rotation points 1 & 2 are done. 
Considering the image rendering should be done using epd_draw_pixel then I could add the examples later since I also want to add PNG rendering to my component. Just tell me when it's the best time and I will do a Pull request so you can review my updates.

vroland, 2021-03-27:
You can always open a pull request and continually update it, and I'll just merge it when it's done ;)

martinberlin, 2021-03-27:
Sounds good. About:

> The only caveat that still don’t know how to solve is the issue of partial update when you are using the screen rotated.

Do you have any idea what would be the way to have partial refresh working also when the display is software rotated? Sounds quite complex for me but maybe I’m missing the right way to do it.

vroland, 2021-03-27:
You mean the partial refresh as implemented by the highlevel driver? This should work pretty much the same, except for rotated corrdinates. The update difference calculation doesn't care if the contents of the framebuffer are rotated, it just outputs the difference. So as long as the coordinate transformation is correct, it shouldn't be a problem.
I.e. the function for updating the complete screen should pretty much work as-is, and the epd_hl_update_area function would need to map the rotated coordinates to the non-rotated framebuffer.

martinberlin, 2021-03-28:
Nice in las commit rotation in partial update is already working for 2 (inverted landscape)
Still could not find why it does not work on portrait but I will try to find out tonight. If you can make a fast review of my _rotate function.

martinberlin, 2021-03-29:
Pending: 

- [ ] Add boundaries and error handling 
- [ ] Test all points again with different display sizes

Rainerlan, 2021-04-01:
@martinberlin,
I was trying out several screen rotations: 0, 1 and 2 for drawing lines, rectangles & text.
Only with rotation 1, the display content (all of it) looks nice and crisp (dark black color).
When I use rotations 1 and 2, the drawn content is very light grey - almost invisible.
When trying out the rotations, I was only changing the rotation in `epd_set_rotation(0)`. All oder code has been left unchanged.
Do I use the API somehow wrong?

martinberlin, 2021-04-01:
>When I use rotations 1 and 2, the drawn content is very light grey - almost invisible.

Hi Rainer, thanks for your feedback, you mean 2 and 3 (Since you said that with 1 looks ok)
I tried over here but I could not reproduce it. Can you post a picture so I can take a deeper look?
Note that I tried only with the Lilygo 4.7 and a 6" epaper so I don't know how it look in another epaper models.

Rainerlan, 2021-04-01:
Hi @martinberlin ,
Many thanks for you quick reply!
I figured out a few things:
- using the highlevel-API, as in your examples, the display seems to works fine (Will try out more drawing functions tomorrow).
- As I use cpp, I had to add "extern C" to the epd_driver.h to use the highlevel-API, as you did.
I will investigate further - but it seems the problem lies in the way, I used the "non-highlevel"-API functions before.
(I had copied parts from the "weather.cpp" from vroland - where he also uses the "non-high-level"-API)
Interestingly, the display worked...somehow. 
I'll continue testing;) Maybe I can figure out what is the problem with the "non-high-level"-API.
Although I am pretty sure now its only the way I use the provided functions.
Btw.: Great work on the rotation. Many thanks for the effort you have spent!

PS: You are right - there is an error in my initial post:
With rotation=1, the output is "dark & crisp" - with **0** & 2 its fading/grey.

martinberlin, 2021-04-02:
Great feedback and research thanks for trying this out. About adding extern C to use high level API you have to show me how you did it. I solved the problem adding [three function hooks that call highlevel directly from Epd-driver in my C++ component](https://github.com/martinberlin/cale-idf/blob/master/components/epd_driver/epd_driver.c#L346). Even though highlevel.c was in the build logs I could not access the functions inside.

Rainerlan, 2021-04-02:
I have attached the epd_driver.h from epd/include directory. With this I can compile/link and use the highlevel API functions without problems.
What I noticed this morning, is that after updading the display, some of the content fades away...
I may jsut use functions incurrectly - but this time I really only do the following:

`
epd_poweron();
int temperature = 25;
EpdDrawError err = epd_hl_update_screen(&hl, MODE_GL16, temperature);
Serial.println("EpdDrawError err:" + String(err));
epd_poweroff();
`

Any feedback appreciated;)

[epd_highlevel.zip](https://github.com/vroland/epdiy/files/6249059/epd_highlevel.zip)

https://user-images.githubusercontent.com/179617/113407965-b4180e00-93ae-11eb-98e3-705947901d84.mp4

martinberlin, 2021-04-02:
Hi Rainer thanks for sharing, this is the weather example or it’s your custom program? 
I have also a Lilygo 4.7 so I can try to reproduce it if I can use the same program. That is rotation 3, right? Is only fronts that appear gray or also other geometric functions?
And do they get back to normal if you use rotation 0 ?

Rainerlan, 2021-04-02:
Hi martinberlin, I used bits and pieces from various examples and "old" code I had. E.g.
- I have created own font-header files for meteofont and moonphases. 
- the code connects to my MQTT server to obtain the openweather-data json data from my fhem server. 

So it will not run out-of-the box as e.g. I loop over a successful MQTT connection.
To share it, I could strip down the code however and use e.g. fixed codes from the fronts (and remove MQTT connection check)
Would be great if you could reproduce it.
After playing around a bit, I found that the original feed_display functions that was used in 
"https://github.com/Xinyuan-LilyGO/LilyGo-EPD47" was completely rewritten.
I suspect that somehow the waveform or some LUT does not perfectly match the display - Still it could be a problem of the configuration I use (Target is to use rotation 3 / 270degree).
What I find noteworthy is, that with each screen-rotation setting, some ohter part of the display shows "faded/low-contrast" areas.
@vroland: In the Readme.md, you list the 4.7" e-paper board by LILYGO as "tested". Have you experienced any difficulties with the contrast?

vroland, 2021-04-02:
Unfortunately I can only test this after easter, when I'm home again. But I did not encounter the fading problem on the LILYGO board, yet (at least that I know of). Does it happen with the master branch as well, or just with the rotation branch? I'm also not quite sure what you mean with the fading, could you point out where to look for in the video?

martinberlin, 2021-04-02:
I tried today to flash the weather example. It took me a while to figure out how to get Arduino as a component.
And I switched the branch in my fork epdiy-rotation from develop (where is rotation implemented) to master (original epdiy)
I could not experience this fading effect. This are the results, notice it looks strange in this screen I think it was done for a bigger epaper:
[image]
[image]

Rainerlan, 2021-04-02:
In the video, when having a close look on the battery percentage (top right) it most obvious:
2s into the video, the screen gets updated - and the small 8pt font of the "100%" is getting dark and crisp
about 500ms later, 3s into the video, the "100%" is diminishing/fading.
I have made several videos shorty after each other - with just the rotation changed. The fading is mostly visible with rotation = 3.
While the effect is always visible, it seems to change during the day.
Using Xinyuan-LilyGO, the text was always crisp & dark.
Could it be also temperature dependent? How big an effect would an offset of eg. 5 deg. make?

https://user-images.githubusercontent.com/179617/113445800-d766ab80-93f6-11eb-9d8c-4cca49e2107b.mp4

https://user-images.githubusercontent.com/179617/113445819-d9c90580-93f6-11eb-9388-3285551db059.mp4

https://user-images.githubusercontent.com/179617/113445828-db92c900-93f6-11eb-9057-a70c58d907d6.mp4

[image]

martinberlin, 2021-04-02:
I’m clueless here, tried to reproduce this using rotation 3 and I see the weather title Berlin with the date correctly and with same contrast that is up there (just inverted)
Let’s wait for Valentin to try it maybe he discovers something. I’m quite sure that this is not related to the rotation otherwise I will get it too when printing a font in the same location.
It could be something about how the partial update is done. I would try first to print a simple font and then keep further. This partial update modes is something that changed quite a lot since Lilygo EPDiy version to this one.

Rainerlan, 2021-04-02:
After having stripped down everything to minimal I have zipped the vsc/.pio project to the attached .zip file - including firmware.elf for rotation swiching cyclically 0->1->2->3.
I agree - this most probably has nothing to do with the rotation. Maybe with the temperature or VCOM, or some other strange effects (With rot=3, the WROVER module sits under the statusbar on the display). 
Everytime I try, the fading looks a little different.
Find attached a photo - 2 shots, taken shorty after each - montaged side-by-side - one with rotation = 0, the other with rotation =3.
The difference is not VERY noticable currently - but it still exists...
[image]
[rotTestFadeMinimal.zip](https://github.com/vroland/epdiy/files/6251520/rotTestFadeMinimal.zip)

Lorysmus, 2021-04-03:
Hi there, 
Nice work with the rotation!
I played a bit with the rotation with the Lilygo 4.7" and this is the result for rotation(1):
For the shapes it looks very nice - I just have to set the Y -zero-point to 240 so the whole y range is 240..1200 and the x range is 0..540.
With the text there is a restricted usable area in the Y-range from 240 to ca 840. It is not possible to write anything in the third third of the display.... please check the pic... 

[image]

martinberlin, 2021-04-03:
@Rainerlan I will try tomorrow when I find some time
Thanks Lorysmus I found the bug that was checking the y boundary without taking in account location (corrected in last commit)

>I just have to set the Y -zero-point to 240 so the whole y range is 240..1200

I didn't get this part. For me the y zero point is 0. I used this draw code to test something similar, repeating a string along the y axis using epd_set_rotation(EPD_ROT_PORTRAIT);

```c
uint8_t* fb = epd_hl_get_framebuffer(&hl);
  epd_poweron();
  epd_clear();
  epd_poweroff(); 
  EpdRect rect = {
    .x = 0,
    .y = 0,
    .width = 30,
    .height = 30,
  };
  // Print small corner squares (0,0 is top left)
  epd_fill_rect(rect,0,fb); // top left
  rect.y = epd_rotated_display_height()-rect.height;
  epd_fill_rect(rect,0,fb); // bottom left
  rect.x = epd_rotated_display_width()-rect.width;
  epd_fill_rect(rect,0,fb); // bottom right
  rect.y = 0;
  epd_fill_rect(rect,0,fb); // top right

  int cursor_x = 100;
  int cursor_y = 20;
  
  EpdFontProperties font_props = epd_font_properties_default();
 
  char srotation[32];
  sprintf(srotation, "Loading demo...Rotation: %d", epd_get_rotation());
  uint8_t counter = 1;
  for (int cursor_y=20; cursor_y < epd_rotated_display_height(); cursor_y=cursor_y+30) {
    cursor_x = 100; // This is because epd_write_string recalculates the cursor and I want to print it in same X
    epd_write_string(&FONT, srotation, &cursor_x, &cursor_y, fb, &font_props);
    printf("counter: %d cursor_x: %d y: %d\n", counter, cursor_x, cursor_y);
    counter++;
  }

  epd_poweron();
  vTaskDelay(100);
  t1 = millis();
  err = epd_hl_update_screen(&hl, MODE_GL16, temperature);
  t2 = millis();
  epd_poweroff();
  printf("EPD update %dms. err: %d\n", t2 - t1, err);
```

martinberlin, 2021-04-03:
@Rainerlan I tried your sketch and can’t reproduce it. For me the top looks crystal clear like all the rest. I don’t see any fading effect. Did you tried to run the examples/demo or grayscale demo to see if there is some hardware issue with the Epaper?
[image]

Rainerlan, 2021-04-10:
@martinberlin, @vroland many thanks for trying out my sketch!
The behaviour of the "fading" seems to be different everyday (sometimes almost invisible, sometimes very bad).
Attached you can find a very bad expample from yesterday (Today it also looks bad - but not as bad as yesterday)
As also from your comments, I now suspect the HW of the LILYGO EPD 4.7.
Probably I will raise an issue in https://github.com/Xinyuan-LilyGO/LilyGo-EPD47

Still I wonder why the effect is most noticable on top of the display when using `epd_set_rotation(3);`.
- Maybe because the ESP32-module is just on the other side (some interferrence)?

Other reasons I may think of:
- Some temperature dependency?
- Inaccuracies of VCOM?

@vroland: From your experience, what  is the impact of a temperature and/or VCOM offset?

Note: I also increased the clockspeed - which had no effect wrt "fading" on my HW.
[image]

vroland, 2021-04-10:
Regarding temperature / VCOM dependence:
 * A wrong VCOM setting could definitely lead to such fading, but in my experiments always accross the whole screen, not on some subsection.
 * I did not explore temperature dependence systematically yet, but especially on the colder side this may have an impact. So it could be that for some reason the components underneath generate excessive heat, but that should also happen on our devices.
 
So I agree that it's probably a hardware problem and I'm a bit out of ideas of what to try. But maybe something fruitful will come from their response.

martinberlin, 2021-05-11:
Since this feature is done and I could not reproduce @Rainerlan on my Lilygo EPD47 using his same sketch the best is to close this one. Just reopen if there is a bug found or some additional fix. Thanks for the collaboration on this one.
I really enjoyed adding this to the component!

closed 2021-05-11 by martinberlin