vroland/epdiy #73: LT1945 issues

M737M737, 2021-04-05:
Hi,
I soldered a v4 board and tested it without the ePaper display connected. Unfortunately, part U2 (LT1945) burnt down between the pins 6 & 7 (top left). As I'm not that experienced at SMD soldering, I gave it a new try and soldered a new board and tested every single connection, it was all fine. However, the part burnt down at exactly the same place.

All parts were ordered from LCSC, LT1945 from AliExpress (https://www.aliexpress.com/item/32863876207.html?spm=a2g0s.9042311.0.0.290d4c4dfx2shh)

Has someone else faced this issue?
Is testing without the ePaper display recommended?

At the 2nd try I didn't solder the esp32 for now. Could this be the problem?

Thanks!

vroland, 2021-04-05:
Hi,
sorry to hear that. Maybe check that the orientation on all the parts is correct? The silkscreen markings seem to be unfamiliar to some people: The elongated line correspons to where the dot on the package should be. So maybe something is wrong there?
Also, which diodes are you using? At the beginning of the project, I had a chip burn out because I used normal diodes instead of schottky diodes.
Lastly, counterfeit chips on AliExpress are not unheard of. If you ruled out the other reasons, could you upload a picture of the package markings? 
Hope you find it!

Regards,

Valentin

M737M737, 2021-04-05:
Hi,
I just double checked the orientation and it should be fine. (see attached photos for reference)
[image]
[image]
[image]

Schottky diodes should be also ok: https://lcsc.com/product-detail/Schottky-Barrier-Diodes-SBD_ON-Semiconductor-MBR0540T1G_C21353.html
For orientation, I reffered to the kicad v4 breakout plan.
[image]

I also attached a picture of the package. I got 10 pc., no further markings or information.
The only interesting thing is the additional "e3" marking on some pieces.
[image]

Any ideas?

Thank you very much!

vroland, 2021-04-05:
Hm, I don't see anything wrong either. I'll compare your photos with my boards when I get home after easter at the end of the week. Unfortunately I can't really do anything more from here :/

M737M737, 2021-04-06:
Alright, thank you!

vroland, 2021-04-09:
Hm, of mine, none have the additional e3... Everything else seems to match though :/

M737M737, 2021-04-10:
Ok, can you send me a link of your LT1945's?

vroland, 2021-04-10:
This is the link I've used: https://www.aliexpress.com/item/2053692331.html?spm=a2g0s.9042311.0.0.27424c4dbMlSvC although the listing seems to have been replaced since then, because it only says 2 orders.

mcer12, 2021-04-11:
@M737M737  There's some other issue, probably a short somewhere, maybe a diode soldered in reverse. Check everything connected to that LT1945. It's 99% not problem with the regulators. I have both types, ones with "e3" in the corner and ones without it and they all work fine.

M737M737, 2021-04-13:
Ok, thank you! I'll check that again.

markbirss, 2021-05-02:
> Hi,
> I just double checked the orientation and it should be fine. (see attached photos for reference)
> [image]
> [image]
> [image]
>
> Schottky diodes should be also ok: [...]

My 5 x boards was made by PCBWay and I have those LT1945's with the e markings as the other was out of stock

[image]

mcer12, 2022-04-22:
To sum this up, use flux, check for solder bridges and buy from good sources. Closing.

closed 2022-04-22 by mcer12