vroland/epdiy #78: Can't draw images or text over non flat color

fdev31, 2021-04-19:
Hi,
I wanted to discuss on matrix but a login/password is requested ...

It's a general issue, I couldn't find a way to draw text or images on top of an existing image, not erasing ALL the pixels.

To illustrate my request, I made a small experiment which works quite well, for fonts, extending the "EPD_DRAW_BACKGROUND" meaning:

```diff
diff --git a/src/epd_driver/font.c b/src/epd_driver/font.c
index f8a3563..280bed9 100644
--- a/src/epd_driver/font.c
+++ b/src/epd_driver/font.c
@@ -151,6 +151,7 @@ static enum EpdDrawError IRAM_ATTR draw_char(const EpdFont *font, uint8_t *buffe
     int color_difference = (int)props->fg_color - (int)props->bg_color;
     color_lut[c] = max(0, min(15, props->bg_color + c * color_difference / 15));
   }
+  char background_needed = props->flags & EPD_DRAW_BACKGROUND;
 
   for (int y = 0; y < height; y++) {
     int yy = cursor_y - glyph->top + y;
@@ -166,12 +167,14 @@ static enum EpdDrawError IRAM_ATTR draw_char(const EpdFont *font, uint8_t *buffe
     for (int xx = start_pos; xx < max_x; xx++) {
       uint8_t bm = bitmap[y * byte_width + x / 2];
       if ((x & 1) == 0) {
-        bm = bm & 0xF;
+          bm = bm & 0xF;
       } else {
-        bm = bm >> 4;
+          bm = bm >> 4;
+      }
+      if (background_needed || bm) {
+          color = color_lut[bm] << 4;
+          epd_draw_pixel(xx, yy, color, buffer);
       }
-      color = color_lut[bm] << 4;
-      epd_draw_pixel(xx, yy, color, buffer);
       byte_complete = !byte_complete;
       x++;
     }
```

On images, I would expect some (optional) parameter/flag, called "transparentColor" or "colorKey" where one could say which is the color that behaves "transparent".

vroland, 2021-04-21:
Hi fdev31,
Yes, that sounds like a useful extension! Since you already mostly implemented this: Would you mind submitting a PR for this?

fdev31, 2021-04-23:
Hi @vroland, sure !
I will do it for the scope of my patch, it turns very practical and I didn't find any issue while using it.
I was wondering how to do for the images... duplicating some functions or making the existing ones more complex...
how would you like to discuss it? Else I might do some research + PR.
Thank you for this excellent piece of work, this library is fantastic.

vroland, 2021-04-24:
That is a good question. I think I'd implement it in this function: https://github.com/vroland/epdiy/blob/04854b5244599cf5bb42c0e05c87cf6ec9cf89c5/src/epd_driver/epd_driver.c#L440 
To maintain API compatibility, the body would need to be moved to an internal function which is called by the original one and the one allowing a transparency color.
I wouldn't bother with implementing an optimized version like the epd_copy_to_framebuffer function. 

A PR for this would be very welcome :)

fdev31, 2021-04-27:
I didn't test it (yet), will see when I have time & motivation to improve my e-paper, but looked into it and since 
the code was crystal clear & very easy to update, so I did some draft:
https://github.com/fdev31/epdiy/commit/91c3658765ab2ccadd2d23e1a49af000f0cb317d

Any comment/remark I may look into before testing this & making a PR ?

vroland, 2021-04-27:
At a quick glance, this looks all right to me. I'm a bit undecided on using a pointer for the transparency color though. It could be a bit confusing if it's the same type as the frame buffers. On the other hand, using an integer allows to pass invalid values. Do you have a better idea on how to make the API clear?

fdev31, 2021-04-27:
I updated the fork to illustrate a bit different API (no pointer), makes it simpler to use I guess and API relatively clear.

vroland, 2021-04-29:
Yes, I think that is clearer :) The private function should be `static` though.
Do you have some example to test this?

fdev31, 2021-04-29:
I'll update my fork with the static definition.
In the meantime I cleaned my WIP code a little bit & pushed it so you can test:

https://github.com/fdev31/LilygoT5-weather-photo-display

vroland, 2021-05-06:
Ah, now I see why the transparency could be needed. Looks very nice! :)

vroland, 2021-05-06:
Can I link to you program e.g. in the readme?

fdev31, 2021-05-06:
Thank you!
Of course...
I may need to write some README too then... ;)
Also those are pictures which - I believe are free - but I'm not 100% certain.
I'm currently taking my own pictures to match different types of weather so then I can replace the background images.
Since I didn't finish yet, do you think this can be an issue ? (I think it's quite unlikely but who knows...)

vroland, 2021-05-06:
Yes, some readme would be nice :)
And while it is unlinkely that someone complains, I'm always cautious about that, too. If someone wants to sabotage you for any reason, or even just scans for copyrighted content automatically, you may loose account access, etc. I guess the more people find it the higher the risk.
But if you're taking your own pictures anyway, just tell me when you're done :)

fdev31, 2021-05-07:
Ok, I used my lunch break to push what I had and add more content to the README... should be "good enough", tell me if something still looks bad ;)

martinberlin, 2022-05-14:
What is the actual status here @fdev31 ?

The actual code has been updated quite a few times since last May so if there is no feedback about this at the end of May we better make a new issue with a clear mission and close this one.

martinberlin, 2023-07-30:
@fdev31 would be possible that you make a pull request from latest version in master to introduce this new function?

fdev31, 2023-07-30:
That's exactly what I did 2+ years ago... I thought it had been merged.
So I return the question, what is the status ? ^^

martinberlin, 2023-07-30:
Looks like it’s merged thanks just tell me and I reopen if not:

https://github.com/search?q=repo%3Avroland%2Fepdiy%20epd_draw_rotated_transparent_image&type=code

sorry but as it’s not really associated the pull request with the Issue, and no one is closing it when done, then it’s hard to guess if it’s finished or not.

closed 2023-07-30 by martinberlin