vroland/epdiy #81: Waveshare 7.8 raw support

viorello, 2021-04-27:
Hi,
amazing work with this board! I was looking for something like this for ages now. 
Do you think there is a way to use this board for a 7.8 inch raw e-epaper from Waveshare? 
You think there is any phisical limitation (like not enough pinouts from the esp32) or just a matter of adapting connection and driver?

I attached the connection pins

Thank you in advance

[image]

vroland, 2021-04-29:
Since this has 16 data lines, I'm not sure how they are driven. In any case, that's too many pins for our current setup :/ It could be possible with some additional shift register, but at a reduced speed. And it will certainly require some experimentation.

martinberlin, 2022-05-10:
Closing here since it will need a lot of research to this. It's also IMHO not possible with existing MCU. We should make a version with S2 or S3 that have a lot more of GPIOs.
Then we can reopen this one and rethink how to do it.

For those 16 data-lines there are already commercial solutions available from Good display and Waveshare, a lot of research has to be done, and no one knows if the speed will be enough to update this epaper models consistently.

closed 2022-05-10 by martinberlin

GoReNeY, 2024-10-16:
Is there any updates on 16 data lines support? With v7 board version maybe?

martinberlin, 2024-10-16:
Yes @GoReNeY please check v7 hardware repository it already supports 16 data lines for displays with the 40 pins connector