vroland/epdiy #84: Issue with Vcom (lowest value -2.5V)

Schmirgus, 2021-05-07:
Hello :)
i accidently got the orientation of the op amp (LM358) wrong. Now i  corrected the orientation and used a new LM358.  The issue i have now is, that i cant get a VCOM voltage lower then -2.5V (max is ~ -6.5V). I desoldered R14 and confirmed that the value is correct. The highest value of the potentiometer is ~163k Ohms and the lowest value 130 Ohms (measured on PCB). Any idea what could be the issue or am i missing something?
When i simulate the schematic in ltspice with my measured values from the PCB it should work fine... 

measured at the lm358:
Pin 8 15V
Pin 4 -15V

best regards,
Marco ;)

Schmirgus, 2021-05-07:
i hate myself... so the last thing i could think of was that maybe of the solder joints might be faulty... and yes that was the issue :( great to find the error everytime i open an issue

closed 2021-05-07 by Schmirgus

vroland, 2021-05-08:
Well, if it helps feel free :D