vroland/epdiy #85: Do you offer a commercial license?

jockm, 2021-05-13:
I am considering developing a small commercial product and would like to use your code as a basis for the epaper driver.  Do you offer a commercial license?

jockm, 2021-05-13:
Or alternately a LGPL version.  I am more than happy to contribute my epaper code back, just not the larger application

vroland, 2021-05-14:
I did not consider it yet, but I'm not against it. Could you send me an e-Mail and describe what kind of product you are developing? You can find my address on my profile.

vroland, 2021-05-18:
For future record: Offering a commercial license is not possible, as significant portions of the code where contributed to epdiy under the GPL. 
For using epdiy in a commercial project, you must comply with the terms of the GPL.

closed 2021-05-18 by vroland

jockm, 2021-05-18:
Thank you again for looking into this