vroland/epdiy #86: FT232h

amadeok, 2021-05-22:
Hey 
I saw your messages about the ft232h just today sorry so how it is going ?

vroland, 2021-05-22:
I'm currently waiting for some parts and a prototype pcb, because the breadboard is just too unstable. Sometimes a waveform changes just by getting my hand near it ;) for the pcb I'm also using some SMD parts, which are rated for higher frequencies.

cxw0102, 2021-08-23:
@vroland,can the new PCB support TF card?

vroland, 2021-08-23:
***@***.***,can the new PCB support TF card?

What do you mean with "TF card"?

cxw0102, 2021-08-24:
@vroland,sorry,i mean micro SD

vroland, 2021-08-24:
There is no card reader on the board. But you can add one with the available gpios. That is, if you are talking about epdiy. The ft232h board is a separate project.

cxw0102, 2021-08-25:
@vroland yes,wait your ft232h board.epdiy board not enough Gpios support SD。

vroland, 2021-08-25:
epdiy should have enough GPIOs. For v5 / v6, 4 GPIOs are available and two more input-only ones. This is enough for connecting an SD card over SPI.
I don't really see a benefit of adding a card reader to the ft232h board, since it only works on a usb-capable computer anyway, so you could attach the SD card to this computer.

martinberlin, 2024-04-07:
@vroland this issue is stale and a bit old, is this ft232 board still going to be released?
Or maybe we can convert this into a discussion?
I think we should move to discussions all this kind of ideas or feature requests that are not really an Issue within the component.

amadeok, 2024-04-08:
> @vroland this issue is stale and a bit old, is this ft232 board still going to be released? Or maybe we can convert this into a discussion? I think we should move to discussions all this kind of ideas or feature requests that are not really an Issue within the component.

hey man i remember he made a couple prototypes but there were some issues, i would imagine since the ft232h has to be clocked very high to achieve usb hi-speed (480mbps). 
the esp32-s3 supports usb 2.0, but only full-speed (like 12mbps ) 
both the the esp32-s3 and the older esp32 board say they supports wifi with "1T1R mode with data rate up to 150 Mbps"  but i never achieve even closed to that with esp32-wrover. i wonder if there's a way to make the wifi work at 150 mbps

martinberlin, 2024-04-08:
I see, it was in fact an interesting experiment but the ESP32 speed cannot cope with the USB speed. What a pity I would have loved to see that working...

amadeok, 2024-04-08:
> I see, it was in fact an interesting experiment but the ESP32 speed cannot cope with the USB speed. What a pity I would have loved to see that working...

there are other microcontrollers that have built-in usb hi-speed 480mbps, like the [stm32f7x3](https://www.st.com/en/microcontrollers-microprocessors/stm32f7x3.html)
however they seem to be harder to find /more expensive than the esp32, and also probably we would need  to write the firmware and probably a custom board

martinberlin, 2024-04-25:
@amadeok if you don't mind I'm moving this issue to Discussions since it's really a different board than epdiy. I'm really willing to try this and it sounds like a very interesting project