vroland/epdiy #87: Unable to use fontconvert.py

hacksics, 2021-05-31:
I'm unable to convert fonts and getting bellow error. I googled it but couldn't find any answer 

```
 python fontconvert.py OpenSans10B 10 OpenSans-Bold.ttf > opensans10b.h
falling back to font 1 for ‐.
Traceback (most recent call last):
  File "fontconvert.py", line 91, in <module>
    face = load_glyph(code_point)
  File "fontconvert.py", line 87, in load_glyph
    raise ValueError(f"code point {code_point} not found in font stack!")
ValueError: code point 8208 not found in font stack!
```

vroland, 2021-05-31:
That's because you specified a character range in the `intervals`-section of the script your font has no glyphs for. You can eiter comment that character range out in https://github.com/vroland/epdiy/blob/master/scripts/fontconvert.py#L27 or specify an additional font that contains it (e.g. Symbola or maybe even OpenSans-Regular.ttf) after OpenSans-Bold.ttf.

hacksics, 2021-06-01:
thanks @vroland  I've tried with commenting the intervals and it worked just file. Previously I tried this commenting few but it looks like I need to comment all except first two. Thanks for the help. I'm closing the ticket.

martinberlin, 2022-07-02:
> That's because you specified a character range in the intervals-section of the script

Sorry but we all due respect it's a pain to use **fontconvert.py** as is right now. 
It's really not user friendly for a developer that is creating something with this component. If the character range specified does not exist then it should skip it but still generate the font leaving a warning in the command-line.

Yardie-, 2022-07-03:
IMHO the epdiy version is actually superior to those of other systems. So making it compatible with other systems would be a big step backwards. I have modified my version of this script and will discuss the changes with the appropriate people.

martinberlin, 2022-07-03:
Sure @Yardie- I completely agree on this point.
But in my opinion the default script ranges should be modified so someone that just tries to use this for the first time does not get such a blocking errors.

DongBH, 2023-06-03:
Hey, I try some modifications, it will skip the code_point with no glyphs.
```
#!python3
import freetype
import zlib
import sys
import re
import math
import argparse
from collections import namedtuple

parser = argparse.ArgumentParser(description="Generate a header file from a font to be used with epdiy.")
parser.add_argument("name", action="store", help="name of the font.")
parser.add_argument("size", type=int, help="font size to use.")
parser.add_argument("fontstack", action="store", help="font files")
parser.add_argument("--compress", dest="compress", action="store_true", help="compress glyph bitmaps.")
parser.add_argument("--additional-intervals", dest="additional_intervals", action="append", help="Additional code point intervals to export as min,max. This argument can be repeated.")
args = parser.parse_args()

GlyphProps = namedtuple("GlyphProps", ["width", "height", "advance_x", "left", "top", "compressed_size", "data_offset", "code_point"])

faces = freetype.Face(args.fontstack)
compress = args.compress
size = args.size
font_name = args.name

# inclusive unicode code point intervals
# must not overlap and be in ascending order
intervals = [
    (0x21, 0x9F9F),
]

add_ints = []
if args.additional_intervals:
    add_ints = [tuple([int(n, base=0) for n in i.split(",")]) for i in args.additional_intervals]

intervals = sorted(intervals + add_ints)

def norm_floor(val):
    return int(math.floor(val / (1 << 6)))

def norm_ceil(val):
    return int(math.ceil(val / (1 << 6)))

faces.set_char_size(size << 6, size << 6, 150, 150)

def chunks(l, n):
    for i in range(0, len(l), n):
        yield l[i:i + n]

total_size = 0
total_packed = 0
all_glyphs = []
height = 0
ascender = 0
descender = 0

def load_glyph(code_point):
    glyph_index = faces.get_char_index(code_point)
    if glyph_index > 0:
        faces.load_glyph(glyph_index, freetype.FT_LOAD_RENDER)
        return faces
    else:
        return None
      
#    raise ValueError(f"code point {code_point} not found in font stack!")

stat = False
intv = [];
start = end = 0;
for i_start, i_end in intervals:
    for code_point in range(i_start, i_end + 1):
        face = load_glyph(code_point)
        if face == None:
            if stat == True:
                intv.append((start, end))
                stat = False
            continue
        
        if stat == True:
            end = code_point
        else:
            stat = True
            start = end = code_point

        bitmap = face.glyph.bitmap
        pixels = []
        px = 0
        for i, v in enumerate(bitmap.buffer):
            y = i / bitmap.width
            x = i % bitmap.width
            if x % 2 == 0:
                px = (v >> 4)
            else:
                px = px | (v & 0xF0)
                pixels.append(px);
                px = 0
            # eol
            if x == bitmap.width - 1 and bitmap.width % 2 > 0:
                pixels.append(px)
                px = 0

        packed = bytes(pixels);
        total_packed += len(packed)
        compressed = packed
        if compress:
            compressed = zlib.compress(packed)

        glyph = GlyphProps(
            width = bitmap.width,
            height = bitmap.rows,
            advance_x = norm_floor(face.glyph.advance.x),
            left = face.glyph.bitmap_left,
            top = face.glyph.bitmap_top,
            compressed_size = len(compressed),
            data_offset = total_size,
            code_point = code_point,
        )
        total_size += len(compressed)
        all_glyphs.append((glyph, compressed))

        height = face.size.height
        ascender =face.size.ascender
        descender = face.size.descender 

glyph_data = []
glyph_props = []
for index, glyph in enumerate(all_glyphs):
    props, compressed = glyph
    glyph_data.extend([b for b in compressed])
    glyph_props.append(props)

print("total", total_packed, file=sys.stderr)
print("compressed", total_size, file=sys.stderr)

print("#pragma once")
print("#include \"epd_driver.h\"")
print(f"const uint8_t {font_name}Bitmaps[{len(glyph_data)}] = {{")
for c in chunks(glyph_data, 16):
    print ("    " + " ".join(f"0x{b:02X}," for b in c))
print ("};");

print(f"const EpdGlyph {font_name}Glyphs[] = {{")
for i, g in enumerate(glyph_props):
    print ("    { " + ", ".join([f"{a}" for a in list(g[:-1])]),"},", f"// {g.code_point:#0x}")
print ("};");

print(f"const EpdUnicodeInterval {font_name}Intervals[] = {{")
offset = 0
for i_start, i_end in intv:
    print (f"    {{ 0x{i_start:X}, 0x{i_end:X}, 0x{offset:X} }},")
    offset += i_end - i_start + 1
print ("};");

print(f"const EpdFont {font_name} = {{")
print(f"    {font_name}Bitmaps,")
print(f"    {font_name}Glyphs,")
print(f"    {font_name}Intervals,")
print(f"    {len(intv)},")
print(f"    {1 if compress else 0},")
print(f"    {norm_ceil(height)},")
print(f"    {norm_ceil(ascender)},")
print(f"    {norm_floor(descender)},")
print("};")
```

DongBH, 2023-06-03:
Hey, I try some modifications, and it will skip the code_point with no glyphs.
`#!python3
import freetype
import zlib
import sys
import re
import math
import argparse
from collections import namedtuple

parser = argparse.ArgumentParser(description="Generate a header file from a font to be used with epdiy.")
parser.add_argument("name", action="store", help="name of the font.")
parser.add_argument("size", type=int, help="font size to use.")
parser.add_argument("fontstack", action="store", help="font files")
parser.add_argument("--compress", dest="compress", action="store_true", help="compress glyph bitmaps.")
parser.add_argument("--additional-intervals", dest="additional_intervals", action="append", help="Additional code point intervals to export as min,max. This argument can be repeated.")
args = parser.parse_args()

GlyphProps = namedtuple("GlyphProps", ["width", "height", "advance_x", "left", "top", "compressed_size", "data_offset", "code_point"])

faces = freetype.Face(args.fontstack)
compress = args.compress
size = args.size
font_name = args.name

# inclusive unicode code point intervals
# must not overlap and be in ascending order
intervals = [
    (0x21, 0x9F9F),
]

add_ints = []
if args.additional_intervals:
    add_ints = [tuple([int(n, base=0) for n in i.split(",")]) for i in args.additional_intervals]

intervals = sorted(intervals + add_ints)

def norm_floor(val):
    return int(math.floor(val / (1 << 6)))

def norm_ceil(val):
    return int(math.ceil(val / (1 << 6)))

faces.set_char_size(size << 6, size << 6, 150, 150)

def chunks(l, n):
    for i in range(0, len(l), n):
        yield l[i:i + n]

total_size = 0
total_packed = 0
all_glyphs = []
height = 0
ascender = 0
descender = 0

def load_glyph(code_point):
    glyph_index = faces.get_char_index(code_point)
    if glyph_index > 0:
        faces.load_glyph(glyph_index, freetype.FT_LOAD_RENDER)
        return faces
    else:
        return None
      
#    raise ValueError(f"code point {code_point} not found in font stack!")

stat = False
intv = [];
start = end = 0;
for i_start, i_end in intervals:
    for code_point in range(i_start, i_end + 1):
        face = load_glyph(code_point)
        if face == None:
            if stat == True:
                intv.append((start, end))
                stat = False
            continue
        
        if stat == True:
            end = code_point
        else:
            stat = True
            start = end = code_point

        bitmap = face.glyph.bitmap
        pixels = []
        px = 0
        for i, v in enumerate(bitmap.buffer):
            y = i / bitmap.width
            x = i % bitmap.width
            if x % 2 == 0:
                px = (v >> 4)
            else:
                px = px | (v & 0xF0)
                pixels.append(px);
                px = 0
            # eol
            if x == bitmap.width - 1 and bitmap.width % 2 > 0:
                pixels.append(px)
                px = 0

        packed = bytes(pixels);
        total_packed += len(packed)
        compressed = packed
        if compress:
            compressed = zlib.compress(packed)

        glyph = GlyphProps(
            width = bitmap.width,
            height = bitmap.rows,
            advance_x = norm_floor(face.glyph.advance.x),
            left = face.glyph.bitmap_left,
            top = face.glyph.bitmap_top,
            compressed_size = len(compressed),
            data_offset = total_size,
            code_point = code_point,
        )
        total_size += len(compressed)
        all_glyphs.append((glyph, compressed))

        height = face.size.height
        ascender =face.size.ascender
        descender = face.size.descender 

glyph_data = []
glyph_props = []
for index, glyph in enumerate(all_glyphs):
    props, compressed = glyph
    glyph_data.extend([b for b in compressed])
    glyph_props.append(props)

print("total", total_packed, file=sys.stderr)
print("compressed", total_size, file=sys.stderr)

print("#pragma once")
print("#include \"epd_driver.h\"")
print(f"const uint8_t {font_name}Bitmaps[{len(glyph_data)}] = {{")
for c in chunks(glyph_data, 16):
    print ("    " + " ".join(f"0x{b:02X}," for b in c))
print ("};");

print(f"const EpdGlyph {font_name}Glyphs[] = {{")
for i, g in enumerate(glyph_props):
    print ("    { " + ", ".join([f"{a}" for a in list(g[:-1])]),"},", f"// {g.code_point:#0x}")
print ("};");

print(f"const EpdUnicodeInterval {font_name}Intervals[] = {{")
offset = 0
for i_start, i_end in intv:
    print (f"    {{ 0x{i_start:X}, 0x{i_end:X}, 0x{offset:X} }},")
    offset += i_end - i_start + 1
print ("};");

print(f"const EpdFont {font_name} = {{")
print(f"    {font_name}Bitmaps,")
print(f"    {font_name}Glyphs,")
print(f"    {font_name}Intervals,")
print(f"    {len(intv)},")
print(f"    {1 if compress else 0},")
print(f"    {norm_ceil(height)},")
print(f"    {norm_ceil(ascender)},")
print(f"    {norm_floor(descender)},")
print("};")
`

DongBH, 2023-06-03:
I try some modification,  it will skip the code_point with no glyphs .
[fontconvert2.zip](https://github.com/vroland/epdiy/files/11642185/fontconvert2.zip)

martinberlin, 2023-06-03:
Nice @DongBH thanks for your collaboration. 
Can you please make a pull request from current master branch so we can review your updates?
That would be the right way to collaborate in an open source repository.

Apart of my critics to the font generator, that I must have written in a day with bad humor, I also like a lot epdiy fonts system.

DongBH, 2023-06-04:
I'll try, tks.

Yardie-, 2023-06-10:
I thought I had already solved this with the string code but maybe not. If you use the string and the code point is not there it should report it and move on.

Yardie-, 2023-06-12:
Yeah I did this in my previous pull request on 26 /9/ 22
  ```
      # handle missing characters in font file
        if( face.get_char_index(code_point) == 0 ):
            print("Character ", chr(code_point), "(", code_point, ") is not in ", font_file, file=sys.stderr)
            continue
 
```

martinberlin, 2023-06-12:
Thanks Yardie.
@hacksics can you please confirm this is solved?
Then we can simply close this issue

martinberlin, 2023-07-30:
Solved by @Yardie-

closed 2023-07-30 by martinberlin

joshbellblogs, 2025-09-29:
I like the idea suggested above of skipping missing glyphs but still providing a clear warning with the exact character and code point.

For anyone experimenting with text styles or checking character coverage before going into heavy scripts like fontconvert.py, I’ve found that using quick font generators or preview tools can save a lot of time. For example, on [Unitextify](https://unitextify.com/)
 you can test different Unicode characters and fonts instantly, which helps spot these gaps before conversion. That way you know upfront if your chosen font actually supports the symbols you need.

I think combining that kind of pre-check with the improvements discussed here would make the workflow much smoother for both new and experienced developers.