vroland/epdiy #88: EC060KH1 color panel support

regishsu, 2021-06-08:
Hi @vroland ,
amazing work to support those panels. 
Do you think there is a way to use this board for a 6 inch CFA (Color Filter Array) panel?
The controller timing is the same as ED060SC4, however it seems need another lookup table to show colors.
Thank you in advance.

Regis

vroland, 2021-06-08:
Do you have a data sheet for this? If the signals are the same it might work, but you'll have to figure out the timings and waveforms yourself.

martinberlin, 2022-02-26:
@regishsu can you please give more information about this display?
Seems like an Eink kaleido display that is used in the Pocketbook. But there is no datasheet online that I could find, neither any info about the color lookup table you mention.
If you have more information I would also want to try one of this epapers but don’t know here to buy them.
@vroland if there is no feedback in the next month in my opinion we could close here.

regishsu, 2022-02-27:
Hi, Martin,
I have dropped this project because Eink doesn't provide the waveform to me.
Thanks for EDPIY project!
Regis

martinberlin, 2022-05-14:
Closing since it seems there is nothing more to do here

vanarebane, 2023-11-20:
Hello, I have the [datasheet](https://1drv.ms/b/s!ApRLRqw9Uoyuvpo5pdrEOj-P9l8xAQ?e=0cSGSV)

martinberlin, 2023-11-20:
By the way got a client request to implement this so I will receive in some weeks one Kaleido display. Closing here and keeping #266 so there is no repeated issues

closed 2023-11-20 by martinberlin