vroland/epdiy #89: Image not getting displayed properly

Schmirgus, 2021-06-08:
Hello :)
two weeks ago i finally found a vendor who allegedly sold me an "ED097OC4". Unfortunately there arent any stickers for the VCOM or the display type. I tried different VCOM voltages in a range of -1 to -3V with no differents. Also i tried the ED970TC2, ED970OC4 and ED970OC4 low quality in the config with no difference. When i unpacked the display the contrast etc seemed fine.
[image]
Has anyone seeing a similiar issue ?
https://www.youtube.com/watch?v=NaMCH7J34ug

thanks :)

Schmirgus, 2021-06-08:
i incread the VCOM to -4V and the pictures is getting better  :/ any ideas?

vroland, 2021-06-08:
Hm, this does indeed look like a VCOM or display type issue. But I've not yet seen displays with VCOM voltages in excess of -3V.
Does this effect change if you select different display types like SC4 or 047TC2?

Schmirgus, 2021-06-09:
I guess the quality of the display isnt that great. There must be a reason why there isnt a frame around the glas and the sticker is missing. Maybe this display got sorted out? I wonder tho why the original picture that was displayed on the screen looked fine.

Nevertheless increasing the VCOM -5V gave me the best quality but then the whites are obviously pretty bad. The ED06SC4 and the 047TC1 settings definitely resulted in a better picture quality.

vroland, 2021-06-17:
Sorry for the late reply. With these displays its always hard to know if they have some previous fault or are just slightly different models. Btw, did you measure that all the voltages are correct on your board? Maybe there is something wrong with them. Maybe test your board with another display?

Schmirgus, 2021-08-24:
Hey ;) sorry for the late reply, i was finishing my thesis. Unfortunately i dont got a second display :/. Unfortunately i dont got an oscilloscope to check all the voltages properly.

vroland, 2021-08-25:
Hm, maybe. Can you upload an image of your board? Maybe some control pins are bridged on the connector? But I'm running out of ideas, so I tend to blame the display :D

mcer12, 2022-04-22:
Closing, stale. @Schmirgus feel free to reopen once you have the required test equipment.

closed 2022-04-22 by mcer12