vroland/epdiy #90: Question concerning battery life with example lilygo-t5-47-epd-platformio

jchodk, 2021-06-08:
Hello,

I'm using a [slightly modified version](https://github.com/jchodk/mRoubaud/) of your lilygo-t5-47-epd-platformio example adding a timer as wake up method with a one minute delay and putting the esp in deep sleep mode just after refreshing the display ; so all the code in *setup*. To my great disappointment with the 18650 version, the battery is fully drained only after few hours, results aren't really better with hourly wake up.  Have you obtained so a limited autonomy in your tests ? 

Furthermore, after few hours running there are two "[dead rows](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/50)". I never see such defect in commercial e-reader. Could it be an hardware defect ?

holgerlembke, 2021-06-09:
just fishing.

try epd_poweroff_all();

jchodk, 2021-06-09:
@holgerlembke, 
Thanks for the tip. I also read that using only timer wake up -my use case does not need button wake up- you could induce a "deeper" sleep and reduce power consumption. 

I would be more motivated to further expriment if someone can confirmed he gets months autonomy with a daily wake up, as I was expecting starting my project

holgerlembke, 2021-06-09:
I know I'm not much help. I could point out https://github.com/vroland/epdiy/issues/21#issuecomment-672070794 whit my experience with an TTGO T5 V 2,3 ESP-32 e-ink paper.

To maximize your frustration (sorry :) I don't think TTGO cares much about what they do and if it works. The eject new boards and move one, documentation to a minimum and no addendums, nothing, wasteland.

Are there any schematics for that board? I found something but who knows. First look and I found a blue LED driven by the batterie. Like mine.

jchodk, 2021-06-09:
I think TTGO isn't the only one working that way and not the worst ; do you know Banana Pi? TTGO is a paramount of professionalism compared to Sinovoip. 
When in deep sleep *and* battery mode, blue led is off with T7 4.7. Well your result with 28 days is one order of magnitude better than mine! Maybe my expectations were absurd, after all my Kindle's battery (in sleep mode ?) dries off after less than a month, and I guess theirs engineers aren't to bad at their craft...

holgerlembke, 2021-06-09:
As for the ESP32 deep sleep starts with 150 µA down to 10 µA, hibernation goes down to 5 µA.

So with a 2300 mAh 18650 cell it should reach more than 600 days. Theoretically.

Looks like there is a different sucker on that board.

vroland, 2021-06-17:
Hm, I did not do a lot of experimentation with the lilygo board I must admit. With the V5 board I got down to 13uA in deep sleep, but I don't thing the lilygo board is very concerned with low deep sleep current, as it looks like the whole serial circuitry is not isolated when in battery mode.

jchodk, 2021-06-17:
Anyway, battery life isn't really my concern anymore. As explained in my first message, after few days my TTGO displayed (more adequately *does not displayed*) two faulty lines. But after my last experiments it's nearly one third of the screen.... The symptoms are more obvious during screen refresh as the concerned area stay white. Same issue with different battery ou with USB power. So the board is quite useless now. too bad.

martinberlin, 2021-06-17:
So it’s the display corrupted? I’m experimenting with one and user interfaces so I stress it to refresh only part of the screen many times a day and I still didn’t see any of this

jchodk, 2021-06-18:
Few days ago my screen shows few "dead" rows using TTGO 
image example : 
[image]
Today I got that :
[image]
Maybe I find the culprit, there is something wrong on the screen corner :
[image]
As I'm pretty sure the board hadn't any shock, I ruined it only by mounting it a frame. Quite a delicate thing...

vroland, 2021-06-22:
Hm, that looks bad :/ And these stripes appear always in the same position?
I didn't have any problems with mine so far, but I it blew a fuse once I had to replace...

jchodk, 2021-06-22:
Yes always at the same place but the stripes are "spreading". I'm not sure if it's visible on the second picture but there is some part of the outer layer of the screen (not the protection film wich was stripped off long ago!) wich appears torned, cracked. Well, it will join my "not even epic fail" drawer... It's quite infuriating : for once I was quite satisfied with my project enclosure :). Thanks for your replies.

vroland, 2021-06-22:
Ok, then it is probably indeed the the delicate traces on the glass breaking. That's unfortunate :(

lixy123, 2021-08-08:
Use master-slave mode. The ink screen is a slave. A low-power Bluetooth module is installed outside the ink screen. After receiving the text information, the host module wakes up the ink screen slave in the air and triggers the slave to display the text. After a few seconds, the ink screen returns to sleep. In essence, the power is transferred to the host.
see my examples:
https://github.com/lixy123/LilyGo-EPD47-BLYNK

Occasionally rotate to display text 1-2 times a day. The 18650 battery can support about 2-3 months.

mcer12, 2022-04-22:
Closing. For anyone else having issues with lilygo and other boards, report please the issues in lilygo repository.

closed 2022-04-22 by mcer12